raspberrypi/firmware #1606: Error: "end kernel panic - not syncing: systems is deadlocked on memory" After full OS upgrade 02.08.2021

2play, 2021-08-03:
First boot after full OS upgrade and latest boot files.

[image]

OS: Linux 5.10.17-v7l+ armv7l GNU/Linux - Raspbian GNU/Linux 10 (buster)
BOOTLOADER: up to date
   CURRENT: Tue 06 Jul 2021 10:44:53 AM UTC (1625568293)
    LATEST: Tue 06 Jul 2021 10:44:53 AM UTC (1625568293)
   RELEASE: stable (/lib/firmware/raspberrypi/bootloader/stable)
            Use raspi-config to change the release.

  VL805_FW: Dedicated VL805 EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1

If i revert to my back up boot partition with files dated 21.07.2021
System will boot to cli but error about kernel cant load overlay modules appear and similar
adding only new elf and dat files give same overlay errors.
updating overlay folder will cause again the title error

Also Im providing a txt file with upgrades took place during today's OS update/upgrade process
[pi4 update 02.08.2021.txt](https://github.com/raspberrypi/firmware/files/6924582/pi4.update.02.08.2021.txt)

tx in advance for any feedback

popcornmix, 2021-08-03:
Can you post contents of config.txt and cmdline.txt files?
If you have a gpu_mem entry then try removing it.

2play, 2021-08-03:
@popcornmix 
I do have. I commented out below, and used `sudo apt-get update; sudo apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel`
this to got to latest stable kernel (now up to 5.10.52)

`#gpu_mem_256=128
#gpu_mem_512=256
#gpu_mem_1024=384
#gpu_mem_2048=512
#gpu_mem_4096=512
#gpu_mem_8192=512
`
All ok! thank you!
ill close the issue...

closed 2021-08-03 by 2play