raspberrypi/firmware #1607: PoE+ HAT Fan not working at lower state

diasdmhub, 2021-08-04:
I've got 2 Pi's with the new PoE+ Hat and both were working fine with the correct fan curve set with "dtoverlay=rpi-poe-plus".
In "_/boot/config.txt_" I've set them like this:
```
dtoverlay=rpi-poe-plus
dtparam=poe_fan_temp0=45000,poe_fan_temp0_hyst=3000
dtparam=poe_fan_temp1=50000,poe_fan_temp1_hyst=3000
dtparam=poe_fan_temp2=62000,poe_fan_temp2_hyst=4000
dtparam=poe_fan_temp3=70000,poe_fan_temp3_hyst=5000
```

After the latest firmware update, I noticed that _fan_temp0_ and _fan_temp1_ are not working anymore.
The fan only spins up when the temperature reaches 62ºC or higher.
When the fan is not spinning it shows state "2" and when it is spinning, shows state 3.
```
$ cat /sys/class/thermal/cooling_device0/cur_state
2
```

It all started out of the blue with both PoE+ HATs, right after I rebooted and the kernal was updated to version "**5.10.52-v8+**". Prior version "**5.10.17-v8+**" was working fine.

"od" shows me the correct trip values I set with dtparam, still states 1 and 2 don't spin the fan up.
```
$ od -An --endian=big -td4 /proc/device-tree/thermal-zones/cpu-thermal/trips/trip?/temperature /proc/device-tree/thermal-zones/cpu-thermal/trips/trip?/hysteresis
       45000       50000       62000       70000
        3000        3000        4000        5000
```

I've reinstalled the Raspberry Pi OS and, with a clean installation, the strange behavior is the same, no spin at states 1 and 2.
It seems not to work at lower speeds anymore.

When I list overlays, nothings shows up.
```
# dtoverlay -l
No overlays loaded
```

And when I try to load "_rpi-poe-plus_" overlay, it fails.
```
$ sudo dtoverlay -v rpi-poe-plus
DTOVERLAY[debug]: using platform 'bcm2711'
DTOVERLAY[debug]: overlay map loaded
run_cmd: which dtoverlay-pre >/dev/null 2>&1 && dtoverlay-pre
DTOVERLAY[debug]: loading file '/boot/overlays/rpi-poe-plus.dtbo'
DTOVERLAY[debug]: wrote 2938 bytes to '/tmp/.dtoverlays/0_rpi-poe-plus.dtbo'
DTOVERLAY[debug]: wrote 2938 bytes to '/sys/kernel/config/device-tree/overlays/0_rpi-poe-plus/dtbo'
* Failed to apply overlay '0_rpi-poe-plus' (kernel)
run_cmd: which dtoverlay-post >/dev/null 2>&1 && dtoverlay-post
```

Also, "dmesg" shows nothing related to PoE.

DTC shows me:
```
# dtc -I fs /proc/device-tree
rpi-poe-fan@0 {
		cooling-levels = < 0x00 0x01 0x0a 0x64 0xff >;
		cooling-max-state = < 0x04 >;
		compatible = "raspberrypi,rpi-poe-fan";
		status = "okay";
		firmware = < 0x06 >;
		phandle = < 0xea >;
		cooling-min-state = < 0x00 >;
		#cooling-cells = < 0x02 >;
	};
```

**Any ideas on why that is happening with both HAT's?**

**MY SYSTEM:**
```
Raspberry Pi 4 Model B Rev 1.4 
```
```
 $ cat /etc/rpi-issue
Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9a3a10bf1019ebb2d59053564dc6b90068bad27d, stage2
```
```
$ vcgencmd version
Aug  3 2021 18:14:56
Copyright (c) 2012 Broadcom
version 40787ee5905644f639a2a0f6e00ae12e517a2211 (clean) (release) (start)
```
```
$ uname -a
Linux [hostname] 5.10.52-v8+ #1441 SMP PREEMPT Tue Aug 3 18:14:03 BST 2021 aarch64 GNU/Linux
```

uzlonewolf, 2021-08-04:
What happens if you delete the dtoverlay= line and leave only the dtparam= lines?  On the 3B+'s I was never able to get it to accept the new temperatures if I included the overlay, but it worked fine if it didn't.  Something about the hat auto-loading the overlay and the dtoverlay= line loading it a 2nd time, wiping the params.

diasdmhub, 2021-08-04:
I commented the "dtoverlay" line and left "dtparam" as you suggested.
```
#dtoverlay=rpi-poe-plus
dtparam=poe_fan_temp0=45000,poe_fan_temp0_hyst=3000
dtparam=poe_fan_temp1=50000,poe_fan_temp1_hyst=3000
dtparam=poe_fan_temp2=62000,poe_fan_temp2_hyst=4000
dtparam=poe_fan_temp3=70000,poe_fan_temp3_hyst=5000
```
After that, "od" shows that "dtparam" is correct, but the odd behavior is the same. There's no fan spin for levels 1 and 2.
```
$ od -An --endian=big -td4 /proc/device-tree/thermal-zones/cpu-thermal/trips/trip?/temperature /proc/device-tree/thermal-zones/cpu-thermal/trips/trip?/hysteresis
       45000       50000       62000       70000
        3000        3000        4000        5000
```

chri-kai-in, 2021-08-10:
Hi,

how many GB RAM have the rpi's ? I face the situation POE+ fan works fine with 8GB derivative but declines running with the 4GB derivative. It will not work with the latest kernel nor firmware updates and even when i adjust the kernel version on 4GB derivative back to 5.0.17-v71+ and adjusted the firmware to have the same release date as on the 8GB derivative (03.09.2020) - no improvement.

I have two POE+ hats which both show correct fan behaviour on the 8GB derivative and no fan running on the 4GB derivative. I can cross-change SD cards and POE+ hats - behavoir stays the same. I expect it must be something depending on the board - firmware, whatsoever...

Meanwhile my 8GB system was updated to kernel 5.0.52-v71+ and firmware is still as of 03.09.2020 and the fan is working.
The same combination does not work on my 4GB derivative.

diasdmhub, 2021-08-10:
> Hi,
> 
> how many GB RAM have the rpi's ? I face the situation POE+ fan works fine with 8GB derivative but declines running with the 4GB derivative. It will not work with the latest kernel nor firmware updates and even when i adjust the kernel version on 4GB derivative back to 5.0.17-v71+ and adjusted the firmware to have the same release date as on the 8GB derivative (03.09.2020) - no improvement.
> 
> I have two POE+ hats which both show correct fan behavior on the 8GB derivative and no fan running on the 4GB derivative. I can cross-change SD cards and POE+ hats - behavior stays the same. I expect it must be something depending on the board - firmware, whatsoever...
> 
> Meanwhile my 8GB system was updated to kernel 5.0.52-v71+ and firmware is still as of 03.09.2020 and the fan is working.
> The same combination does not work on my 4GB derivative.

Both RPi's have 8GB.

chri-kai-in, 2021-08-11:
Too bad - another theory scattered...

With the 03.09.2020 release of the firmware it worked on my 8GB derivative with fan on as well in level 2.

Btw. i moved the rasbian image from the 4GB to the 8GB device and the fan was started but only on level 3 as you state it. 

In addition - after updating the firmware on the 8GB derivative to latest release the behavoir with the raspbian image changed as well to fan on with level 3 only.

After switching back to kernel 5.0.17-V71+ and bootloader firmware as of 03.09.2020 the behavior stays the same - fan working but only starting with level 3.

Maybe the cooling levels in the *.dts file for level 1 and 2 are too low ?

```
	compatible = "brcm,bcm2835";

	fragment@0 {
		target-path = "/";
		__overlay__ {
			fan0: rpi-poe-fan@0 {
				compatible = "raspberrypi,rpi-poe-fan";
				firmware = <&firmware>;
				cooling-min-state = <0>;
				cooling-max-state = <4>;
				#cooling-cells = <2>;
				cooling-levels = <0 1 10 100 255>;
				status = "okay";
			};
		};
	};
```
If the levels reflect PWM values to be applied to the output i doubt that a PWM level of 1 and 10 will let the fan start running for cooling level 1 & 2.

Unfortunately my level of knowledge for devicetree and creation of overlay files or manipulating the device tree does not allow me to prove this theory...

chri-kai-in, 2021-08-11:
Proven !

Following steps i applied:

1) derive initial *.dts file from exisitng *,dtbo fiel in folder /boot/overlays
``` sudo dtc -I dtb -O dts -o /boot/overlays/rpi-poe-plus-overlay.dts /boot/overlays/rpi-poe-plus.dtbo```

2) adjust the "cooling levels line" within this file to desired values in nano/vi and save the changes when leaving the editor
``` sudo nano /boot/overlays/rpi-poe-plus-overlay.dts```

In my case the following values worked, but i reccomend to use more uneven values to scatter the frequencies of the fan and thereby maybe lower the overall noise.
```cooling-levels = < 0x00 0x28 0x46 0x80 0xff >;```

3) backing up the original *.dtbo file, so we would be able to step back (just in case)
``` sudo mv /boot/overlays/rpi-poe-plus.dtbo /boot/overlays/rpi-poe-plus-orig.dtbo```

4) creating a replacement *.dtbo file from the adjusted *.dts file
``` sudo dtc -@ -Hepapr -I dts -O dtb -o /boot/overlays/rpi-poe-plus.dtbo /boot/overlays/rpi-poe-plus-overlay.dts```

5) make sure the changes ended up in the *,dtbo file (search for the cooling levels line and compare with your chosen values)
```  sudo fdtdump /boot/overlays/rpi-poe-plus.dtbo```

If all looks good go for a reboot...
``` sudo shutdown -r now ```

Of course - everything stated here worked only on my machine - so do it on you own risk on your machine.

With these changes the fan starts as well in level 1 and level 2. allowing you to adjust the temperature tripping values to your needs.

chri-kai-in, 2021-08-11:
..still this will not improve my limitations with my 4GB derivative - POE+ fan not working on my RPI 4B, 4GB RAM with this configuration. If i move the sd-card from the 4GB derivative into my 8 GB derivative everything works like a charm...

diasdmhub, 2021-08-11:
> _Maybe the cooling levels in the *.dts file for level 1 and 2 are too low ?_

I thought of that as well. They seem too small to spin the fan. Although, they were spinning before the kernel update.

Anyhow, I replicated the first step you suggested
```
$ sudo dtc -I dtb -O dts -o /boot/overlays/rpi-poe-plus-overlay.dts /boot/overlays/rpi-poe-plus.dtbo
/boot/overlays/rpi-poe-plus-overlay.dts: Warning (unit_address_vs_reg): /fragment@0/__overlay__/rpi-poe-fan@0: node has a unit name, but no reg property
/boot/overlays/rpi-poe-plus-overlay.dts: Warning (unit_address_vs_reg): /fragment@3/__overlay__/rpi-poe-power-supply@0: node has a unit name, but no reg property
```

Backed up the original _.dtbo_
`# cp /boot/overlays/rpi-poe-plus.dtbo /boot/overlays/rpi-poe-plus.bkp.dtbo`

Set cooling levels like this (0=0, 1=30, 2=50, 3=100, 4=255):
`cooling-levels = < 0x00 0x1e 0x32 0x64 0xff >`

Then, I compiled the new _.dts_ file
```
# dtc -@ -Hepapr -I dts -O dtb -o /boot/overlays/rpi-poe-plus.dtbo /boot/overlays/rpi-poe-plus-overlay.dts
/boot/overlays/rpi-poe-plus.dtbo: Warning (unit_address_vs_reg): /fragment@0/__overlay__/rpi-poe-fan@0: node has a unit name, but no reg property
/boot/overlays/rpi-poe-plus.dtbo: Warning (unit_address_vs_reg): /fragment@3/__overlay__/rpi-poe-power-supply@0: node has a unit name, but no reg property
/boot/overlays/rpi-poe-plus.dtbo: Warning (cooling_device_property): /fragment@1/__overlay__/cooling-maps/map0:cooling-device: cell 0 is not a phandle reference
/boot/overlays/rpi-poe-plus.dtbo: Warning (cooling_device_property): /fragment@1/__overlay__/cooling-maps/map1:cooling-device: cell 0 is not a phandle reference
/boot/overlays/rpi-poe-plus.dtbo: Warning (cooling_device_property): /fragment@1/__overlay__/cooling-maps/map2:cooling-device: cell 0 is not a phandle reference
/boot/overlays/rpi-poe-plus.dtbo: Warning (cooling_device_property): /fragment@1/__overlay__/cooling-maps/map3:cooling-device: cell 0 is not a phandle reference
```

Despite the warnings, all went well and the new _.dtbo_ file has new cooling levels.
```
# fdtdump /boot/overlays/rpi-poe-plus.dtbo | grep levels
                cooling-levels = <0x00000000 0x0000001e 0x00000032 0x00000064 0x000000ff>;
```

After rebooting I can see that the fan is (finally!) spinning again at level 2 for both my RPi's.
```
$ cat /sys/class/thermal/cooling_device0/cur_state
2
```

And I can confirm that the CPU temperature is slowly dropping, as it should.

The best part is that the new values are very silent.
It would be interesting to see the actual fan RPM, so we could evaluate those cooling levels.
But, for now, I settled.

With this issue, seems to me that the original cooling levels are not well optimized for the Raspberry PoE+ HAT.
chri-kai-in, I confirm your theory that it works with the 8GB model. Unfortunately I cannot test with a 4GB RPi.

Thank you a lot.

popcornmix, 2021-08-16:
@XECDesign did you see this, and should we change the cooling levels?

pelwell, 2021-08-16:
It sounds like they are too low in some cases. If @XECDesign agrees I'll increase them to something safer.

chri-kai-in, 2021-08-16:
Maybe a different spreading of the PWM values would bei helpful. Either following the proposal of diasdmhub (0 30 50 100 255) or apply another distribution (0 32 64 128 255).

pelwell, 2021-08-18:
Having laid my hands on a POE+ HAT, along with a first-generation PoE HAT, I can confirm that the cooling values work with the older board but not the newer version.

The proposed values of 0 32 64 128 255 are appealing - level 2 is still very quiet at 64 but spinning that bit fast should make the cooling more effective.

There's a new Pull Request at https://github.com/raspberrypi/linux/pull/4536, if anyone else would like to test it.

popcornmix, 2021-08-19:
Improvement should be in latest rpi-update kernel.