raspberrypi/firmware #1609: HDMI broken after Update from Videocore May 27 2021 14:02:15 to Aug  3 2021 18:14:56

schwebke, 2021-08-08:
**Describe the bug**
After updating Pi OS, HDMI is no longer working. I am using a Samsung LS34J55 ultrawide with 3440x1440@50Hz and a Pi4 8GB. This setup was working fine for some months.

Checking with a second Pi4 4GB shows the same behaviour. Reverting the update (sd card image backup from July) leads to a working setup again.

**To reproduce**
I was able to narrow down the error to these package updates:
```
iu   libraspberrypi-bin                      1:1.20210527-1      1:1.20210805-1
iu   libraspberrypi-dev                      1:1.20210527-1      1:1.20210805-1
iu   libraspberrypi-doc                      1:1.20210527-1      1:1.20210805-1
iu   libraspberrypi0                         1:1.20210527-1      1:1.20210805-1
iu   raspberrypi-bootloader        +57.3 kB  1:1.20210527-1      1:1.20210805-1
```

**Expected behaviour**
System booting up to graphical login (display manager), normal and stable working possible.

**Actual behaviour**
The system boots normally, showing the blinking text cursor and for some seconds the display manager login screen is showing up correctly. Sometimes even login and working for some seconds is possible.

Then the screen turns black, showing "no signal" and entering a loop switching between a black image and "no signal".

Comparing `raspinfo` outputs before and after the update I see `hdmi: HDMI1:EDID error reading EDID block 0 attempt ...` messages repeatetly in vcdbg log after the update, which where not there in the previous, working version.

**System**
`raspinfo` output from working state before update and broken state after updating:
* [raspinfo-working.txt](https://github.com/raspberrypi/firmware/files/6950253/raspinfo-working.txt)
* [raspinfo-broken.txt](https://github.com/raspberrypi/firmware/files/6950254/raspinfo-broken.txt)

JeffInCO, 2021-08-17:
I have the exact same issue, but with a Dell U3415W which is the same native resolution (3440x1440) as the original report above.

The gui works fine with an SD card freshly formatted with the May 7th, 2021 Raspberry Pi OS image.

With the fresh image and nothing else installed or changed, if I do an update then full-upgrade, I get the behavior described above a few seconds after the gui starts.  I.e., alternating between "no signal" and a black screen.

This is with a Raspberry Pi4, 8GB, which is the only Raspberry Pi4 that I currently own.  I am also plugged into the HDMI port closest to the USB-C power connector, same as the original poster.

alanbork, 2021-09-19:
have you tried this with kms and fkms both?

LordPossum, 2021-11-09:
I have the same problem described here with my Pi4 4GB and a slightly different setup, consisting in a 1920x1080 Philips  227E7 monitor connected to the Pi with a HDMI-to-DVI cable. Tinkering a bit made me aware of two indipendents "workarounds":
-using FKMS in /boot/config.txt
-connect the Pi to the monitor using the HDMI port of the monitor. Swapping out the cable after booting brings always to no desktop, though.
I tried adding hdmi_force_hotplug=1 to /boot/config.txt and nothing changed (and I wasn't using it before when all was working fine)

popcornmix, 2021-11-09:
@LordPossum are you saying it works correctly with firmware from before May 27 2021, and fails with firmware after that point?

LordPossum, 2021-11-09:
@popcornmix Yes. Up untill the bump to kernel 5.10.39 of May 25 2021 included all was working fine, from 5.10.42 I got black screen.

popcornmix, 2021-11-09:
@LordPossum can you just confirm:
`sudo rpi-update bb46e15`
`sudo rpi-update 6263bdf`
The former (5.10.39) and the latter (5.10.42) fails? That seems surprising based on the changes in the update.

You've described using fkms as a workaround. Are you saying fkms works, and kms fails? (or no fkms/kms fails?)

LordPossum, 2021-11-09:
Ok, just messed up some stuff because I upgraded the Pi's eeprom, sorry to waste a little more of your time.
I can confirm `bb46e15` works and `6263bdf` doesn't with pieeprom-2020-09-03.
I then upgraded pieeprom to the version `1619712685` dated April 29 2921, and now the problem exists between firmware `894ca75`, which is not working, and `fb7c402`, which boots fine to desktop.

As for your last question, if I understood the last part correctly: using FKMS always work, booting to desktop, while using KMS fails after `894ca75` included.

popcornmix, 2021-11-09:
It seems strange that a pieeprom update would move where the failure started.
Are you confident fb7c402 boots fine multiple times, and 895ca75 fails every time? (rather than the failure being intermittent).

https://github.com/raspberrypi/linux/pull/4371 does note that the change will stop a display that doesn't report hotplug (either a broken hdmi cable or a broken display) from working.

`hdmi_force_hotplug=1` doesn't work on kms, but you should be able to get the same effect on kms with `video=HDMI-A-1:D` in cmdline.txt.

schwebke, 2021-12-26:
@alanbork 
> have you tried this with kms and fkms both?
No, up to now just with default fkms on raspbian buster.

With new default kms on raspbian bullseye my UW display is just working without any config or tweaks.

closed 2021-12-26 by schwebke