raspberrypi/firmware #1610: POE+ hat fan not working on RPI 4B, 4GB derivative

chri-kai-in, 2021-08-11:
With proven dtoverlay configuration and latest kernel and firmware POE+ hat fan does not work on RPI 4B, 4GB variant.

Description:
Fan does not show movement on RPI 4B, 4GB variant with kernel version 5.0.52-vr71+ and firmware/bootloader as of 06.07.2021.
Fan was as well is not moving with older releases of firmware and kernel on this device.

To reproduce:
 Update kernel to latest release:
``` sudo rpi-update ```

Update the firmware to the latest release:
``` sudo rpi-eeprom-update ```

Add dtoverlay configuration to /boot/config.txt
```
#POE+ Hat Fan Control
dtoverlay=rpi-poe-plus
dtparam=poe_fan_temp0=60000
dtparam=poe_fan_temp0_hyst=2000
dtparam=poe_fan_temp1=65000
dtparam=poe_fan_temp1_hyst=2000
dtparam=poe_fan_temp2=70000
dtparam=poe_fan_temp2_hyst=2000
dtparam=poe_fan_temp3=75000
dtparam=poe_fan_temp3_hyst=5000
``` 
Change dtoverlay file to provide higher PWM values for colling level 1 & 2 -  see #1607 on how this was achieved.

Restart RPI with
``` sudo shutdown -r now```

Expected behaviour:
FAN shall start moving, when cooling level 2 is reached. Fan speed shall be increased with every next cooling level required.

Actual behaviour:
Fan does not move, even if core temperature is above 80°C and system is throttled to limit overal temperature.

System:
``` 
pi@octopi:~ $ sudo uname -r
5.10.52-v7l+
```

``` 
pi@octopi:~ $ sudo rpi-eeprom-update
BOOTLOADER: up to date
   CURRENT: Di 6. Jul 10:44:53 UTC 2021 (1625568293)
    LATEST: Di 6. Jul 10:44:53 UTC 2021 (1625568293)
   RELEASE: stable (/lib/firmware/raspberrypi/bootloader/stable)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1
pi@octopi:~ $ vcgencmd version
Aug  5 2021 13:48:53
Copyright (c) 2012 Broadcom
version fc9eef11715b244b59c2da056d5aa540fc31255f (clean) (release) (start_x)
```
```
pi@octopi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2020-12-02
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, cce27bd6f44a3b2e83855645986b3e21f771e852, stage2
```

Logs:
``` dmesg | grep fan``` showed no results

Additional context:
- Same configuration and system (µSD-Card swapped) works perfect in RPI 4B, 8GB derivative.
- POE+ hat and fan works if placed on RPI 4B, 8GB system
- Different working system from RPI 4B, 8GB shows same behavior if swapped to RPI 4B, 4GB device.

pelwell, 2021-08-11:
How many instances of a 4GB Pi4B have you tested?

We could rule out a few possibilities by running both systems (4GB + 8GB) with `total_mem=3072` (3GB) in config.txt.

Also, `tail /proc/cpuinfo` on both systems would be useful.

chri-kai-in, 2021-08-11:
Until now - the one instance which is available to me. It can of course be an issue of this instance, which i would favor more than any other result - as long there is an option to fix it ...

```
pi@8GBRPI4B:~ $ tail /proc/cpuinfo
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2711
Revision        : d03114
Serial          : 10000000001c03f5
Model           : Raspberry Pi 4 Model B Rev 1.4

```
pi@$GBRPI4B:~ $ tail /proc/cpuinfo
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2711
Revision        : c03114
Serial          : 10000000945d3092
Model           : Raspberry Pi 4 Model B Rev 1.4

````

I will run both systems with the RAM limitation option in /boot/config.txt. What information do you need when doing so ?

pelwell, 2021-08-11:
Only if the fan works or not, thanks.

chri-kai-in, 2021-08-11:
OK - no fan activity on the 4GB derivative and fan activity on the 8GB derivative.

dmesg states a maximum of 3072 MB RAM:

```
4GB:
[    0.000000] Memory: 2573948K/3014656K available (10240K kernel code, 1364K rwdata, 3188K rodata, 2048K init, 891K bss, 113028K reserved, 327680K cma-reserved, 2228224K highmem)
```
```
8GB:
[    0.000000] Memory: 2626728K/3067904K available (10240K kernel code, 1364K rw           data, 3188K rodata, 2048K init, 891K bss, 113496K reserved, 327680K cma-reserved           , 2281472K highmem)
```

chri-kai-in, 2021-08-11:
I don`t know if this would be of relevance, bt the fan starts moving for a short period of time when attaching the powered network cable. Could a local, specific fail on the GPIOs influence the fan - maybe a PWM output,SPI or I2C interface ?

If i could check something on my machine that would exclude these HW related constraints, i would gladly support.

diasdmhub, 2021-08-11:
I'm not sure it is related to the issue, but I've noticed that my RPi's have different hardware than your, @chri-kai-in.
Both are BMC2835
```
$ tail -4 /proc/cpuinfo
Hardware        : BCM2835
Revision        : d03114
Serial          : 100000007568d97d
Model           : Raspberry Pi 4 Model B Rev 1.4
```

chri-kai-in, 2021-08-14:
I now have a 2nd RPI 4, 4GB and with the same POE+ HAT my system and my configuration works out of the box and the POE+ fan spins.

```
pi@octopi:~ $ tail /proc/cpuinfo
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2711
Revision        : c03114
Serial          : 10000000e85fe232
Model           : Raspberry Pi 4 Model B Rev 1.4
```
**With that result, i want to downgrade this issue to a local problem with my original board.**

How can i investigate HW fails on my initial 4GB board that might influence the fan activity on the POE+ HAT ?

I mapped the i2c-0 bus temporarily to ID=9 ( following the hints at [https://github.com/raspberrypi/hats/tree/master/eepromutils](url))

```
sudo dtoverlay i2c-gpio i2c_gpio_sda=0 i2c_gpio_scl=1 bus=9
```
If i access the bus with i2cdetect, i get the following response, so i would say the i2c-bus 0 on this board is the cause for this defect. On those boards where everything runs smooth i get a sharp response "51" at address 51, on all other adresses there is a "--"

```
pi@octopi:~ $ i2cdetect -y 9
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
10: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f
20: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f
30: 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f
40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f
50: 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f
60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f
70: 70 71 72 73 74 75 76 77
```
Any other idea what to check, probe or configure to get the HAT bus back alive ? All inputs appreciated, before i close this issue.

abashammakh, 2021-08-15:
I am having the same issue as you (fan not working at levels 1 and 2).

I tried to change PWM values as you mentioned in #1607. And It worked! after that, I reverted to the original file and the fan stopped working again.

I also tried to access i2c bus, as you showed above (with the original .dtbo file). and I am getting the expected value of 51 at address 51. 

I don't know the cause of this weird behavior.

here is cpuinfo if it might help:

```
pi@raspberrypi-1:~ $ tail /proc/cpuinfo
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2711
Revision        : c03114
Serial          : 10000000bb4e6797
Model           : Raspberry Pi 4 Model B Rev 1.4
```

I think I will wait for an 'official' fix, rather than changing the PWM values.

Thanks.

closed 2021-08-20 by chri-kai-in