raspberrypi/firmware #1611: [Raspberry Pi 3+] Disabling WiFi breaks Bluetooth in some cases

MichaIng, 2021-08-14:
**Describe the bug**
On some Raspberry Pi 3+ devices, when onboard WiFi is disabled via `dtoverlay=disable-wifi`, Bluetooth also stops working. According to some forum threads this seems to be the case when WiFi and Bluetooth share the same reset line:
- https://www.raspberrypi.org/forums/viewtopic.php?t=276582
- https://www.raspberrypi.org/forums/viewtopic.php?t=284221

However, there are Raspberry Pi 3+ devices without this issue: https://github.com/MichaIng/DietPi/issues/4512#issuecomment-878613153
This is unexpected since there is only a single revision `a020d3`. This makes it difficult to know for installers/scripts whether the WiFi disable overlay needs to be removed as well to enable Bluetooth or not.

**To reproduce**
```
sed -i '/dtoverlay=disable-/d' /boot/config.txt
echo 'dtoverlay=disable-wifi' > /boot/config.txt
reboot
apt install pi-bluetooth
journalctl -u hciuart
```

**Expected behaviour**
Disabling WiFi does not disable Bluetooth as well.

**Actual behaviour**
Disabling WiFi disables Bluetooth as well.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? `Pi3B+`
* Which OS and version (`cat /etc/rpi-issue`)? `Raspberrry Pi OS Buster`
* Which firmware version (`vcgencmd version`)? `#1421`
* Which kernel version (`uname -a`)? `5.10.17-v7+ #1421 SMP Thu May 27 13:59:01 BST 2021 armv7l `

**Logs**
```
systemd[1]: Starting Configure Bluetooth Modems connected by UART...
btuart[209]: Initialization timed out.
btuart[209]: bcm43xx_init
systemd[1]: hciuart.service: Control process exited, code=exited, status=1/FAILURE
systemd[1]: hciuart.service: Failed with result 'exit-code'.
systemd[1]: Failed to start Configure Bluetooth Modems connected by UART.
```

**Additional context**
It this is a hardware-wise limitation, not sure if there is a way to have `dtoverlay=disable-wifi` ignored automatically, as long as Bluetooth is not disabled as well on affected devices? On the other hand I'm not sure whether it's more important to have Bluetooth enabled, when it's expected to be, or to have WiFi disabled, when it's set explicitly via overlay. But probably more important for scripting is whether there is a way to find out in the first place, if the RPi model/system is affected by the issue or not, as the revision code is not sufficient.

pelwell, 2021-08-16:
After some discussion in the office the view is that it is possible that the version number wasn't bumped for the 3B+ change, which is a shame. The rev 1.3 3B+ schematic I have still shows the enable signals as two distinct traces, while the 3A+ clearly shows that WL_ON is also wired to the BT_ON of the modem for layout reasons.

Without a revision change it's going to be difficult to distinguish them. Apart from giving the user a way to override the default for the 1.3 3B+ based on experience, the best bet would probably be to probe for the BT modem with the usual BT_ON (GPIO 0 on the expander) low.

You can use a test script something like this:
```
#!/usr/bin/perl
my $fd;
system("sudo stty -F /dev/serial1 115200 raw time 2 min 0");
die "* failed to open UART\n" if (!open($fd, '+<', "/dev/serial1"));
$fd->autoflush(1);
print $fd ("\x01\x03\x0c\x00");
my $len = sysread($fd, $buf, 10);
printf("BT: %s (%d)\n", $len ? "on" : "off", $len);
close($fd);
```

This test would be run during installation or before Bluetooth starts up, but for testing purposes disable the HCI startup script (on RPiOS, `sudo systemctl disable hciuart`).

With the BT_ON line to the modem enabled, the script will provoke it to return a short sequence (usually 7 bytes). When disabled, there is no response and the read returns a length of 0. A manual test might look something like this:
```
pi@raspberrypi:~$ gpioset gpiochip1 0=0
pi@raspberrypi:~$ ./btprobe
BT: off (0)
pi@raspberrypi:~$ gpioset gpiochip1 0=1
pi@raspberrypi:~$ ./btprobe
BT: on (7)
```
On a 3B+ with BT_ON slaved to WL_ON both btprobes would succeed.

For a production script you would probably replace the printf with a return code indicating whether or not the modem responded.

MichaIng, 2021-08-16:
Many thanks for your detailed answer. Good to know that it is a known behaviour on 3A+.

If I understand it right, this means that `dtoverlay=disable-bt` has not effect on the affected models, but `dtoverlay=disable-wifi` disables both, WiFi+BT, and in absence, both are enabled?

Related to the above, does it play a role whether WiFi/BT is enabled or disabled via device tree overlay for doing the probe, or is the state basically overridden by the GPIO state?

pelwell, 2021-08-16:
> If I understand it right, this means that dtoverlay=disable-bt has not effect on the affected models, but dtoverlay=disable-wifi disables both, WiFi+BT, and in absence, both are enabled?

The `disable-bt` should have exactly the same effect on both 3B+ variants in that it releases ttyAMA0 for console (or other) use, disables UART1, doesn't enable any UART onto GPIOs 30-33, configures the serial<n> aliases appropriately and leaves enough clues to effectively disable the hciuart service. But in the sense that it also causes the firmware to set GPIO0 on the expander to 0 then yes - it has no effect on devices with the shared enables.

`disable-wifi` disables the SDIO interface, meaning that the WiFi half of the 43455 should not be accessed. However, it also causes the firmware to pull WL_ON (and hence BT_ON in these cases) low. That's the result of a power-saving patch introduced late in 2019 (it can save 450mW if both WiFi and BT and disabled), but as you've discovered it has undesirable effects on a subset of the 1.3 3B+s.

If all you are wanting to do is disable WiFi then you shouldn't have to deal with this corner case - the firmware should leave WL_ON alone. Given that the big power savings are to be had with both disabled, I think it's reasonable in this case to disable the WL_ON lowering unless the BT is also disabled. This logic is probably going to have to apply to all 1.3 3B+s.

MichaIng, 2021-08-16:
Okay, thanks for clarification. So `disable-bt` must not be set for the probe to work, and when changing `GPIO 0` state has an effect on the BT probe, `disable-wifi` breaks BT, if `GPIO 0` has no effect on the probe result, `disable-wifi` needs to be absent/off for BT to function.

It is practically not that trivial to apply for a config script, as, if `disable-bt` is currently set, the Pi needs to be rebooted first, before the probing can be done.

The alternative is indeed to simply remove `disable-wifi` on all RPi 3+ models when Bluetooth is enabled and skip the probe. We also blacklist the Broadcom WiFi modules, so WiFi will be definitely disabled, no interface present, either way and the additional benefit of  `disable-wifi` is, as you stated, negligible.

Many thanks for taking the time to dig into this and I think we have collected all information required to deal with it properly. Not a big issue after all, when known. Is it worth a note in the [README](https://github.com/raspberrypi/firmware/blob/5bcc8bc/boot/overlays/README#L696-L699)?

MichaIng, 2021-08-16:
Ah, got it, that is great 👍.

pelwell, 2021-08-16:
There's a trial firmware (just start.elf and fixup.dat) that should fix this issue: https://drive.google.com/file/d/10gBNYX6UcFFwjZircjbfgOd7uKzBSnzA/view?usp=sharing

popcornmix, 2021-08-19:
Fix should be in latest rpi-update firmware.

MichaIng, 2021-08-19:
Many thanks, I'm still waiting for reply/test of our user. I'll also inform users in the forum threads.

popcornmix, 2021-08-19:
I've posted to those threads (directing here) in case any users are following.

miodzicho, 2025-03-17:
```
Revision        : a020d4
Serial          : 00000000e549ca59
Model           : Raspberry Pi 3 Model B Plus Rev 1.4
```

`Linux GurkenRick3 6.12.19-v8+ #1863 SMP PREEMPT Thu Mar 13 14:21:42 GMT 2025 aarch64 GNU/Linux`

```
Mar 16 2025 21:10:06
Copyright (c) 2012 Broadcom
version e5748958ed4a2d0290edf0011e8c2825fb07df10 (clean) (release) (start_cd)
```

```
dtparam=i2c_arm=on
dtparam=audio=off
camera_auto_detect=0
display_auto_detect=1
auto_initramfs=1
dtoverlay=vc4-kms-v3d
max_framebuffers=2
disable_fw_kms_setup=1
arm_64bit=1
disable_overscan=1
arm_boost=1
otg_mode=1
[cm5]
dtoverlay=dwc2,dr_mode=host
[all]
dtparam=i2c1=on
dtparam=i2c_arm=on
dtparam=spi=on
gpu_mem=1
dtoverlay=dwc2
enable_uart=1
[pi0]
dtoverlay=spi0-2cs
#dtoverlay=disable-wifi
[pi3]
dtoverlay=spi0-2cs
dtoverlay=pi3-disable-wifi (tried also disable-wifi)
[pi4]
dtoverlay=spi0-2cs
#dtoverlay=disable-wifi
[pi5]
dtoverlay=spi0-2cs
#dtoverlay=disable-wifi

```

pelwell, 2025-03-17:
Try this: https://drive.google.com/file/d/1MKE2BDLGE9rW1vX3OzIIgi0tLydGqA3y/view?usp=sharing

miodzicho, 2025-03-18:
I have copied start.elf and fixup.dat to /boot/firmware. Here is dmesg : 
```
[    9.133023] Bluetooth: Core ver 2.22
[    9.133150] Bluetooth: HCI device and connection manager initialized
[    9.133171] Bluetooth: HCI socket layer initialized
[    9.133182] Bluetooth: L2CAP socket layer initialized
[    9.133205] Bluetooth: SCO socket layer initialized
[    9.178405] Bluetooth: HCI UART driver ver 2.3
[    9.178436] Bluetooth: HCI UART protocol H4 registered
[    9.178554] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    9.179772] Bluetooth: HCI UART protocol Broadcom registered
[   11.453557] Bluetooth: hci0: command 0xfc18 tx timeout
[   11.453607] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[   11.453619] Bluetooth: hci0: Failed to set baudrate
[   12.667531] Bluetooth: hci1: urb 000000008b33648a failed to resubmit (2)
[   12.667604] Bluetooth: hci1: Opcode 0x0c03 failed: -2
[   12.667617] Bluetooth: hci1: urb 00000000e254ff3f failed to resubmit (2)
[   12.667633] Bluetooth: hci1: urb 00000000aa086f45 failed to resubmit (2)
[   13.473574] Bluetooth: hci0: command 0xfc18 tx timeout
[   13.473638] Bluetooth: hci0: BCM: Reset failed (-110)
[   14.239654] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   14.239682] Bluetooth: BNEP filters: protocol multicast
[   14.239702] Bluetooth: BNEP socket layer initialized
[   14.845635] Bluetooth: hci1: Opcode 0x0c03 failed: -110
```
And bluetoothctl : 
```
pi@GurkenRick3:~ $ sudo bluetoothctl
Agent registered
[bluetooth]# scan on
No default controller available
[bluetooth]#
```
```
pi@GurkenRick3:~ $ systemctl status bluetooth
● bluetooth.service - Bluetooth service
     Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; preset: enabled)
     Active: active (running) since Tue 2025-03-18 08:47:46 GMT; 15min ago
       Docs: man:bluetoothd(8)
   Main PID: 450 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 821)
        CPU: 199ms
     CGroup: /system.slice/bluetooth.service
             └─450 /usr/libexec/bluetooth/bluetoothd --noplugin=sap,a2dp

Mar 18 08:47:46 GurkenRick3 bluetoothd[450]: Starting SDP server
Mar 18 08:47:46 GurkenRick3 bluetoothd[450]: Excluding (cli) sap
Mar 18 08:47:46 GurkenRick3 bluetoothd[450]: Excluding (cli) a2dp
Mar 18 08:47:46 GurkenRick3 bluetoothd[450]: profiles/audio/vcp.c:vcp_init() D-Bus experimental not enabled
Mar 18 08:47:46 GurkenRick3 bluetoothd[450]: src/plugin.c:plugin_init() Failed to init vcp plugin
Mar 18 08:47:46 GurkenRick3 bluetoothd[450]: profiles/audio/mcp.c:mcp_init() D-Bus experimental not enabled
Mar 18 08:47:46 GurkenRick3 bluetoothd[450]: src/plugin.c:plugin_init() Failed to init mcp plugin
Mar 18 08:47:46 GurkenRick3 bluetoothd[450]: profiles/audio/bap.c:bap_init() D-Bus experimental not enabled
Mar 18 08:47:46 GurkenRick3 bluetoothd[450]: src/plugin.c:plugin_init() Failed to init bap plugin
Mar 18 08:47:46 GurkenRick3 bluetoothd[450]: Bluetooth management interface 1.23 initialized
pi@GurkenRick3:~ $
```

pelwell, 2025-03-18:
The trial firmware solves the problem for me on a rev 1.4 3B+, so there must be something different about your environment.

1. Can you confirm that the updated firmware is installed correctly with `vcgencmd version`?
2. Do you also have a USB Bluetooth dongle present? Can you retest without it? In fact, remove any external hardware which isn't absolutely necessary.

pelwell, 2025-03-18:
I wouldn't be surprised if you are actually experiencing an undervoltage situation. Disabling the onboard WiFi should save some power, but it will also change the timing which might exacerbate the problem.

Is your power supply and power cable adequate? The official micro USB supply provides 2.5A at 5.1V, but even with that a 3B+ can struggle if there is a demanding USB device attached.

miodzicho, 2025-03-18:
I'm using PiSugar3 to power RPI. However didn't noticed anything related to undervoltage, stability issues or reboots.

pelwell, 2025-03-19:
It may be fine most of the time, but the kernel and systemd try to turn on everything as early as possible to reduce boot time, and that can really stress the supply.