raspberrypi/firmware #1613: Raspistill hangs causing RPi to freeze

CuriousDran, 2021-08-16:
Hello, I have been working on an optical sensor system powered by RPi 4B (8GB) and HQ camera (IMX477). I've been struggling to resolve this issue since April! In its current configuration, I have 4 HQ cameras linked into the RPi using an Arducam Multicamera Adapter https://www.amazon.com/Arducam-Camera-Adapter-Raspberry-Cameras/dp/B07TYL36MC. I have trouble-shot the system with and without this adapter, with the same results, so I do not think it is the source of the problem.

I have already tried the solutions posted in those threads with no luck.

**Describe the bug**
The system is intended to be used continuously for at least a couple weeks at a time. Currently the system freezes after a few hours. It is my understanding that the Raspistill function hangs when the bug occurs. This has proved very difficult to track down because the problem tends to reproduce itself at inconsistent intervals - the longest I've managed to run it for is 12hours before it freezes. 
 
**To reproduce**
Continuous, image capture using rpi 4B and HQ camera. 

**Expected behaviour**
I expect the system to be able to operate continuously for at least a couple weeks at a time. Best I can get is half a day

**Actual behaviour**
After a varied amount of time raspistill hangs during close (based on the --verbose output option) and the raspberry pi freezes. the only way to regain operation is to do a hard reset (power cycle) on the system.

**System**
pi@raspberrypi:~ $ raspinfo
System Information
------------------

Raspberry Pi 4 Model B Rev 1.2
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2021-03-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 461109415073d2eb67083709662ba983cc191f14, stage4

Linux raspberrypi 5.10.17-v7l+ #1421 SMP Thu May 27 14:00:13 BST 2021 armv7l GNU/Linux
Revision	: c03112
Serial		: 1000000076733945
Model		: Raspberry Pi 4 Model B Rev 1.2
Throttled flag  : throttled=0x0
Camera          : supported=1 detected=1

Videocore information
---------------------

May 27 2021 14:02:45 
Copyright (c) 2012 Broadcom
version 7d9a298cda813f747b51fe17e1e417e7bf5ca94d (clean) (release) (start_db)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       59798140 4867856  52448208   9% /
devtmpfs         1801156       0   1801156   0% /dev
tmpfs            1933252       0   1933252   0% /dev/shm
tmpfs            1933252   16952   1916300   1% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs            1933252       0   1933252   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   48790    209305  19% /boot
tmpfs             386648       4    386644   1% /run/user/1000

Filename				Type		Size		Used	Priority
/var/swap                               file		102396		0	-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: 1.20201210+nmu1
raspberrypi-sys-mods:
  Installed: 20210310
openbox:
[... 567 lines cut ...]
[   11.842700] Bluetooth: RFCOMM ver 1.11

pi@raspberrypi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2021-03-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 461109415073d2eb67083709662ba983cc191f14, stage4

pi@raspberrypi:~ $ sudo vcgencmd version
Mar 26 2021 16:33:56 
Copyright (c) 2012 Broadcom
version edf2e9c318863999c97c50cdb74eee235ede3af5 (clean) (release) (start_db)

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.17-v7l+ #1421 SMP Thu May 27 14:00:13 BST 2021 armv7l GNU/Linux

**Logs**
Link to my responses in another GitHub thread, happy to provide more specific info upon request.
https://github.com/raspberrypi/firmware/issues/1552#issuecomment-873276822

**Additional context**
 This is not my first post on the matter, and I have found other posts that seem to describe my issue well (see two links below). 

- https://www.raspberrypi.org/forums/viewtopic.php?t=304989&sid=8a87bd5519da361639ab16fd238cade3
- https://github.com/raspberrypi/firmware/issues/1552

I am at my whits end with this thing, I've tested it to the brink of insanity. Please help! I need to get this thing completed ASAP.

hastou, 2021-09-28:
@CuriousDran This bug exists since the rpi 4 release and was never solved. Here are all the issues I found this last 2 years describing the problem :
[raspberrypi/userland/issues/686](https://github.com/raspberrypi/userland/issues/686)
[raspberrypi/userland/issues/687](https://github.com/raspberrypi/userland/issues/687)
[raspberrypi/userland/issues/564](https://github.com/raspberrypi/userland/issues/564)
[raspberrypi/firmware/issues/1574](https://github.com/raspberrypi/firmware/issues/1574)
[raspberrypi/firmware/issues/1298](https://github.com/raspberrypi/firmware/issues/1298)
[raspberrypi/linux/issues/4047](https://github.com/raspberrypi/linux/issues/4047)
[waveform80/picamera/issues/574](https://github.com/waveform80/picamera/issues/574)
[raspberrypi/userland/issues/599](https://github.com/raspberrypi/userland/issues/599)

CuriousDran, 2021-09-28:
@hastou thank you, I'll take a look at those links to see if anything there can help.

@JamesH65 The number of captures is inconsistent. Sometime it lasts for three hours, other time it crashes after 5 mins. In each case it requires a hard reset to bring the system back. 

The command I am using is:

raspistill -m 3 -o %s -ss %d -ag %c

CuriousDran, 2021-09-28:
@hastou thanks for the links, I'll check them out.

@JamesH65 The number of captures is inconsistent. Sometimes crashing after three hours, other times crashing in 5 mins. Changing the capture speed hasn't solved the problem. Currently I am taking two images every 30 seconds.

The prototype is currently with the client, I'll need to double check the exact command line arguments being used to capture each image. I'll respond with that info once I have it.

It is worth noting that I did find one of the administrators mentioned that this could be primarily a hardware issue? Is it possible I mishandled the camera? The thing is, I was initially using 4 cameras all connected through an Arducam Multicamera Adapter Board (link: https://www.amazon.com/Arducam-Camera-Adapter-Raspberry-Cameras/dp/B07TYL36MC). Though the problem persisted even when I switched to just one HQ camera. I have tried each camera individually and found the problem stays each time, so unless I managed to damage all four cameras, I don't think it was caused by something I did.

The system will be used in a laboratory environment (indoors) so I don't believe the environment is causing the problems, but I could be mistaken.

I have been fighting this thing for almost 6 months. I am a Graduate student in ECE and this little glitch has set this project back a while. The current solution was to download Ubuntu with a 3rd party GUI. Now the captures happen indefinitely without fail (problem solved). Unfortunately this solution came with its own issues (outside the scope of this discussion). 

I'll post another reply with the command line information you were asking for once I have the prototype system back...

Thank you very much for your replies! I had honestly lost hope of finding a working solution to this.

JamesH65, 2021-09-29:
We will need dmesg, vcdbg logs etc to even have a chance of thinking about this. See https://www.raspberrypi.org/documentation/computers/os.html#vcdbg

CuriousDran, 2021-09-29:
I believe those outputs are in the original post (vcdbg log messages, dmesg log, etc.) Did I miss something? I'll need to reinstall the Raspian OS microSD to provide more readouts for you. I can have that info later today, please just let me know specifically what more information you require.

I got the prototype back from the client yesterday afternoon. Over the past 6 months I've tried a bunch of different command line options for raspistill. Here is the currently used call:

`cmd = "raspistill -e jpg -md 3 -ex verylong -awb off -awbg 3.0,3.0 -ss %d -ag %1.1f -dg %1.1f -n -o %s -h 460 -w560" % (shutterSpeed, analogGain, digitalGain, imgFilename)`
`errchk = subprocess.call([cmd],shell=True)`

The code is set up to receive argv[] command line options during call to provide user control over filename, shutter speed, analog gain, and digital gain.

soundfx37, 2022-04-29:
I'm having a similar problem.  At some point (no discernible pattern of when) the mmal will fail to close/destroyed and much of the system hangs.  I can still connect to the pi over ssh, but it seems like all the IO is stuck.  Mouse, keyboard, SPI, and GPIO buttons all non-responsive.

Every time I have a crash it happens somewhere after "Closing down" and before "Close down completed..."

     `if (state.verbose)
         fprintf(stderr, "Closing down\n");

      if (output_file)
         fclose(output_file);
    if (state.preview_parameters.wantPreview && state.preview_connection)
         mmal_connection_destroy(state.preview_connection);
      if (state.encoder_connection)
         mmal_connection_destroy(state.encoder_connection);
      if (state.splitter_connection)
         mmal_connection_destroy(state.splitter_connection);
      // Disable all our ports that are not handled by connections
      check_disable_port(encoder_output_port);
      check_disable_port(splitter_preview_port);
      
      // Disable components 
      if (state.encoder_component)
         mmal_component_disable(state.encoder_component);
	  
     if (state.veye_camera_isp_state.isp_component)
         mmal_component_disable(state.veye_camera_isp_state.isp_component);
      if (state.veye_camera_isp_state.camera_component)
         mmal_component_disable(state.veye_camera_isp_state.camera_component);

      raspipreview_destroy(&state.preview_parameters);
      destroy_splitter_component(&state);
      destroy_encoder_component(&state);
      destroy_veye_camera_isp_component(&state.veye_camera_isp_state);
	
      if (state.verbose)
         fprintf(stderr, "Close down completed, all components disconnected, disabled and destroyed\n\n");`