raspberrypi/firmware #1614: HDMI hotplug using tvservice stopped working on Pi 3b+

cyph84, 2021-08-19:
**Describe the bug**
After upgrading from an old firmware (Nov 23 2018 19:36:38 version d8054e0f49c3f96936956a4079b2290b11a5d428 (clean) (release)) I found the ability to switch from composite output to HDMI output using tvservice -p no longer works.

It seems like once the firmware has booted into composite mode, it's not possible to switch to HDMI or to get the firmware to even recognize that HDMI is now attached. This is tested using just console framebuffer without any desktop environment running.

**To reproduce**

1. Boot with HDMI unplugged
2. tvservice -s  state == 0x40000 (VC_SDTV_NTSC)
3. Plug in HDMI cable
4. tvservice -M responds with `[I] HDMI is attached`
5. But tvservice -s  state is still 0x40000

**Expected behaviour**
On older firmware (Nov 23 2018 19:36:38 version d8054e0f49c3f96936956a4079b2290b11a5d428 (clean) (release)):

1. Boot with HDMI unplugged
2. tvservice -s  state == 0x40001 (VC_HDMI_UNPLUGGED VC_SDTV_NTSC)
3. Plug in HDMI cable
4. tvservice -M responds with `[I] HDMI is attached`
5. vcdbg log msg shows firmware reading EDID `269418.784: hdmi: HDMI:>>>>>>>>>>>>>Rx sensed, reading EDID<<<<<<<<<<<<<`
6.  tvservice -s  state == 0x40002 (VC_HDMI_ATTACHED VC_SDTV_NTSC)
7. tvservice -p will configure and activate HDMI

**System**

System is a Pi3B+
Kernel: 5.10.52-7-ARCH
Firmware:
Aug 11 2021 14:00:57 
Copyright (c) 2012 Broadcom
version 5c8758dbac5b27911d2384c5dcd5db843528b606 (clean) (release) (start)

popcornmix, 2021-08-19:
@JamesH65 any thoughts? "Fix composite mode on devices up to Pi3B+"  was one of yours.

cyph84, 2022-05-24:
This is all tested on kernel framebuffer console. So I don't think I can restart it without rebooting. Also ran the same tests on X with xrandr.

Bottom line is: it seems that once you boot in RCA mode, you can't switch to HDMI. Once upon a time I remember you could do that with the firmware.