raspberrypi/firmware #1616: Kernel header package is missing

eriktews, 2021-08-24:
At least when using the 64 bit beta of Raspbian, there is currently no package that provides the kernel headers for the default kernel. Currently, the default kernel is 1:1.20210805-1 (5.10.17-v8+) but the only kernel header packages (linux-headers-....) are for 4.19.194-3. It would be great when the kernel headers would be provided here as well so that one can easily build additional modules for the kernel without having to install these headers manually or having to reinstall the entire kernel.

popcornmix, 2021-08-24:
This is in't the right place for packaging issues.
Try here: https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/

closed 2021-08-24 by popcornmix

XECDesign, 2021-08-31:
To answer the question anyway, the headers package is `raspberrypi-kernel-headers`.