raspberrypi/firmware #1617: repeat RPI_MBOX_NOTIFY_XHCI_RESET calls, result in XHCI failures.

jlinton, 2021-08-24:
It appears that the XHCI reset function has problems if it is called multiple times, particularly if the pcie translation is changed without the system being hard reset.

The UEFI/EDK2 firmware is blanket configuring the pcie root port without a DMA translation on all bcm2711 SoCs and enforcing a 3G DMA limit in ACPI mode (and mixed ACPI+DT). This works fine except when recent Linux kernels are booted with the official rpi foundation DT which is suggesting an inbound DMA translation on the more recent devices. In that case the kernel reprograms the pcie root port per the DT, and issues the mailbox call, but the XHCI cannot seem to recover.

Weirdly enough, if the uefi firmware is patched to dynamically modify the DT's to match its programmed translation the controller will work in DT mode reliably until the kernel issues the XHCI mailbox reset. Once that happens it behaves like there is a DMA translation problem with some subset of devices attached to the XHCI controller continuing to work and others failing. This is odd because if the DT has an error, I would expect there to be DMA issues in the case where the XHCI is never reset, and that doesn't happen.

**Expected behaviour**

I would hope that following the kernel reconfiguring the PCIe root port the low-level firmware notices this and adjusts the way it sets up the XHCI firmware window.

**Actual behaviour**
XHCI failures in Linux following any NOTIFY_XHCI_RESET calls.

**System**

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW

* 4B-8G

* Which OS and version (`cat /etc/rpi-issue`)?

* Kernel 5.13/14 tested.
 
* Which firmware version (`vcgencmd version`)?

* I tried all the start4. firmware images from the past few months, and the behavior seems consistent.
 
* Which kernel version (`uname -a`)?

5.14rc6

**Logs**

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.14.0-rc6PCIQUIRK+ (jlinton@uriel.jhome.net) (aarch64-linux-gnu-gcc (GCC) 11.2.1 20210728 (Red Hat Cr1
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] efi: EFI v2.70 by EDK2
[    0.000000] efi: ACPI 2.0=0x33b20018 SMBIOS=0x37200000 SMBIOS 3.0=0x371e0000 MEMATTR=0x353b9018 RNG=0x373db798 MEMRESERVE=0x3377 
[    0.000000] efi: seeding entropy pool
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000027c00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] NUMA: NODE_DATA [mem 0x1fefa9c80-0x1fefbffff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000]   Device   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x00000000001fffff]
[    0.000000]   node   0: [mem 0x0000000000200000-0x0000000033843fff]
[    0.000000]   node   0: [mem 0x0000000033844000-0x0000000033b1ffff]
[    0.000000]   node   0: [mem 0x0000000033b20000-0x0000000036ffffff]
[    0.000000]   node   0: [mem 0x0000000037000000-0x00000000373dffff]
[    0.000000]   node   0: [mem 0x00000000373e0000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA32: 19456 pages in unavailable ranges
[    0.000000] On node 0, zone Normal: 16384 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 23 pages/cpu s54680 r8192 d31336 u94208
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[... 78 lines cut ...]
[    6.416876] random: fast init done
[    6.432328] bcmgenet fd580000.ethernet: GENET 5.0 EPHY: 0x0000
[    6.441323] mmc1: new high speed SDIO card at address 0001
[    6.446718] pci_bus 0000:00: root bus resource [bus 00-ff]
[    6.453338] pci_bus 0000:00: root bus resource [mem 0x600000000-0x63fffffff] (bus address [0xc0000000-0xffffffff])
[    6.476736] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    6.482838] pci 0000:00:00.0: PME# supported from D0 D3hot
[    6.499916] pci 0000:01:00.0: [1106:3483] type 00 class 0x0c0330
[    6.505999] pci 0000:01:00.0: reg 0x10: [mem 0x638000000-0x638000fff 64bit]
[    6.507026] libphy: bcmgenet MII bus: probed
[    6.540653] pci 0000:01:00.0: PME# supported from D0 D3cold
[    6.578456] pci 0000:00:00.0: BAR 14: assigned [mem 0x600000000-0x6000fffff]
[    6.578476] pci 0000:01:00.0: BAR 0: assigned [mem 0x600000000-0x600000fff 64bit]
[    6.578491] pci 0000:00:00.0: PCI bridge to [bus 01]
[    6.578496] pci 0000:00:00.0:   bridge window [mem 0x600000000-0x6000fffff]
[    6.578886] pcieport 0000:00:00.0: PME: Signaling with IRQ 46
[    6.583184] pcieport 0000:00:00.0: AER: enabled with IRQ 46
[    6.612070] xhci_hcd 0000:01:00.0: xHCI Host Controller
[    6.612233] xhci_hcd 0000:01:00.0: new USB bus registered, assigned bus number 1
[    8.767368] random: crng init done
[    8.770808] random: 7 urandom warning(s) missed due to ratelimiting
[   16.612262] xhci_hcd 0000:01:00.0: can't setup: -110
[   16.617291] xhci_hcd 0000:01:00.0: USB bus 1 deregistered
[   16.622895] xhci_hcd 0000:01:00.0: init 0000:01:00.0 fail, -110
[   16.629057] xhci_hcd: probe of 0000:01:00.0 failed with error -110

timg236, 2021-08-25:
NOTIFY_XHCI_RESET should only be called after the PCIe fundamental reset is asserted which will cause the VLI to be reset. There doesn't appear to be a VLI 'software reset' mechanism available and the firmware can't track the PCIe state to see whether it should make the mbox request a NOOP.
Before starting the 'ARM stage' the firmware asserts PCIe reset. UEFI / UBOOT etc should do the same. It's ok to do other PCIe init but bus-mastering should not be enabled on the XHCI until the system is ready to call NOTIFY_XHCI_RESET

jlinton, 2021-08-25:
I believe that the pcie-brcmstb DT code is doing a full reset sequence when it is probed.
note: https://elixir.bootlin.com/linux/v5.14-rc7/source/drivers/pci/controller/pcie-brcmstb.c#L880
and then:
https://elixir.bootlin.com/linux/v5.14-rc7/source/drivers/pci/controller/pcie-brcmstb.c#L962

timg236, 2021-09-23:
It's worth making sure that the .dtb files are up to date, otherwise, Linux won't find the reset controller.

https://github.com/raspberrypi/linux/blob/rpi-5.10.y/arch/arm/boot/dts/bcm2711-rpi-cm4.dts#L288

jlinton, 2021-09-27:
Fairly certain I was running with a new enough DT that the mailbox was being triggered. I added some extra pr_xx's around the reset and mailbox send, which printed information as each step was executed. I've been wondering if there is a timing problem around how long it takes between the reset and the mailbox call, or whether the PERST sequence is being held long enough because if I modify the DT to support non translating DMA's everything will work if the mailbox controller to send the XHCI notification is disabled even if the bus is reset.

timg236, 2025-03-22:
Debugging something else, I've spotted a potential improvement for resetting/reloading the VLI firmware.

With the attached start4.elf/fixup4.dat the following reset sequence seems reliable. 

It's an old Issue, but I think this may help simplify/remove other workarounds and would appreciate any test feedback

```
sudo su -c "echo 1 > /sys/bus/pci/devices/0000\:00\:00.0/remove"
sudo su -c "lspci"
sudo su -c "lsusb"
sleep 1
sudo su -c "echo 1 > /sys/bus/pci/rescan"
sleep 3
sudo su -c "lspci"
```
Debug added to vclog
```
sudo vclog -m
```

[vl805-reset.zip](https://github.com/user-attachments/files/19402547/vl805-reset.zip)

timg236, 2025-05-08:
The updated firmware binaries were released to the firmware repo in this commit. Closing as resolved.
https://github.com/raspberrypi/firmware/commit/d828cc856bc81901a2a3fe5e1ad6231e72f21c97

closed 2025-05-08 by timg236