raspberrypi/firmware #1618: aux uart baud rate wrong, despite setting enable_uart=1

cleverca22, 2021-08-31:
```
[root@nixos:~]# cat /boot/config.txt 
enable_uart=1
avoid_warnings=1
arm_64bit=1
kernel=kernel.img
initramfs initrd followkernel
enable_uart=1
uart_2ndstage=1
[root@nixos:~]# cat /boot/cmdline.txt 
boot.shell_on_fail console=tty1 console=serial0,115200 loglevel=4

[root@nixos:~]# cat /proc/cpuinfo
Revision        : c03112
Serial          : 10000000d328cdce
Model           : Raspberry Pi 4 Model B Rev 1.2
```
with firmware rev 63b192231130f1bdd074f3ae0794889d53bdeb06, /dev/ttyS0 works perfectly
with firmware rev 2d83e2c8e07a81ab7ca4aa645773aa0feed8d7dd, the baud rate is wrong
problem still exist on 24f05a6e0eadcf001159e3618759cfb51761fd0e (current master)

kernel and os version where not changed while testing the differences

stsquad, 2021-09-29:
For reference the instructions: https://xenproject.org/2020/09/29/xen-on-raspberry-pi-4-adventures/ will fail on current firmware. Downgrading the firmware gets me to a point that I can see serial output all the way to booting Xen.

pelwell, 2021-10-04:
I use the UART every day, and it's working fine for me, 32- and 64-bit, with and without `dtoverlay=disable-bt`:
```
pi@raspberrypi:~$ sudo reboot
[   29.282782] reboot: Restarting system
Read start4.elf bytes  2241216 hnd 0x000137bc
Read fixup4.dat bytes     5413 hnd 0x0001827a
Firmware: 778b6a4f3c7d8d48bb63c02c47bcfbac79417bea Sep 28 2021 11:27:35
0x00c03114 0x00000000 0x00000fff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Starting start4.elf @ 0xfec00200 partition 0
PCI reset
+

MESS:00:00:04.033170:0: arasan: arasan_emmc_open
MESS:00:00:04.197913:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:04.200631:0: brfs: File read: 124 bytes
MESS:00:00:04.325382:0: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:04.334898:0: HDMI1:EDID error reading EDID block 0 attempt 1
MESS:00:00:04.344406:0: HDMI1:EDID error reading EDID block 0 attempt 2
MESS:00:00:04.353920:0: HDMI1:EDID error reading EDID block 0 attempt 3
MESS:00:00:04.363430:0: HDMI1:EDID error reading EDID block 0 attempt 4
MESS:00:00:04.372945:0: HDMI1:EDID error reading EDID block 0 attempt 5
MESS:00:00:04.382453:0: HDMI1:EDID error reading EDID block 0 attempt 6
MESS:00:00:04.391967:0: HDMI1:EDID error reading EDID block 0 attempt 7
MESS:00:00:04.401477:0: HDMI1:EDID error reading EDID block 0 attempt 8
MESS:00:00:04.410991:0: HDMI1:EDID error reading EDID block 0 attempt 9
MESS:00:00:04.415489:0: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:04.429855:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:05.552415:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:05.559708:0: *** Restart logging
MESS:00:00:05.561133:0: brfs: File read: 124 bytes
MESS:00:00:05.635596:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:05.645637:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 1
MESS:00:00:05.659425:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 2
MESS:00:00:05.669524:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 3
MESS:00:00:05.679611:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 4
MESS:00:00:05.689707:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 5
MESS:00:00:05.699798:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 6
MESS:00:00:05.709893:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 7
MESS:00:00:05.719980:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 8
MESS:00:00:05.730076:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 9
MESS:00:00:05.735149:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:05.745802:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:05.755898:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 1
MESS:00:00:05.765988:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 2
MESS:00:00:05.776083:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 3
MESS:00:00:05.786171:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 4
MESS:00:00:05.796267:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 5
MESS:00:00:05.806357:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 6
MESS:00:00:05.816452:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 7
MESS:00:00:05.826539:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 8
MESS:00:00:05.836635:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 9
MESS:00:00:05.841707:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:05.847311:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:05.856066:0: HDMI0: hdmi_pixel_encoding: 300000000
MESS:00:00:05.861541:0: HDMI1: hdmi_pixel_encoding: 300000000
MESS:00:00:05.872731:0: dtb_file 'bcm2711-rpi-4-b.dtb'
MESS:00:00:05.891397:0: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
MESS:00:00:05.894682:0: Loading 'bcm2711-rpi-4-b.dtb' to 0x100 size 0xc2a9
MESS:00:00:05.932858:0: brfs: File read: 49833 bytes
MESS:00:00:05.954754:0: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
MESS:00:00:06.117415:0: brfs: File read: 1667 bytes
MESS:00:00:06.124493:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:06.148822:0: brfs: File read: 124 bytes
MESS:00:00:06.154091:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:06.156748:0: Read command line from file 'cmdline.txt':
MESS:00:00:06.162592:0: 'console=serial0,115200 console=tty1 root=PARTUUID=b051a65c-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait splash plymouth.ignore-serial-consoles'
MESS:00:00:06.299204:0: brfs: File read: 199 bytes
MESS:00:00:06.986085:0: brfs: File read: /mfs/sd/kernel8.img
MESS:00:00:06.988643:0: Loading 'kernel8.img' to 0x80000 size 0x78a5af
[... 25 lines cut ...]
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 996912
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:91:36:7B vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000 console=ttyS0,115200 console=tty1 root=PARTUUID=b051a65c-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait splash plymouth.ignore-serial-consoles
...
[    5.808433] systemd[1]: Starting Apply Kernel Variables...
[    5.825678] systemd[1]: Started Journal Service.

Raspbian GNU/Linux 11 raspberrypi ttyS0

raspberrypi login: pi
Password:
Linux raspberrypi 5.10.63-v8+ #1457 SMP PREEMPT Tue Sep 28 11:27:02 BST 2021 aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Oct  4 13:40:07 BST 2021 on tty1
pi@raspberrypi:~$ vcgencmd version
Sep 28 2021 11:27:35
Copyright (c) 2012 Broadcom
version 778b6a4f3c7d8d48bb63c02c47bcfbac79417bea (clean) (release) (start)
pi@raspberrypi:~$
```

cleverca22, 2021-10-04:
let me repeat my testing, and see exactly what is going on...

cleverca22, 2021-10-04:
closing until i can get around to testing it further, it clearly works on raspi-os, i'll reopen whenever i get to testing more, if i can confirm the issue is in the firmware side

closed 2021-10-04 by cleverca22

CarlFK, 2021-11-27:
me too.

Raspberry Pi 3 Model B Plus Rev 1.3

enable_uart=1
avoid_warnings=1
arm_64bit=1
enable_uart=1
uart_2ndstage=1
# core_freq=250 (adding this 'fixes' it)

$ tio /dev/ttyUSB0 
[tio 23:56:33] tio v1.32
[tio 23:56:33] Press ctrl-t q to quit
[tio 23:56:33] Connected

Raspberry Pi Bootcode
Read File: config.txt, 73
Read File: start.elf, 2965664 (bytes)
Read File: fixup.dat, 7285 (bytes)
MESS:00:00:21.089558:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:21.093687:0: brfs: File read: 73 bytes
MESS:00:00:21.169568:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:22.022977:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:22.031330:0: *** Restart logging
MESS:00:00:22.033819:0: brfs: File read: 73 bytes
MESS:00:00:22.057897:0: HDMI0: hdmi_pixel_encoding: 162000000
MESS:00:00:22.067240:0: dtb_file 'bcm2710-rpi-3-b-plus.dtb'
MESS:00:00:22.117805:0: brfs: File read: /mfs/sd/bcm2710-rpi-3-b-plus.dtb
MESS:00:00:22.122901:0: Loading 'bcm2710-rpi-3-b-plus.dtb' to 0x100 size 0x7376
MESS:00:00:22.143892:0: brfs: File read: 29558 bytes
MESS:00:00:22.153174:0: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
MESS:00:00:22.230676:0: brfs: File read: 1667 bytes
MESS:00:00:22.237474:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:22.250974:0: brfs: File read: 73 bytes
MESS:00:00:22.257485:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:22.261474:0: Read command line from file 'cmdline.txt':
MESS:00:00:22.267350:0: 'boot.shell_on_fail console=tty1 console=serial0,115200 loglevel=4'
MESS:00:00:22.286933:0: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
MESS:00:00:23.085861:0: brfs: File read: 281 bytes
MESS:00:00:34.685633:0: brfs: File read: /mfs/sd/kernel8.img
MESS:00:00:34.689605:0: Loading 'kernel8.img' to 0x80000 size 0x78a2b4
MESS:00:00:36.765508:0: Kernel relocated to 0x200000
MESS:00:00:36.768776:0: Device tree loaded to 0x2eff8800 (size 0x77ef)
MESS:00:00:36.776473:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:36.782768:0: uart: Baud rate change done...
MESS:00:00:36.786200:0: uart: Baud rate�2�
                                          09�|L8�������2@�#B�|��p��2�|2npLb��NN�2�Npb�n��
                                                                                         0���l�
                                                                                               0��|2�
                                                                                                     �n����n�Lnn��n����Nnr��n��Nr|�N�|�b��nNpnb�