raspberrypi/firmware #1622: Feature: kernel config to support BPF programs

mazzy89, 2021-09-11:
Currently it's not possibile to run certain programs that make use of eBPF functionalities see for instance Cilium when using the official kernel. It requires certain kernel config module to be enabled as reported in here https://docs.cilium.io/en/v1.9/operations/system_requirements/#linux-kernel

I've spent some time and I've noticed that running the latest firmware version, it misses these two ones 

```
# missing
CONFIG_NET_CLS_BPF
CONFIG_BPF_JIT
```

while these two are set be optionally loaded 

```
CONFIG_NET_SCH_INGRESS
CONFIG_CRYPTO_USER_API_HASH
```

I would prefer to have them upstream rather to bake my own kernel and use it.

closed 2021-09-13 by mazzy89

seamusdemora, 2022-04-19:
@mazzy89 wrote:

> I would prefer to have them upstream rather to bake my own kernel and use it. 

I just read an article about BPF & eBPF, and immediately began searching to learn if it is included in RPi. Your comment _suggests_ that it may not be. Can you confirm that?

mazzy89, 2022-05-01:
The current RPI firmware does not ship with any kernel parameter enabled to support eBPF programs.