raspberrypi/firmware #1625: RPI 2B Crashes due to cpu frequency changes

segalale, 2021-09-15:
We are having a weird issue with RPI model 2B rev 1.1/1.2.

Our RPI is located in a closed system where the ambient temperature is around 50C most of the time (measured by vcgencmd measure_temp).
Once in a while (1-40 hours), the RPI freezes and reboots itself.
System load is not high (Neither cpu usage nor IO). There is absolutely nothing about it in any of the log files (systemd is in persistent mode), and no console messages (over UART).
The power supply is solid, and we have tried several RPI's and several different brands of SD cards.
The OS is the latest Raspbian Lite, however we have also tried installing Arch Linux, downgrading Raspbian to kernel 4.19 and even installing FreeBSD, all without success.

To reproduce the problem, we have found that it is enough to wait until the system runs a few minutes with ambient temperature above 50C (in a closed box), and then alternate the cpu frequencies between 600Mhz and 900Mhz in a loop. The system crashes in a few seconds - minutes.

If the cpu is set to 600Mhz in config.txt using:
arm_freq=600
the problem is not reproduced.

popcornmix, 2021-09-15:
> Once in a while (1-40 hours), the RPI freezes and reboots itself.

Do you really mean it spontaneously reboots?
Do you have a watchdog service or external hardware that issues a reboot?

Does it occur if you don't have the high ambient temperature?

segalale, 2021-09-15:
Yes, it spontaneously reboots. Once in a while it even goes to shutdown mode.
We have tried disabling the built-in watchdog, with the same result.

It does not occur (at least not easily reproduced) without the high temperature. 
I have managed to reproduce this behavior on my table, with a clean install, just by covering the Pi with a few pieces of cloth (for the high temperature), running:

stress-ng -c 4
for a while, and then changing the frequencies by cpufreq-set in a loop.

I should also note that it does not occur on RPI 4B.

pelwell, 2021-09-15:
Could you try adding a small over-voltage in config.txt? `over_voltage=2` for example.

segalale, 2021-09-15:
Sorry I haven't mentioned it. I've tried playing with over_voltage (positive and negative), without success.

pelwell, 2021-09-15:
No problem - it's just something to eliminate.

pelwell, 2021-09-16:
Have you tested with `force_turbo=1`, which will keep the cores at their maximum frequency?

segalale, 2021-09-16:
No, I haven't, but I suspect that it will not crash.
When running stress-ng, the cores are at their maxmium all the time, and איק RPI doesn't crash. The only way I've managed to reproduce it is by altering the frequencies in a loop, using cpufrequtils.

popcornmix, 2021-09-16:
> We have tried disabling the built-in watchdog, with the same result.

The watchdog isn't enabled by default. Are you saying you had initially enabled it?

I'm questioning this as I've never seen a case of a Pi rebooting without either the watchdog being enabled, or a power supply issue (either under-voltage on the 5V line, or over-current from the PMIC). The usual symptom of other failures is a hang (that typically needs a power cycle to recover).

Can you add a `vcgencmd get_throttled` to your frequency changing loop and report what is output before the reboot.

segalale, 2021-09-16:
> The watchdog isn't enabled by default. Are you saying you had initially enabled it?
>
> I'm questioning this as I've never seen a case of a Pi rebooting without either the watchdog being enabled, or a [...]

Yes, we have enabled the watchdog. However, I've tried it on a clean installation as well, with similar results.

> Can you add a `vcgencmd get_throttled` to your frequency changing loop and report what is output before the reboot.

Will do, thanks.

segalale, 2021-09-16:
I have added  `vcgencmd get_throttled` to the loop, and the result was 0x0 all the way.

popcornmix, 2021-09-17:
Can you provide the script you use to provoke this?
Ideally something that will run on Pi2 with a clean install of RPiOS.
We can try running it here (with cloth insulating it if needed).

Ruffio, 2021-09-17:
> Sure, the following script crashes my RPI2 (with cloth for insulation) in
> a little more than 10 minutes (run as root):
>
> `#!/bin/bash
>
> stress-ng -c 4 &
>
> echo "Waiting 10 minutes"
> sleep 10m
>
> echo "Changing CPU frequencies"
> while true
> do
> cpufreq-set -f 600MHz
> cpufreq-set -f 9000Mhz
> done
> `
>
> It requires cpufrequtils and stress-ng.

Damn 9000 MHz?

segalale, 2021-09-17:
Sorry, it's a typo. Works with 900Mhz.

segalale, 2021-09-23:
> Can you provide the script you use to provoke this?
> Ideally something that will run on Pi2 with a clean install of RPiOS.
> We can try running it here (with cloth insulating it if needed).

Sorry for bumping this up. Did you have any success reproducing the issue?