raspberrypi/firmware #1626: 480i no longer working under fkms

alanbork, 2021-09-17:
As of efe79c1df6df24e666c08dd68e1636ed24210fc1, switching into 480i causes my TV to report "no signal". This is outside of x, using modetest. I don't see any error messages in dmesg.

reverting back to 9dbcd6dfed16cc7de4a67b9ca445c4847f68c4c2 fixes it.  I can search through the revisions in between if you have any suggests of likely candidates, but it's a lot to just try blindly.

popcornmix, 2021-09-18:
Can you try https://github.com/Hexxeh/rpi-firmware/commit/f8357e4d1dbc6b98dcec9acae9dbba5d9fb86040 and the one before? That's one commit that would affect 480i.

alanbork, 2021-09-19:
f8357e4d1dbc6b98dcec9acae9dbba5d9fb86040 still has the problem. 
64132d67d3e083076661628203a02d27bf13203c (which seems to be the next oldest release) fixes it.  So it seems that f8357e4d1dbc6b98dcec9acae9dbba5d9fb86040 is where the regression happens.

bahsura, 2022-08-14:
Was this ever fixed?

popcornmix, 2022-08-15:
I believe interlaced output is now working on fkms and kms.
Have you tried recently?
You'll probably want to test rpi-update firmware/kernel currently - although apt should be updated to match that this week.