raspberrypi/firmware #1627: Video decoder crash every few minutes

Malvineous, 2021-09-19:
There seems to be some bugs in the H264 decoder.  I am streaming video from another Raspberry Pi, and it was working fine, until I made the video a few pixels taller so I could add the time, date and other info in a bar along the bottom of the video.  (The video went from 1296x972 to 1280x1008.  I had to reduce the width due to #1608 but I don't think it's related.)

Since then, the playback stops randomly.  I am playing two video streams at the same time (only one of which has had its size changed and started this problem), and when this bug is triggered, both streams break.  Sometimes part of the picture from one video ends up showing in the other video's window, sometimes the video becomes very badly corrupted (turns green and magenta) but mostly the video just stops updating leaving only the last frame visible.

In order to fix it I have to exit both video players and restart - closing only one player is not enough.  Sometimes only restarting one player will show no video, but sometimes it comes up with the last stuck frame still visible.  Exiting all players and reloading them all gets this working again, but only for another few minutes until it crashes again.

I get various messages in the logs when this happens.  Here is one exerpt:

```
[ 1645.031404] bcm2835_mmal_vchiq: timed out waiting for sync completion
[ 1648.103401] bcm2835_mmal_vchiq: timed out waiting for sync completion
[ 1648.103439] bcm2835-codec bcm2835-codec: bcm2835_codec_stop_streaming: Failed disabling o/p port, ret -62
[ 1650.119420] bcm2835-codec bcm2835-codec: bcm2835_codec_stop_streaming: Timeout waiting for buffers to be returned - 19 outstanding
[ 1650.119534] ------------[ cut here ]------------
[ 1650.119577] WARNING: CPU: 2 PID: 773 at drivers/media/common/videobuf2/videobuf2-core.c:1945 __vb2_queue_cancel+0x1b0/0x220 [videobuf2_common]
[ 1650.119585] Modules linked in: rpcsec_gss_krb5 option usb_wwan vc4 usbserial btsdio cec bcm2835_codec(C) bcm2835_isp(C) bluetooth bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) drm_kms_helper v4l2_mem2mem ecdh_generic videobuf2_vmalloc ecc videobuf2_dma_contig snd_soc_core videobuf2_memops videobuf2_v4l2 snd_compress videobuf2_common snd_pcm_dmaengine videodev v3d gpu_sched raspberrypi_hwmon iproc_rng200 mc snd_pcm snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops rpivid_mem uio_pdrv_genirq nvmem_rmem uio sch_fq_codel drm fuse drm_panel_orientation_quirks ip_tables x_tables ipv6 sr_mod cdrom brcmfmac cfg80211 brcmutil
[ 1650.119832] CPU: 2 PID: 773 Comm: video_decoder Tainted: G         C        5.10.52-4-ARCH #1
[ 1650.119842] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[ 1650.119855] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO BTYPE=--)
[ 1650.119874] pc : __vb2_queue_cancel+0x1b0/0x220 [videobuf2_common]
[ 1650.119893] lr : __vb2_queue_cancel+0x34/0x220 [videobuf2_common]
[ 1650.119901] sp : ffffffc01a61baf0
[ 1650.119909] x29: ffffffc01a61baf0 x28: 0000007f94b91750 
[ 1650.119925] x27: ffffffc010fdc000 x26: ffffffc01a61be50 
[ 1650.119941] x25: ffffff8043dab868 x24: ffffff8069c05f40 
[ 1650.119955] x23: ffffff8043d602a0 x22: ffffff8043baa300 
[ 1650.119971] x21: ffffff80428d7028 x20: ffffff80428d70d0 
[ 1650.119985] x19: ffffff80428d7028 x18: 0000000000000030 
[ 1650.120000] x17: 0000000000000000 x16: 0000000000000000 
[ 1650.120015] x15: ffffff8069c05e80 x14: 00000000000001d2 
[ 1650.120029] x13: ffffff8069c05e80 x12: ffffffc01a61b760 
[ 1650.120043] x11: ffffffc01123bae0 x10: 00000000fffff000 
[ 1650.120058] x9 : ffffffc01123bae0 x8 : ffffffc0111e3ae0 
[ 1650.120072] x7 : ffffffc01123bae0 x6 : 0000000000000000 
[ 1650.120086] x5 : 0000000000000000 x4 : ffffff80fb85d6c8 
[ 1650.120101] x3 : ffffffc010fdc000 x2 : 6c8354effb15d400 
[ 1650.120116] x1 : 00000000000055ac x0 : 0000000000000013 
[ 1650.120132] Call trace:
[ 1650.120152]  __vb2_queue_cancel+0x1b0/0x220 [videobuf2_common]
[ 1650.120170]  vb2_core_queue_release+0x24/0x50 [videobuf2_common]
[ 1650.120188]  vb2_queue_release+0x10/0x20 [videobuf2_v4l2]
[ 1650.120210]  v4l2_m2m_ctx_release+0x20/0x40 [v4l2_mem2mem]
[ 1650.120225]  bcm2835_codec_release+0x5c/0xf4 [bcm2835_codec]
[ 1650.120288]  v4l2_release+0xbc/0xf0 [videodev]
[ 1650.120304]  __fput+0x78/0x230
[ 1650.120314]  ____fput+0x10/0x20
[ 1650.120327]  task_work_run+0x80/0x140
[ 1650.120339]  do_exit+0x32c/0xa10
[ 1650.120348]  do_group_exit+0x38/0xa0
[ 1650.120359]  get_signal+0x10c/0x844
[ 1650.120372]  do_notify_resume+0x188/0x9d0
[ 1650.120381]  work_pending+0xc/0x43c
[ 1650.120390] ---[ end trace 5fd4ec3580a88085 ]---
[ 1650.120406] videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000ac96b1de in active state
[ 1650.120420] videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000003219955b in active state
[ 1650.120432] videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000e7922934 in active state
[ 1650.120444] videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000253d780c in active state
[ 1650.120454] videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000003380b69f in active state
[ 1650.120463] videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000002566c086 in active state
[ 1650.120474] videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000005c8d5e2a in active state
[ 1650.120485] videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000009293a3da in active state
[ 1650.120494] videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000f9455a5d in active state
[ 1650.120504] videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000b31fb82e in active state
[ 1650.120515] videobuf2_common: driver bug: stop_streaming operation is leaving buf 0000000060988941 in active state
[ 1650.120525] videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000001d57d7a5 in active state
[ 1650.120535] videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000000688a4ea in active state
[ 1650.120544] videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000007eb7ac4f in active state
[ 1650.120556] videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000f938963d in active state
[ 1650.120565] videobuf2_common: driver bug: stop_streaming operation is leaving buf 0000000089ad923c in active state
[ 1650.120576] videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000000a274fa8 in active state
[... 30 lines cut ...]
[ 1660.139496] bcm2835-codec bcm2835-codec: bcm2835_codec_create_component: failed to create component ril.video_decode
[ 1663.207476] bcm2835_mmal_vchiq: timed out waiting for sync completion
[ 1663.207516] bcm2835_mmal_vchiq: vchiq_mmal_component_init: failed to create component -62 (Not enough GPU mem?)
[ 1663.207535] bcm2835-codec bcm2835-codec: bcm2835_codec_create_component: failed to create component ril.video_decode
[ 1663.626347] bcm2835_mmal_vchiq: received invalid message context 5!
[ 1663.628625] bcm2835-codec bcm2835-codec: bcm2835_codec_create_component: failed to create component ril.video_decode
[ 1663.629371] bcm2835_mmal_vchiq: received invalid message context 1!
[ 1663.629395] bcm2835_mmal_vchiq: received invalid message context 1!
[ 1663.629626] bcm2835_mmal_vchiq: received invalid message context 1!
[ 1668.839518] bcm2835-codec bcm2835-codec: bcm2835_codec_stop_streaming: Timeout waiting for buffers to be returned - 19 outstanding
```
I tried increasing the GPU mem as suggested.  It was initially `gpu_mem=128` which was fine until I enlarged the video, now I get the same messages even if I specify `gpu_mem=384`.  I can't go large than this or the Pi won't boot.  I have `cma=300M` but changing this value doesn't seem to do anything.

**System**

* Which model of Raspberry Pi? Pi 4
* Which OS and version (`cat /etc/rpi-issue`)? Arch Linux ARM 64-bit
* Which firmware version (`vcgencmd version`)?
```
Sep 14 2021 16:51:17 
Copyright (c) 2012 Broadcom
version 1f74fc33fa785915fe0d7bd6d1cdfa220bc1ed7b (clean) (release) (start)
```
* Which kernel version (`uname -a`)?
`Linux 5.10.63-8-ARCH #1 SMP PREEMPT Thu Sep 16 14:24:31 UTC 2021 aarch64 GNU/Linux`

Malvineous, 2021-11-06:
I ordered some metal cases with "pillars" pressing down on the chips and thermal pads to fit between the Pi chips and the pillars.  The [first one](https://www.aliexpress.com/item/4000907704024.html) (a box shape with grooves in the top) lowered the temperature from 85°C down to a maximum of 53°C and completely stopped the freezing problem.

The [second case](https://www.aliexpress.com/item/4000240410050.html) had a fan as well and dropped the temperature by a further 10°C to 43°C and likewise the Pi has been running for over 24 hours now without a single unexpected reboot.

So this means the problem is definitely due to overheating, and apparently the throttling not being able to throttle back enough.

I know the ARM core gets throttled when the temperature is too high, but does the VideoCore get throttled?  In my case the ARM is sitting at around 22% CPU (~50% CPU on two cores, 0% on the other two cores) however I am using V4L2M2M to hardware decode two H264 streams.  It would seem the H264 decoding is what is contributing most of the heat, and possibly that isn't getting throttled which is why it kept overheating?

popcornmix, 2021-11-08:
@Malvineous have you tried with `over_voltage=2` as suggested earlier?
If your h264 silicon is borderline at default voltage, it may fail when temperature is higher.

Note: the h264 hardware creates negligible heat when compared to the arm.

Malvineous, 2021-11-10:
Sorry no I never got that far.  I took the top of the heatsink case off to try it - 32°C ambient today so good day to test.  I left it as-is to confirm it would reboot without any heatsink which it did, once it got to around 80°C.

I then switched it off, set `over_voltage=2`, then switched it back on again.  Despite being off for 30 seconds or so, it still locked up on the rainbow screen.  I then set `over_voltage=6` and that time it booted successfully, said 76°C just after the H264 playback started, then about 15 seconds later it rebooted.  When it rebooted it got past the rainbow screen but then froze on a black screen, before any kernel output appeared.

I then removed `over_voltage` and put the heatsink case back on, and it's been behaving normally for around an hour now, reporting just under 44°C.

I had a pedestal fan in the room for the first test, and this accidentally provided enough of a breeze to keep the temperature at around 72°C and it was behaving normally there too.  It wasn't until I blocked off enough airflow for it to get above 80°C that it rebooted.

So looks like it's down to the throttling not being aggressive enough (or this unit being sensitive above 80°C rather than 85°C) and it's just overheating.

Malvineous, 2021-11-16:
Going to close this one now as it definitely looks like it was due to overheating.  I suppose you could argue that it's still a bug in that the thermal throttling isn't fully working, but then you can also make the case that running it right up past the 80 degree mark is not very wise either.

closed 2021-11-16 by Malvineous