raspberrypi/firmware #1628: Booting from SD card partitioned with GPT with Hybrid MBR still does not work.

sallevan, 2021-09-26:
Albeit there is a comment under #1457 that some commit in a time fixes the problem, I disagree with that -- booting with hybrid MBR/GPT still does not work properly.

I have Raspberry Pi 4b with 4GB RAM for experiments and a pretty good SanDisk 256GB SD Card. GPT partition table was made the following way:

```
-. Primary GPT
1. EFI System Partition -- 200 MB FAT32
2. VideoCore Boot Partition -- 50 MB FAT16
3. Microsoft Reserved Partition -- 128 MB according to Microsoft requirements
4. Operating System Root Partition
5..8. Other partitions I need
-. Secondary GPT
```

Hybrid MBR, in theory, can be constructed in two ways:

```
1. EFI Protective Partition (type 238, it has to start at LBA 1 and to end at last LBA of GPT partition 1)
2. Primary DOS >=32MB (type 6, start and end LBA should be the same as of GPT partition 2)
3. Unused
4. Unused
```
This layout scheme might be known as "Boot Camp" hybrid MBR and was invented by UNIX world.

```
1. Primary DOS >=32MB (type 6, start and end LBA should be the same as of GPT partition 2)
2. Unused
3. Unused
4. EFI Protective Partition (type 238, it has to start at LBA 1 and to end at last LBA of GPT partition 1)
```
This layout scheme is known as "Windows-compatible" hybrid MBR as Windows before Win10 uses media partitioned this way as MBR-partitioned while UNIX sees it as GPT-partitioned.

If I stuck to VideoCore firmware dated 20200601 **and** use secondary MBR layout, RPi boots properly because _start4.elf_ from those days ignored GPT completely, firmware with dates between 20200601 and 20210107 is known not to work with such setups, but **newer firmware does not work for me too.**

When I use "Windows-compatible" layout, EEPROM loader correctly reads _start4.elf_, _fixup4.elf_ and _config.txt_ from MBR partition 1, which can be seen via serial console port with appropriate `enable_uart` and `uart_2ndstage` settings and then starts start4.elf, which, in turn, chokes up with this partition layout as I have described in https://github.com/raspberrypi/firmware/issues/1465#issuecomment-814005398 -- I've copied serial console log (large!) there.

When I use "Boot Camp" layout, RPi just keeps silent. Red (power) LED is constantly red, green (activity) LED is constantly green, HDMI video and serial output shows nothing.

I have to kindly ask to our fellow developers to pay an attention to the fact that Windows machines with Windows version less than 10 still exist with enough quantity, and non-working hybrid MBR booting prevents straight-and-easy data exchange with just pulling SD card off RPi and inserting it to Windows machine.

timg236, 2021-09-27:
"When I use "Boot Camp" layout, RPi just keeps silent. Red (power) LED is constantly red, green (activity) LED is constantly green, HDMI video and serial output shows nothing."

Has this ever worked?

sallevan, 2021-09-27:
> "When I use "Boot Camp" layout, RPi just keeps silent. Red (power) LED is constantly red, green (activity) LED
> is constantly green, HDMI video and serial output shows nothing."
> 
> Has this ever worked?

Where?

In Raspberry Pi world to the best of my knowledge, no.

In macOS/Windows hybrid world -- yes, and it is the only supported mass storage layout by Apple and Microsoft. :) Every piece of Apple hardware which uses Boot Camp (macOS/Windows dual boot solution provided by Apple) hard drive will be partitioned that way during Microsoft Windows installation preparation phase.

timg236, 2021-09-27:
Ok, just verifying that this is a feature request. I thinks it’s unlikely that we’ll add support for more boot partition schemes in the near future. There’s a fix for Hybrid / GPT in the related PR.

sallevan, 2021-09-27:
Yes, I completely agree with that it's nothing more than nice-to-have feature. Thank you again for fixing hybrid-style partitioning boot!

closed 2021-09-27 by sallevan