raspberrypi/firmware #1630: cannot find sd card on pi 4

jiangfenghua123, 2021-09-29:
https://carleton.ca/rcs/qnx/installing-qnx-on-raspberry-pi-4/

# MMU: 16-bit ASID 44-bit PA TCR_EL1=b5183519
GICv2: 256 interrupts
GICv2: routing SPIs to gic cpu 0
cpu0: MPIDR=80000000
cpu0: MIDR=410fd083 Cortex-A72 r0p3
cpu0: CWG=4 ERG=4 Dminline=4 Iminline=4 PIPT
cpu0: CLIDR=a200023 LoUU=1 LoC=2 LoUIS=1
cpu0: L1 Icache 48K linesz=64 set/way=256/3
cpu0: L1 Dcache 32K linesz=64 set/way=256/2
cpu0: L2 Unified 1024K linesz=64 set/way=1024/16
cpu0: GICv2 cpu interface 0
Loading IFS...done
cpu1: MPIDR=80000001
cpu1: MIDR=410fd083 Cortex-A72 r0p3
cpu1: CWG=4 ERG=4 Dminline=4 Iminline=4 PIPT
cpu1: CLIDR=a200023 LoUU=1 LoC=2 LoUIS=1
cpu1: L1 Icache 48K linesz=64 set/way=256/3
cpu1: L1 Dcache 32K linesz=64 set/way=256/2
cpu1: L2 Unified 1024K linesz=64 set/way=1024/16
cpu1: GICv2 cpu interface 1
cpu2: MPIDR=80000002
cpu2: MIDR=410fd083 Cortex-A72 r0p3
cpu2: CWG=4 ERG=4 Dminline=4 Iminline=4 PIPT
cpu2: CLIDR=a200023 LoUU=1 LoC=2 LoUIS=1
cpu2: L1 Icache 48K linesz=64 set/way=256/3
cpu2: L1 Dcache 32K linesz=64 set/way=256/2
cpu2: L2 Unified 1024K linesz=64 set/way=1024/16
cpu2: GICv2 cpu interface 2
cpu3: MPIDR=80000003
cpu3: MIDR=410fd083 Cortex-A72 r0p3
cpu3: CWG=4 ERG=4 Dminline=4 Iminline=4 PIPT
cpu3: CLIDR=a200023 LoUU=1 LoC=2 LoUIS=1
cpu3: L1 Icache 48K linesz=64 set/way=256/3
cpu3: L1 Dcache 32K linesz=64 set/way=256/2
cpu3: L2 Unified 1024K linesz=64 set/way=1024/16
cpu3: GICv2 cpu interface 3

System page at phys:0000000000012000 user:ffffff8040207000 kern:ffffff8040204000
Starting next program at vffffff80600603c0
All ClockCycles offsets within tolerance
Welcome to QNX Neutrino on the Raspberry Pi 4 (Armv8-A Cortex-A72 core)
Starting watchdog...
Starting random service...
Starting serial driver (/dev/ser1)
Starting serial driver (/dev/ser3)
Starting I2C driver (/dev/i2c1)...
Starting SPI 0 driver (/dev/spi0)...
bcm2711_cfg:target_sclk=5000000, dvif=100.000000, div=100bcm2711_cfg: sclk=5000000, divisor=100bcm2711_cfg:target_sclk=5000000, dvif=100.000000, div=100bcm2711_cfg: sclk=5000000, divisor=100SPI opmode: 1Starting SDMMC driver for SD card (SDIO0)...
Starting PCI Server...
Inform vc to load vl805 firmware
00000000 
Starting USB xHCI controller in the host mode (/dev/usb/*)...
xpt_configure:  No sdmmc interfaces found
Unable to open device: No such file or directory
Unable to open device: No such file or directory
mount: Can't mount / (type default) 
mount: Possible reason: Invalid argument 
mkqnx6fs: unable to access '/dev/sd0t179' - No such file or directory
mount: Can't mount /ti_fs (type qnx6) 
mount: Possible reason: Invalid argument 
mkdir: /ti_fs/etc: Function not implemented
mkdir: /ti_fs/etc/ssh: Function not implemented
Generating public/private dsa key pair.
Saving key "/ti_fs/etc/ssh/ssh_host_dsa_key" failed: No such file or directory
Generating public/private rsa key pair.
Saving key "/ti_fs/etc/ssh/ssh_host_rsa_key" failed: No such file or directory
Starting Network driver...
Unable to load host key: /ti_fs/etc/ssh/ssh_host_rsa_key
Unable to load host key: /ti_fs/etc/ssh/ssh_host_dsa_key
sshd: no hostkeys available -- exiting.
Starting Audio driver...

JamesH65, 2021-09-29:
QNX is not a OS supported by Raspberry Pi, you need to talk to the QNX devs.

closed 2021-09-29 by JamesH65

jiangfenghua123, 2021-09-30:
Ok