raspberrypi/firmware #1631: TFTP boot in start4.elf (RPi 4B) ignores duplicate block numbers, increments ack counter, hangs

jtbraun, 2021-10-05:
During TFTP boot on a Raspberry Pi Model 4B, sometimes boot hangs. It occurs when this happens
* Normally, the TFTP sends a block, the Pi acks, it, continues
* Server (for whatever reason) sends block N twice (duplicate transmission w/o receiving the ack from the Pi
* The Pi's TFTP client must ignore the block number in the second packet, and increment the "last block received counter" to N+1, as the next ACK is for block N+1, not N.
* The TFTP server never receives an ACK for block N, so continues to send block N
* Inconsistent w/ my "ignores the block counter" theory above, the Pi continues to send multiple acknowledgements for block N+1 (even though it only received N multiple times, not N+1)

**To reproduce**
Boot over TFTP many times. For me, happens ~1/20-30 times

**Expected behaviour**
I expect the Pi to ONLY send acknowledge packets for block numbers it's received, not future ones.

**Actual behaviour**
The Pi acks a TFTP block it never received.

**System**
* Which model of Raspberry Pi? **Pi4B**
* Which OS and version (`cat /etc/rpi-issue`)? Pre-OS boot, n/a.
* Which firmware version (`vcgencmd version`)? `commit 6b855fb1f6e753522d38a121eca9a534bf46081f (HEAD -> stable, tag: 1.20210928`
* Which kernel version (`uname -a`)? Pre-OS boot, n/a.

**Logs**
Screencap from wireshark. In this case, N = 2153 (note duplicate block sent at the cursor). An ack is received from the Pi for block 2154 (never transmitted). Since 2153 is never acked, the server continues to try to send it, and the Pi continues to ack 2154, and boot never happens. Presumably the data from 2153 is now incorrectly placed at 2154 as well.
[image]

This is with in.tftp from a Fedora 33 system. I've noticed a similar problem with TFTPD64_SE on Windows, where I'll get a bunch of error messages occasionally in it's logs. In that instance, I think TFTPD just keeps sending new blocks, and warns that it's receiving N+1 acks for them, because boot will progress, but will always hang/crash/fail (probably because the wrong data is in the wrong location in memory from these off-by-X errors)

peterharperuk, 2021-10-05:
Can you check the bootloader version? I think this might have been fixed.

## 2021-02-16 - Change VC version info & TFTP fix - BETA
<snip>
   * Change TFTP to ACK data blocks which it has already ACK'd instead of ignoring them.
<snip>

This became the "default" bootloader on 2021-04-19

jtbraun, 2021-10-05:
I updated the bootloader using the RPI imager sometime in September, and the the serial console reports:
`RPi: BOOTLOADER release VERSION:c305221a DATE: Sep  3 2020 TIME: 13:11:46 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1599135103 0x5926197b 0x00c03114`

From the wireshark trace, you can see that the bootloader is not ignoring the repeated blocks (it's still acking repeated blocks multiple times), but it's acking the WRONG block number. It's using the erronous N+1 number from it's internal state, rather than acking the N block it just received.

I still maintain that the first ack of 2154 (two down from the cursor) should NEVER have been sent, as only block 2153 has been received at that point. There's still an unintentional increment of the internal block counter in the bootloader, probably in the case where it received two block messages in a row before it's had a chance to ack the first one.

jtbraun, 2021-10-05:
Oops, that's Sep 3 2020, not 2021. I'll see about updating to the latest stable bootloader.

jtbraun, 2021-10-05:
Odd. Using `rpi-eeprom-update` it reports that Thus Sep 3 12::11:43 PM UTC 2020 is the current and latest version of the bootloader. Is there a better way to get the "default" 2021-04-19 bootloader on here?

jtbraun, 2021-10-05:
I managed to get the bootloader updated via SDCARD... the bootloader recovery path doesn't work with large FAT32 partitions. If I just used the imager to create a recovery card, I got no LEDs blinking on the board and no video output. I manually created a 16GB FAT32 partition and copied the recovery files over, and it flashed the bootloader. I'll file a bug against the imager for that.  I'll run with this for a bit and let you know if I see any boot failures.

jtbraun, 2021-10-05:
FYI: https://github.com/raspberrypi/rpi-imager/issues/283

jtbraun, 2021-10-05:
Okay, I got this running, and now occasionally see in the bootloader's serial logs messages like:
`Duplicate block 210 expected 211`

And you can see this happening in the logs:
[image]

The RPI now (correctly) responds with an ack of the duplicated block number, and the transfer continues.

It looks like the dpkg for `rpi-eeprom` that installs on the 64-bit ubuntu image from the latest Raspberry Pi installer is really out-of-date, it's latest `critical` bootloader is the Sept 3, 2020 one I was using. I had to manually fetch the Apr 29, 2021 one and stick it in /lib/firmware/... to allow rpi-eeprom-config to let me enable the UART and install this newest bootloader.

closed 2021-10-05 by jtbraun