raspberrypi/firmware #1646: Stable firmware after 1.20200902 cause Home Assistant OS on RPi 4 to hard freeze

agners, 2021-11-08:
**Is this the right place for my bug report?**
This repository contains the GPU firmware used on the Raspberry Pi. This software is the closed source part of the Raspberry Pi system, it includes booting (including network booting and USB booting), low-level power and clock control, FKMS and legacy HDMI control (not full KMS that is in the linux kernel), hardware legacy codecs (MPEG2, H264 and VC1), encode hardware including the ISP (image sensor pipeline) and camera control, audio output (analogue and HDMI audio).

If you believe that the issue you are seeing is within this area, this is the right place. If not, we have other repositories for the linux kernel at [github.com/raspberrypi/linux](https://github.com/raspberrypi/linux) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
When using Raspberry Pi 4 with firmware `1.20201022` the system occasionally (typically around 1-5 days) to hard freeze. The freeze leaves no traces: The `journalctl` is empty

[    0.112709] raspberrypi-firmware soc:firmware: Attached to firmware from 2021-09-30T19:21:03, variant start
[    0.116727] raspberrypi-firmware soc:firmware: Firmware hash is 51215b4f6e3966401ecd99652a35cf1c25069113

**To reproduce**
List the steps required to reproduce the issue.

**Expected behaviour**
Add a clear and concise description of what you expected to happen.

**Actual behaviour**
Add a clear and concise description of what actually happened.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)?
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

pelwell, 2021-11-08:
I see the "still editing" comment, but questions to answer include:
* Is this definitely a firmware (as opposed to kernel) issue?
* What is the Last Known Good firmware?
* Do you have any suggestions as to how on earth we can debug such an infrequent freeze on a third-party distribution?

auanasgheps, 2021-12-05:
Hello, sorry to push, is there any progress on this?

pelwell, 2021-12-05:
Not being RPiOS reduces the chance of an investigation. The likely cause of a freeze after prolonged use is a resource leak of some kind.

agners, 2021-12-10:
> The likely cause of a freeze after prolonged use is a resource leak of some kind.

Any hints what kind of resource that could be? I assume it must be managed by the firmware somehow?

What we did meanwhile is try to pin down the offending commit on the master branch. What strikes me a bit odd is that the Sep 1 firmware on master branch ([a34f263ce6a](https://github.com/raspberrypi/firmware/commit/a34f263ce6a)) does not match the Sep 2 firmware on stable branch ([2b41f509710](https://github.com/raspberrypi/firmware/commit/2b41f509710)). However, Aug 19 on stable ([ef72c17bcaa](https://github.com/raspberrypi/firmware/commit/ef72c17bcaa)) matches Aug 19 on master ([f293685f683](https://github.com/raspberrypi/firmware/commit/f293685f683)). It seems that stable diverged a bit there. Since we know that Aug 19 on stable is fine, the community started to test each firmware from then on on the master branch.

It turned out that on master branch the first non-working firmware is the one from Aug 24 [031e8d266108](https://github.com/raspberrypi/firmware/commit/031e8d26610874cc43a5726cb07feef15c919052)...

The first change strikes me as a good candidate: The problem seems to only affect some people. Slight variations in production can cause PLLs to behave different.... 

```
firmware: pi4: allow pllb changes while running
See: #1431
```

Thoughts? Would it maybe possible to get a recent firmware with that change reverted?

pelwell, 2021-12-10:
That stable firmware was a special bug-fix release, comprising the previous firmware with a minimum number of necessary patches applied.

Narrowing a regression to a single commit is always helpful, and that change (from Aug 24 _2020_) is to an area ripe for unintentionally introducing seemingly random problems, i.e. clocking.

Any thoughts, @popcornmix ?

popcornmix, 2021-12-10:
@agners does HAOS have any different config.txt settings compared to RPiOS? (if there are any settings like arm_freq/arm_freq_min that may explain the fact HAOS has issues and RpiOS doesn't appear to).

agners, 2021-12-10:
@popcornmix we use U-Boot, hence `kernel=u-boot.bin`, but other than that its pretty much the same as RPiOS is using (see https://github.com/home-assistant/operating-system/blob/dev/buildroot-external/board/raspberrypi/config.txt). I am also not aware that `config.txt` got changed on the affected systems.

@auanasgheps can you confirm that for your case?

jcsouthworth, 2021-12-10:
I believe auanasgheps and I have identically presenting issues. my config.txt:

disable_splash=1
kernel=u-boot.bin
arm_64bit=1
dtparam=sd_poll_once=on  #removing this line has no effect.
dtoverlay=w1-gpio  #removing this line has no effect.

I have not tried changing the first three lines.

popcornmix, 2021-12-10:
Does it run with ondemand governor? Is it any more stable with performance governor (or using force_turbo=1 which should have  a similar effect).

jcsouthworth, 2021-12-10:
@popcornmix testing now

auanasgheps, 2021-12-10:
The only thing I have changed in confix.txt is GPIO configuration for [Raspbee device](https://phoscon.de/en/raspbee2), which should not have any effect on this issue.

```
# Enable GPIO RaspBee
enable_uart=1
dtoverlay=pi3-miniuart-bt
```

I am going to enable force_turbo=1 on [24-Aug-2020](https://github.com/raspberrypi/firmware/commit/031e8d26610874cc43a5726cb07feef15c919052)

@jcsouthworth what version are you testing on?

jcsouthworth, 2021-12-10:
My apologies for not listing it, I went back to using the stock HAOS FW for 6.6 which is 27-May-2021 commit 1eb8921. I can change to another version if someone thinks there is a better one to test. running fine and confirmed my clock stays at max.

# cat /sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_cur_freq
1500000

auanasgheps, 2021-12-10:
I can confirm the same clock frequency. 
Can take some days for the crash to occur.

auanasgheps, 2021-12-14:
So it's been almost four days after the change to performance governor and no crash has occurred yet. 
In the past some crash has occurred on the fourth day (most of them under 3 days), so I'll wait two more days to declare it "stable". I guess @jcsouthworth is seeing the same since he hasn't given an update either. 

If it doesn't crash, what are the next steps? Use this workaround indefinitely or it's a symptom of a problem you could help diagnose further?

pelwell, 2021-12-14:
Your findings suggest that there's a problem with changing the ARM clock frequency, which is what popcornmix suspected. At best it will be a hard-to-find software bug that causes invalid values to be written to the clock generator and that gives no further problems once fixed. It's more likely to be a timing-dependent problem with the sequence of operations that just occasionally causes a short clock pulse or something triggered by an unfortunate ratio between two clocks within the device - also arguably a software bug, but I'd prefer to think of it more as a lack of a workaround for fussy hardware.

We'd prefer users not to have to rely on using the performance governor because it is causes devices to use more power and run hotter than necessary. Whether we can pinpoint the issue and fix it properly is going to depend heavily on whether or not we can reliably reproduce the issue in a reasonable timeframe - so far that's not looking promising.

timg236, 2021-12-14:
A fix to the change in behaviour in 1.20201022 was made in this commit
https://github.com/raspberrypi/firmware/commit/aa57c6755b9a08092f79ff7e01961538621dc350#diff-4509beb0ab401d71fa4a5cd94a55c9a74f13332776ae4019c5bfc4c2005157ff

This changes how the ARM PLL is managed. I think it's only in rpi-update master (maybe oldstable) but you'll need the following release or newer.
https://github.com/raspberrypi/firmware/commit/1a0297bfbf7c970f657c11edc42b22a671170d18

The behavioural change was 'theoretical' based upon advice from Broadcom about how this PLL should be managed. We've not seen direct evidence of the failure but it might only affect a narrow window of chips / clock changes.

popcornmix, 2021-12-14:
As Tim suggests, please test with rpi-update firmware and return to the ondemand governor and report if this helps.

jcsouthworth, 2021-12-14:
As @auanasgheps suspected, I am at 4 days of uptime suggesting stability on 27-May-2021 stable 1eb8921 with force_turbo=1. I will switch to 1a0297b removing force_turbo and report back. Thank you RPi team for the help on this!

auanasgheps, 2021-12-14:
> but it might only affect a narrow window of chips / clock changes.

The issue is "narrow" (85 people following the HASSOS issue), but affects every RPi4 model regardless of RAM SKU or production date. 
We still don't know how to replicate it other than using the RPi4s that already crash using FW after 02-Sep-20, on these boards we can always replicate the crash.

I'll come back in a couple of days to confirm if the performance governor at least works, but we need the evidence from  @jcsouthworth too.

pelwell, 2021-12-14:
I think your time would be better spent testing https://github.com/raspberrypi/firmware/commit/1a0297bfbf7c970f657c11edc42b22a671170d18, but the choice is yours.

auanasgheps, 2021-12-14:
Can you please clarify what we should do? 
Should we use start4.elf/fixup4.dat from 1a0297b and go back to ondemand?

I'll do it in a couple of days.

popcornmix, 2021-12-14:
> Should we use start4.elf/fixup4.dat from 1a0297b and go back to ondemand?

Yes. There is a potential clock fix in that commit and we'd like to know if it does fix your issue.

jcsouthworth, 2021-12-17:
I am at 3 days, 2 hours of uptime with no issues on HA OS 6.6 with firmware 1a0297b. I am going out of town for a bit so I won't be able to diagnose much, but I can verify if it remains up with a VPN. I'd estimate > 75% of my crashes associated with this issue have been < 3 days, so this is looking promising!

auanasgheps, 2021-12-20:
I am almost at 5 days. Stable for me too! 

I'd say this is the fix for our problem. We'll ask more people at HomeAssistantOS to confirm but I'm sure we'll be fine.

I don't know if you made the fixt because of this GitHub issue, but you deserve my gratitude since this problem has affected me personally for 6 months.
Please tell me how can I support/donate and I'll send a small contribution to the team or the project.

Thank you very much!

@agners are you synced up? Is there anything else you need to implement this update in the next HASSOS release?

jcsouthworth, 2021-12-20:
I agree with @auanasgheps - looks stable and looking for the preferred way to support you all! Thank you all very much for your help!

one last clarification for the RPi team - do you know if this fix impacts only RPi4 or if it impacts other RPi variants?

timg236, 2021-12-20:
> I agree with @auanasgheps - the new FW looks stable and I am looking for the preferred way to support you all! Thank you all very much for your help!
> 
> one last clarification for the RPi team - do you know if this fix impacts only RPi4 or if it impacts other RPi variants?

It’s specific to the BCM2711, therefore, Pi4, Pi400 and Compute Module 4

pelwell, 2021-12-20:
To answer the other question, the best way to show your appreciation is to support other RPi users when you can.

agners, 2021-12-21:
Thanks to all who contributed to this thread, really appreciated!

@auanasgheps currently it seems that the fix is only on the master branch. For a stable HAOS release, I'll wait until the firmware makes it to the stable branch.

popcornmix, 2021-12-21:
The stable firmware branch (updated Dec 1st) contains the commit "clock-2711: Keep PLLB in VCO high range", so should be good.

agners, 2021-12-23:
@popcornmix not for me, last commit on https://github.com/raspberrypi/firmware/commits/stable is Nov 18, 2021 as of today :monocle_face:

timg236, 2021-12-23:
I think it’s in oldstable Buster/legacy but stable for bullseye isn’t updated yet

agners, 2021-12-23:
Ok, so oldstable Buster/legacy is using firmwares from the master branch then? For Home Assistant OS we rely on the `stable` branch of this repository.

timg236, 2021-12-23:
Oldstable was forked after Bullseye was released because Bullseye was frozen. I think stable will get refreshed from master in the new year to pull in this plus other fixes.

popcornmix, 2021-12-27:
Correct. We updated oldstable for the legacy Buster release.
That update contains all the existing stable commits plus some extra ones (including the VCO one that you require).
I've updated stable to match that, so that may be a sensible point for you to update to.

agners, 2021-12-27:
Oh I see, there is also `oldstable` in this repo. Ok, all I wanted to avoid is using a firmware build not already being tested by RPi OS users. But being on oldstable means it got pushed many users of the RPi OS Buster release, so that was good enough as well.

FWIW, I usually rely on the tags (https://github.com/raspberrypi/firmware/tags), those also trigger a GitHub notification :smile:

agners, 2022-01-03:
HAOS 7.1 with firmware `oldstable_20211201` has now been on our beta channel for several days and got promoted to stable today. Multiple people confirmed that the new firmware/the new OS release fixed the stability issues. So we can consider this issue fixed! Thanks for your help to get this resolved, very much appreciated!

closed 2022-01-03 by agners