raspberrypi/firmware #1648: RPi4B/Kodi goes to black screen with vc4-kms-v3d after commit 19272ccd69049aaf42c78a235a0bf37dbabd5ea7

mkreisl, 2021-11-09:
This commit leads to a black screen (TV says no signal) after Kodi is started.
This occurs with all tested kernel versions (5.10.60 - 5.10.75 and 5.14.14) and with both Kodi 19 and Kodi 20.

In dmesg the messages appear:
```
[  113.892587] [drm:drm_crtc_commit_wait [drm]] *ERROR* flip_done timed out
[  113.892713] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] commit wait timed out
[  124.128612] [drm:drm_crtc_commit_wait [drm]] *ERROR* flip_done timed out
[  124.128920] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:91:plane-6] commit wait timed out
[  134.368625] [drm:drm_crtc_commit_wait [drm]] *ERROR* flip_done timed out
[  134.368995] vc4-drm gpu: [drm] *ERROR* Timed out waiting for commit
[  144.608695] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[  154.848659] [drm:drm_crtc_commit_wait [drm]] *ERROR* flip_done timed out
[  154.849120] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] commit wait timed out
[  165.088657] [drm:drm_crtc_commit_wait [drm]] *ERROR* flip_done timed out
[  165.089117] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:91:plane-6] commit wait timed out
[  175.328649] [drm:drm_crtc_commit_wait [drm]] *ERROR* flip_done timed out
[  175.329113] vc4-drm gpu: [drm] *ERROR* Timed out waiting for commit
[  185.568601] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[  188.481021] init: passwd-change main process (5796) terminated with status 1
[  190.432635] cec-vc4: message ff 84 20 00 06 timed out
[  192.736595] cec-vc4: message ff 87 00 15 82 timed out
[  195.040635] cec-vc4: message f0 timed out
[  197.344605] cec-vc4: message f0 timed out
[  199.136617] [drm:drm_crtc_commit_wait [drm]] *ERROR* flip_done timed out
[  199.136926] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] commit wait timed out
[  199.648658] cec-vc4: message f0 timed out
[  201.952655] cec-vc4: message f0 timed out
[  204.256584] cec-vc4: message 11 timed out
[  206.560662] cec-vc4: message 11 timed out
[  208.864603] cec-vc4: message 11 timed out
[  209.376644] [drm:drm_crtc_commit_wait [drm]] *ERROR* flip_done timed out
[  209.376953] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:91:plane-6] commit wait timed out
[  211.168584] cec-vc4: message 11 timed out
[  213.472618] cec-vc4: message 22 timed out
[  215.776619] cec-vc4: message 22 timed out
[  218.080648] cec-vc4: message 99 timed out
[  219.616604] [drm:drm_crtc_commit_wait [drm]] *ERROR* flip_done timed out
[  219.616874] vc4-drm gpu: [drm] *ERROR* Timed out waiting for commit
[  220.384652] cec-vc4: message 99 timed out
[  222.688572] cec-vc4: message ff 84 20 00 01 timed out
[  224.992583] cec-vc4: message ff 87 00 15 82 timed out
[  227.296649] cec-vc4: message 10 timed out
[  229.600630] cec-vc4: message 10 timed out
[  229.856628] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[  231.904602] cec-vc4: message 10 timed out
[  234.208649] cec-vc4: message 10 timed out
```

If I **use vc4-fkms-v3d** (which I have always used so far) this problem no longer occurs

This means the firmware files that are present in /boot, not the kernels itself

Until one commit before the one above, the problem does not occur, so it must definitely be the commit **arm_loader: Add message to release firmware framebuffer 19272ccd69049aaf42c78a235a0bf37dbabd5ea7**

popcornmix, 2021-11-10:
That commit did have a later fix:
```
1e494f150046bd04331ab0b19e4e5322eb7b7773 firmware: arm_loader: Allow hvs interrupt during SET_NOTIFY_DISPLAY_DONE
```
Does the problem still occur with that? If it does can you post your edid, and I'll see it I can reproduce.

mkreisl, 2021-11-10:
How much later is the fix? Of course I also tried it initially about 2-3 days ago with the latest version with the same error

popcornmix, 2021-11-10:
I'd like the raw edid. e.g. paste output of:
```
base64 /sys/devices/platform/gpu/drm/card1/card1-HDMI-A-1/edid
```

mkreisl, 2021-11-10:
No problem
```
AP///////wAeVUhEAQAAACUWAQOAeUR4KuxroFZHnCUQSEonzwABAQEBgUCBgIGPiwABAalAAjqA
0HI4LUAQLEWAoFoAAAAeAjqAGHE4LUBYLEUAoFoAAAAeAAAA/QAxTA9RCAAKICAgICAgAAAA/ABH
UlVORElHIFdVWEdBAcMCAytyUBQFEwQSAxECFgcVBgGfkCYmFQdQCQcHbgMMACAAgC0g0AQBQQc/
GyFQoFEAHjDQiGUEoFoAAAAcDh8AgFEAHjDAgEcEoFoAAAAcaCmg0FGEIjDomJYEoJAAAAAcjy94
0FEaJ0DokAQIoJAAAAAcAAAAAAAAAAAAAAAA9A==
```

gradientskier, 2021-11-15:
Can this be related?

https://beta.raspberrypi.org/forums/viewtopic.php?p=1938658#p1938658

pelwell, 2021-11-15:
What effect does chasing changing `dtoverlay=vc4-kms-v3d` to `dtoverlay=vc4-fkms-v3d` have?

gradientskier, 2021-11-15:
> What effect does changing `dtoverlay=vc4-kms-v3d` to `dtoverlay=vc4-fkms-v3d` have?

It works!

mkreisl, 2021-11-16:
@popcornmix I tried to locate the problem further yesterday and found what I was looking for.
I still use the old commits in the Kodi sources which use the PI:HDMI audio device. and this initialize now fails. But this is already heavy that this breaks the hdmi output completely, so would still rate it as a bug in the firmware

popcornmix, 2021-11-16:
Using PI:HDMI (i.e. firmware driving hdmi audio) together with vc4-kms-v3d (i.e. arm driving hdmi hardware) is not a supported configuration. Both ends can access hdmi registers without knowledge of what the other side is doing, and crashes are not unexpected.

We should perhaps try harder to reject use of the firmware hdmi audio driver when kms is enabled.

mkreisl, 2021-11-16:
> Using PI:HDMI (i.e. firmware driving hdmi audio) together with vc4-kms-v3d (i.e. arm driving hdmi hardware) is not a supported configuration. Both ends can access hdmi registers without knowledge of [...]

I know that and I have also read several posts about how this should be prevented.
But unfortunately that does not seem to work properly yet

Brunnis, 2021-11-29:
Maybe the issue I'm seeing is related to this: https://forums.raspberrypi.com/viewtopic.php?p=1944172#p1943655

I am seeing this on Lakka, which is based on Libre-ELEC. I did not have this issue on 5.10.39, but had it sporadically (maybe 40% of boots) on 5.10.63 and 100% of the time on 5.15.0. So far, I've only seen it happen on my rev 1.4 Pi 4 and only on my 4K TV (LG OLED65CX). Changing from KMS to FKMS makes the issue disappear. See the linked post for full description and log printouts.

Brunnis, 2021-11-30:
Sorry for slightly misleading information earlier. The sporadic nature of this issue and my lack of time for testing led me to make some incorrect conclusions. The rev 1.1 and the rev 1.4 boards seem about equally affected. Both have major issues with 4K when testing with the 5.10.63 and 5.15.0 kernels on Lakka. 5.10.39 has no issues on any of the boards, whether running 4K or 1080P. Below are the full test results. Each config was booted five times and the result recorded. Each boot was a cold boot.

[image]

popcornmix, 2021-11-30:
@Brunnis are you able to narrow it down further?
Are you purely changing the kernel between tests (and not firmware or other parts of system)?

Brunnis, 2021-11-30:
Actually, I am just switching between the Lakka images, so not changing the system from defaults in any way so far. Can you suggest any steps/commands that would help in narrowing it down?

popcornmix, 2021-11-30:
Easiest to reproduce with RPiOS bullseye (then we'll know what you are running and have a better chance of reproducing).
If it's a kernel issue you should see the same issue on RPiOS.

From there you can identify the exact update which caused this. See:
https://github.com/raspberrypi/rpi-firmware/commits/master

If you click on each commit the end of the url contains a git hash. Run
```sudo rpi-update <hash>```
to revert back to that version. Report the first version with the error.

Brunnis, 2021-11-30:
I've ran a bunch of reboots using Bullseye lite 32-bit, but cannot reproduce the issue there. Lakka is 64-bit, but both use the same kernel (5.10.63). The output of vcgencmd version is:

**Lakka 3.6 64-bit:**
Oct 29 2021 10:48:24
Copyright (c) 2012 Broadcom
version b8a114e5a9877e91ca8f26d1a5ce904b2ad3cf13 (clean) (release) (start_x)

**Bullseye 32-bit lite:**
Oct 29 2021 10:47:33
Copyright (c) 2012 Broadcom
version b8a114e5a9877e91ca8f26d1a5ce904b2ad3cf13 (clean) (release) (start)

Last week I tried Bullseye 64-bit desktop and that also seemed to work fine. Don't really know how to proceed, to be honest.

popcornmix, 2021-11-30:
It seems lakka is using start_x. You can add start_x=1 to bullseye config.txt just in case that makes a difference (I wouldn't expect it to).

Are they both using `dtoverlay=vc4-kms-v3d` driver?

But if the issue doesn't occur with RPiOS bullseye, then there's not much we can do.
We don't really know what the differences are with lakka.

pelwell, 2021-11-30:
It's probably worth checking that Lakka includes overlay_map.dtb in /boot (or wherever they mount the boot partition). You can also see if `sudo vcdbg log msg` shows any errors from the firmware (if the utility is installed).

Brunnis, 2021-11-30:
@popcornmix  start_x=1 made no difference. Both are indeed using vc4-kms-v3d.

@pelwell Lakka does not include overlay_map.dtb. Is that an issue?

pelwell, 2021-11-30:
It would mean the usual config.txt file is loading the Pi 0-3 version of the kms overlay. Try changing the line to `dtoverlay=vc4-kms-v3d-pi4`.

pelwell, 2021-11-30:
Ah - I should have said to check in the `overlays` subdirectory.

Brunnis, 2021-12-01:
Ahh, Lakka is actually using dtoverlay=vc4-kms-v3d-pi4 by default. Sorry for overlooking that earlier.

Regarding the overlays subdirectory, that was actually where I looked, since I figured that would be the correct place. 😊

I'll let you know if I get anywhere with further testing.

Brunnis, 2021-12-06:
I found a successful workaround for the no HDMI issue that at least works on Lakka 3.6 with the 5.10.63 kernel: Add "video=HDMI-A-1:1920x1080M@60" to cmdline.txt. So, it seems something sporadically fails (more than 50% of boots) when trying to use 4K 30Hz. Don't know yet if this particularly affects the LG OLED CX, as I don't have any more 4K TVs around. I actually belive I tested this workaround early on in my testing of kernel 5.15.0 and it didn't help there, but I need to confirm that.

Doubtful if this is related to the original issue described here, but I thought I'd let you know of this slight progress.

david-barbion, 2021-12-31:
I also have the black screen problem. However, when using a plain micro hdmi to hdmi cable all is fine. When using a micro hdmi adapter and a standard hdmi cable, it fails with a black screen.

Tested on Raspbian bullseye and Recalbox 8.0.0 (both of them use kms by default)

Going back to fkms makes both cables work, so this does not look like a cable problem.

popcornmix, 2022-01-01:
> However, when using a plain micro hdmi to hdmi cable all is fine. When using a micro hdmi adapter and a standard hdmi cable, it fails with a black screen.

Sounds a bit like the adapter is breaking something (possibly reading the edid).
In each case can you show output of one of:
```
base64 /sys/devices/platform/gpu/drm/card0/card0-HDMI-A-1/edid
base64 /sys/devices/platform/gpu/drm/card1/card1-HDMI-A-1/edid
```
(one will report missing file, the other should show something).

david-barbion, 2022-01-01:
With the adapter plugged-in, edid is empty
```
# cat /sys/devices/platform/gpu/drm/card1/card1-HDMI-A-1/edid
# cat /sys/devices/platform/gpu/drm/card1/card1-HDMI-A-2/edid
# 
```
(card-0-HDMI-A-1 does not exist)

`vcdbg log msg` reports errors when reading edid. I've copied the log output here https://pastebin.com/gsEZUues

The issue appeared between kernel 5.10.13 and 5.10.79.

Also, at power-up, a signal is sent to the TV until vc4 (drm) is loaded . This is even more visible with raspbian where one can see the linux system loading and at some point (when vc4 is loaded, I think), the screen goes black and no signal is sent anymore.

6by9, 2022-01-02:
I'd guess that the hotplug line isn't following the HDMI spec.
The firmware tries reading the EDID regardless of the hotplug status. The kernel doesn't as it follows the HDMI spec more closely, and if hotplug isn't asserted then it won't read the EDID.

digitalLumberjack, 2022-01-15:
Hello @david-barbion @popcornmix  o/

We still have many users with the issue on @recalbox on the latest kernel + firmware. Could we help anyway making other tests for you ?

popcornmix, 2022-01-15:
If you can't read the edid using a specific adapter then the adapter is faulty (most likely no not connecting the hotplug detect line).
You can try adding `video=:D` to end of cmdline.txt which may override it.

digitalLumberjack, 2022-01-15:
Is there something that could explain that it works flawlessly on old kernels with the same adapter ?

popcornmix, 2022-01-15:
There was a bug in the kernel driver where we would read the edid without hotplug being asserted which is against spec (and caused issues with re-enabling scrambling after switching hdmi input).

david-barbion, 2022-01-15:
Finally, the video option solved this issue. Thank you @popcornmix.

ModMike, 2022-01-29:
So what was the solution?

digitalLumberjack, 2022-01-29:
@popcornmix we just released the new Recalbox 8.0.1 with the `video=HDMI-A-1:D video=HDMI-A-2:D` in cmldline.txt that seems to fix the issue of the black screen where people had black screens.

`video=:D` did not fix the issue so we wanted to try the working one.

But it cause other issues, as it seems to avoid the kernel to take any arguments of config.txt (hdmi_mode in this case)

Is there a documentation or source code anywhere that could let us understand how this video=XXX is parsed and used ?

popcornmix, 2022-01-29:
`video=:D` should not be applied universally as a default. It is generally only needed with faulty hardware (e.g. a hdmi cable that doesn't connect the hotplug line, or an Argon case which are very cause a lot of unreliability issues for hdmi).

Info on the `video=` setting here: https://www.kernel.org/doc/html/v5.10/fb/modedb.html

digitalLumberjack, 2022-01-29:
Thank you @popcornmix for the informations. I'm sorry but I must admit we don't know what to think about this issue.

100% of the hdmi worked well.
Since the bug fix we talked about, many people had a black screen.

Does that mean that a certain percentage of hdmi cables/tv were working by chance ? Cannot it be considered as a feature and not a bug then ?

I may be wrong but I think several distributions are touched by this bug, and right now we are even thinking about rolling back to the old kernel and firmware version as we don't know what to do.

Please help me not doing this 😅

ModMike, 2022-01-29:
@digitalLumberjack I tried it and it doesn't work. I did have a bridged solder point on HDMI which I fixed and validated with my my multimeter.

Am I supposed to put the whole statement video=HDMI-A-1:D video=HDMI-A-2:D in the cmdline.txt or just one or the other?

ModMike, 2022-01-29:
@popcornmix I understand how much disdain you have for the argon and others like it but I do "love" mine, specifically for the ssd functionality. Software work arounds are par for the course in development.

digitalLumberjack, 2022-01-29:
> @digitalLumberjack I tried it and it doesn't work. I did have a bridged solder point on HDMI which I fixed and validated with my my multimeter.
> 
> Am I supposed to put the whole statement video=HDMI-A-1:D video=HDMI-A-2:D in the cmdline.txt or just one or the other?

We put that in cmdline.txt at the end of the line. Don't know if it fixes anything for the argon though.

ModMike, 2022-01-29:
@digitalLumberjack Thanks, I appreciate you trying to help. Will try outside of case and report back.

pelwell, 2022-01-29:
"disdain" is a strong word that doesn't apply here - saying that that "cause a lot of unreliability issues for hdmi" is just a statement of fact.

ModMike, 2022-01-29:
Test results:

1. Adding video=HDMI-A-1:D video=HDMI-A-2:D to cmdline.txt did not work for Argon
2.  suggested by @popcornmix, system works fine when removed from Argon case
3. Leaving in video=HDMI-A-1:D video=HDMI-A-2:D in cmdline.txt when out of Argon case causes washed out video and seems to affect audio interface choices

Possible issue: Did Argon not connect the hot plug detect and purposely ground it?

bkg2k, 2022-01-29:
> "disdain" is a strong word that doesn't apply here - saying that they "cause a lot of unreliability issues for hdmi" is just a statement of fact.

None of our users reported faultly Argon case until then. Maybe they were all luky.
Unfortunately, now, there are plenty of users reporting blackscreen. The situation is even worse with the `video=...:D` as @digitalLumberjack said.

The situation for end users is a lot worse since this "fix". Our users are not technical people at all. They are mostly players. Lambda users don't care or even know if they have "fautly hardware". The only things they see is "It's no longer working".
Asking lamba users to add some cmdline option only when it does'nt work is not an option.
The firmware should work out of the box, as before, for as many hardware as possible, faultly or not.

pelwell, 2022-01-29:
@popcornmix has spent years fixing the corner cases in the old firmware-based HDMI driver. It will take a while for the remaining issues with the new driver to be resolved, but they will.

ModMike, 2022-01-29:
I agree with @bkg2k 's sentiment but also understand @popcornmix 's desire to do things the right way but as I've said before, not everything is done the "right" way and this change has trashed a lot of systems. Electronics are littered with corrections and or errors that we usually find a way around.

I hope @popcornmix does find a way to fix things so Libreelec is compatible with more devices. I think the am logic boxes are getting expensive and hard to find, which is why I switched to the RPI 2 years ago.

This thread is getting dangerously close to a flame war so I will let @popcornmix get on with it and no longer comment. Furthermore I am offering a $40 donation bounty when this issue is fixed. I know it's not much but I think we should ALL encourage the Libreelec team.

ModMike, 2022-01-29:
Asking lamba users to add some cmdline option only when it does'nt work is not an option. 

Did I miss something? What fix is this?

ModMike, 2022-01-29:
> > What effect does changing `dtoverlay=vc4-kms-v3d` to `dtoverlay=vc4-fkms-v3d` have?
> 
> It works for me! Is this the official solution, or there will be an update fixing this issue?

So this gives me video on the Argon but I lose audio outputs. Is there a way to fix it? I tried both HDMI ports.

bkg2k, 2022-01-29:
> This thread is getting dangerously close to a flame war so I will let @popcornmix get on with it and no longer comment. Furthermore I am offering a $40 donation bounty when this issue is fixed. I know it's not much but I think we should ALL encourage the Libreelec team. I would encourage others in this thread to add to the bounty, ESPECIALLY if you are a vendor. Fair is fair.

No flame war here. We're in a serious thread.
The original fix seems legit and I don't blame @popcornmix or anyone else for it.
I'm just here to alert that the fix seems to have severe side-effects that affect many people. Most likely more people than those affected by the original bug.
So for me, the balance pros/cons is on the wrong side, and maybe a rollback should be considered until further investigations.

ModMike, 2022-01-29:
@bkg2k Are you talking about changing dtoverlay=vc4-kms-v3d to dtoverlay=vc4-fkms-v3d? I lost all audio capability when I did that but I DO get Argon video.

ModMike, 2022-01-29:
@popcornmix I noticed you committed a change to the RPI bistro last night to address no sound with fems. Did this make it into the libreelec nightlies?

ModMike, 2022-01-29:
@popcornmix I noticed you committed a change to the RPI bistro last night to address no sound with fkms. Did this make it into the libreelec nightlies?

popcornmix, 2022-01-31:
> @popcornmix I noticed you committed a change to the RPI GitHub last night to to fix the fkms no sound issue. Did this make it into the libreelec nightlies?

I don't believe there has been any recent change like this. You may have been seeing a spurious github notification that occurs when rebasing newer kernel trees - if a commit references an issue you get a notification on each rebase.

Brunnis, 2022-03-02:
As I've mentioned previously, I worked around my issues with black screen during boot on my 4K TV by setting 1920x1080 via the kernel command line. This was using Lakka 3.6 and solved the problem 100 % for me. Now Lakka 3.7 has finally been released and they've determined that the issue was caused by more recent Pi firmware. A downgrade to 1.20210831 has apparently solved the issue (as verified by others).

See release notes mentioning this here: https://www.lakka.tv/articles/2022/03/01/lakka-3.7/

@ToKe79 may know more, since he implemented this change.

rijotech, 2022-07-25:
> > > What effect does changing `dtoverlay=vc4-kms-v3d` to `dtoverlay=vc4-fkms-v3d` have?
> > 
> > 
> > It works for me! Is this the official solution, or there will be an update fixing this issue?
> 
> So changing this in the flash/distro.config gives me video on the Argon but I lose audio outputs. Is there a way to fix it? I tried both HDMI ports.
> 
> This is what I have now in distro.cfg: arm_64bit=1 kernel=kernel.img dtoverlay=vc4-fkms-v3d,cma-512 dtoverlay=rpivid-v4l2 dtoverlay= disable_overscan=1 disable_fw_kms_setup=1
> 
> and cmdline.txt:
> 
> boot=UUID=2801-4448 disk=UUID=6f37a636-034d-4f7d-9eeb-a1453c0cf6ba quiet snd-bcm2835.enable_compat_alsa=1

> So changing this in the flash/distro.config gives me video on the Argon but I lose audio outputs. Is there a way to fix it? I tried both HDMI ports.
> 
> This is what I have now in distro.cfg: arm_64bit=1 kernel=kernel.img dtoverlay=vc4-fkms-v3d,cma-512 dtoverlay=rpivid-v4l2 dtoverlay= disable_overscan=1 disable_fw_kms_setup=1
> 
> and cmdline.txt:
> 
> boot=UUID=2801-4448 disk=UUID=6f37a636-034d-4f7d-9eeb-a1453c0cf6ba quiet snd-bcm2835.enable_compat_alsa=1

Where is that file location `flash/distro.config`? I'm facing the same problem.

popcornmix, 2022-07-25:
> Where is that file location flash/distro.config? I'm facing the same problem.

`/flash/distro.cfg` is a libreelec specific file (they just include this file from config.txt for their own default settings).
On another distribution `/boot/config.txt` would be the file to edit.

popcornmix, 2023-01-24:
I believe these issues are all related to being unable to read the edid.
This is usually down to a faulty hdmi cable, or less likely a faulty display.

There are workarounds for this, using manual configuration, but fixing the hardware is the better solution.

Adding something like `video=HDMI-A-1:1920x1080@60D` to cmdline.txt (end of existing line, assuming that is a supported mode) is likely to get a picture up.

fuchsmich, 2023-01-31:
Listen to @popcornmix 👆👆 !!

My RPI4 with kodi on ArchARM had symptoms described here and in other locations. It used to work. Last year it started having troubles. After trying a lot of things with fkms in config.txt, passing edid.dat to the kernel and what not I finally bought a new HDMI cable (DIGITUS HDMI/A ST-Micro HDMI ST 2m) replacing a Micro-HDMI Adapter and a Standard HDMI Cable and everything works as indented.

Thank you for your patience with us fools.

geiseri, 2023-08-27:
I had the same experience as @fuchsmich.  I had my RPI in storage for a few months, and when I powered it back on I got this issue.  After tons of messing around, I purchased a brand new HDMI cable, and everything returned to working.   So I guess it was a layer-1 issue?

closed 2025-02-14 by popcornmix