raspberrypi/firmware #1649: AutoDeploy script

xevilmaxx, 2021-11-10:
Hi,
**It's not a bug report.**
Just wanted to say that it would be nice to have some king of script to auto-install latest firmware at least on raspbian os.
So here is the one i created for myself, maybe you will find some use of it.
Obviously there is plenty of room for improvement, but it works for me.
At least on **RaspiOs arm64** bit (probably for standard raspbian is identical).

```bash
#!/bin/bash

if [ "$EUID" -ne 0 ]
  then echo "Please run as root"
  exit
fi

echo "Execute remove sources once done, if you are sure, backups also may be dropped"
options=("Upgrade" "Restore_Backup" "Remove_Backup" "Remove_Sources")
select opt in "${options[@]}"
do
    case $opt in
        'Upgrade')
                
                
                
                echo "Deleting old files"
                #select directory in which to download
                cd /home/pi/Downloads
                #remove previous download
                rm -rfv ./firmware
                #clone fresh repo
                echo "Cloning fresh repo"
                git clone --depth=1 https://github.com/raspberrypi/firmware

                #Predispose backup folders if not already exists
                echo "Creating Bak directories"
                sudo mkdir -p /BakBoot
                sudo mkdir -p /BakBoot/Boot
                sudo mkdir -p /BakBoot/OptVc
                sudo mkdir -p /BakBoot/LibModules

                #Do backup of things we are going to replace / update
                echo "Bak boot"
                sudo cp -fR /boot/ /BakBoot/Boot
                echo "Bak opt/vc"
                sudo cp -fR /opt/vc/ /BakBoot/OptVc
                echo "Bak lib/modules"
                sudo cp -fR /lib/modules/ /BakBoot/LibModules

                #Start Upgrade
                echo "Replacing boot"
                sudo cp -fr /home/pi/Downloads/firmware/boot/* /boot/
                echo "Replacing lib/modules"
                sudo cp -fr /home/pi/Downloads/firmware/modules/* /lib/modules/

                #Maybe facoltative, but i will keep it (needed for GPU acceleration in some cases)
                echo "Replacing opt/vc"
                sudo cp -fr /home/pi/Downloads/firmware/opt/* /opt/

                echo "Anyway only /boot/kernel8.img (other kernel* are useless for our case) and lib/modules/5.10.78-v8+ are needed for rpi 3b+ or 4, others are facoltative"
                echo " "
                echo "So you can remove surplus /boot/kernel.img and /lib/modules/ if you dont mind"
                echo " "
                echo "Run> sudo rm -rf ./BakBoot after reboot if all works (in order to free space on SD), otherwise reverse changes"

                
                
                
                exit
                ;;
        'Restore_Backup')
                
                
                
                #Start Restore
                echo "Restoring Boot dir"
                sudo cp -fr /BakBoot/Boot/* /
                
                echo "Restoring Modules dir"
                sudo cp -fr /BakBoot/LibModules/* /lib/
                
                echo "Restoring opt/vc"
                sudo cp -fr /BakBoot/OptVc/* /opt/
                
                
                exit
                ;;
        'Remove_Backup')
                
                
                
                #Start removing
                echo "Removing /BakBoot folder"
                sudo rm -rf /BakBoot
                
                
                exit
                ;;
        'Remove_Sources')
                
                
                
                #Start removing
                echo "Removing /home/pi/Downloads/firmware folder"
                sudo rm -rf /home/pi/Downloads/firmware
                
                
                exit
                ;;
        *)
                echo
                echo "!!!Unhandled Case!!!"
                echo
                exit 1
                ;;
    esac
done

exit 0

```

pelwell, 2021-11-10:
[`rpi-update`](https://github.com/raspberrypi/rpi-update)?

xevilmaxx, 2021-11-10:
Oh, nevermind, just accidentally found this:
https://github.com/raspberrypi/rpi-update
which i assume is almost same

closed 2021-11-10 by xevilmaxx