raspberrypi/firmware #1650: Video modes capture half the expected resolution then upscale it

Malvineous, 2021-11-15:
I have noticed that video captured with both V4L2 in raw YUV mode, and using `raspivid` appear to return video taken at a lower resolution and then upscaled to the requested resolution.  It seems to be halving the requested resolution and then later doubling it, causing the video to become pixellated.  Sort of like the 2x2 binning process except it's happening one too many times.

For example, capturing video at 1296x972 returns a 1296x972 video as expected, but it only contains the detail of a 640x480 image, upscaled to 1296x972.  In order to get a 1296x972 video with 1296x972 detail, one must override the mode selection to choose 2592x1944 mode, so that after the extra resolution halving step, one ends up with true 1296x972 video.

# Steps to reproduce

1. Take some still photos which select the correct sensor mode:
```
$ raspistill -w 640 -h 480 -o raspistill.640x480.jpg
$ raspistill -w 1296 -h 972 -o raspistill.1296x972.jpg
$ raspistill -w 2592 -h 1944 -o raspistill.2592x1944.jpg
```
2. Take some video and compare the detail against the still images:
```
$ raspivid -md 0 -w 1296 -h 972 -o raspivid.640x480.h264
```
The modes given in the `-md` option appear to behave like this:

<table>
<tr>
<td>-md option</td>
<td>Expected sensor resolution</td>
<td>Apparent sensor resolution</td>
<td>Comment</td>
</tr>
<tr>
<td>0</td>
<td>auto</td>
<td>640x480</td>
<td>Seems to select mode 4 as expected, but mode 4 appears to be wrong</td>
</tr>
<tr>
<td>1</td>
<td>1920x1080</td>
<td>1296x972? Could be 2592x1944</td>
<td>Video zoomed in a lot, seems to be a crop from a larger sensor resolution</td>
</tr>
<tr>
<td>2</td>
<td>2592x1944</td>
<td>1296x972</td>
<td>full FoV, much more detail, this is what I expected to get for a 1296x972 video capture, it matches the detail of the 1296x972 still image</td>
</tr>
<tr>
<td>3</td>
<td>2592x1944</td>
<td>1296x972</td>
<td>had to use -fps 1 (as expected), same as <tt>-md 2</tt></td>
</tr>
<tr>
<td>4</td>
<td>1296x972</td>
<td>640x480</td>
<td>same detail as 640x480 still image</td>
</tr>
<tr>
<td>5</td>
<td>1296x730</td>
<td>640x480</td>
<td>video zoomed in, but very pixellated, same detail as 640x480</td>
</tr>
<tr>
<td>6</td>
<td>640x480</td>
<td>320x200</td>
<td>*really* pixellated, looks like 320x200, purple tint</td>
</tr>
<tr>
<td>7</td>
<td>640x480</td>
<td>320x200</td>
<td>same as <tt>-md 6</tt> but normal colours</td>
</tr>
</table>

It looks like the `auto` option is working correctly, but all the actual modes are incorrectly halving the resolution for some reason, but only in video mode, not in still mode.

What tells me it's a bug is that if I use `-md 1` I get a true 1296x972 capture, and `-md 7` which should select a 640x480 capture appears to be a 320x200 one.

**System**

* Which model of Raspberry Pi? Pi1B+, Pi3B, Pi3B+ all with version 1 of the Pi camera
* Which OS and version (`cat /etc/rpi-issue`)? Arch Linux ARMv7
* Which firmware version (`vcgencmd version`)?
```
Nov  5 2021 18:17:26 
Copyright (c) 2012 Broadcom
version b99321a10f46c9bd662fc87998419c5d6974927a (clean) (release) (start_db)
```
* Which kernel version (`uname -a`)? `Linux 5.10.77-1-raspberrypi-ARCH #1 SMP Fri Nov 5 17:26:12 UTC 2021 armv7l GNU/Linux`

6by9, 2021-11-15:
Stills captures ALWAYS use the full resolution mode unless overridden by ```-md N```. All your raspistill lines will therefore be reading out the full 5MPix from the sensor and downscaling. If you add the ```-r``` to add the raw Bayer frame to the JPEG, you'll see they all come out largely the same size, and using https://github.com/6by9/dcraw/ or similar to process the raw will give you a 5MPix output file.

Malvineous, 2021-11-15:
Fair enough - but if I take a still at 640x480 and then enlarge it to 1296x972, I get a blurry/pixellated 1296x972 image as expected, however this matches exactly the look of the video when capturing at 1296x972.  Why does the 1296x972 video only take data off the sensor at 640x480 and then scale it up?  From reading the docs it looks like it should use the 2x2 binning mode to read data off the sensor at the exact 1296x972 resolution, yet that doesn't seem to be what happens.  It seems to use the equivalent of a 4x4 binning mode to get a 640x480 image, then enlarges the frame up to 1296x972 by duplicating pixels and that's what you end up with.

You can see this by downscaling a captured 1296x972 video to 640x480 and then upscaling it back to 1296x972 again and it looks exactly the same, there's no loss of detail at all.

However if you capture at 1296x972 with `-md 2` and do the same exercise, there is a very noticeable loss of detail when you downscale to 640x480 and back to 1296x972.

6by9, 2021-11-15:
Run ```vcgencmd set_logging level=0x400```
Run ``` raspivid -t 200 -md 4 -w 1296 -h 972 -o md4.h264```
Run ```sudo vcdbg log msg |& grep switch``` and I get the output:
```
1094090.916: camplus_switch_mode: enter flags=0x0, user_mode: name=(null), width=1296, height=972, xfm=0, user_mode->flags=0x1
1094090.977: camplus_switch_mode [0]: request_mode: name=2x2_BINNED, width=1296, height=972, xfm=3
1094133.250: camplus_switch_mode: exit, mode used 2x2_BINNED
1094204.785: camplus_switch_mode: enter flags=0x0, user_mode: name=(null), width=1296, height=972, xfm=0, user_mode->flags=0x1
1094204.853: camplus_switch_mode [0]: request_mode: name=2x2_BINNED, width=1296, height=972, xfm=3
1094207.151: camplus_switch_mode: exit, mode used 2x2_BINNED
```

Repeat but with ```-md 2``` and I get
```
1157662.574: camplus_switch_mode: enter flags=0x0, user_mode: name=(null), width=1296, height=972, xfm=0, user_mode->flags=0x1
1157662.637: camplus_switch_mode [0]: request_mode: name=2592x1944, width=2592, height=1944, xfm=3
1157706.105: camplus_switch_mode: exit, mode used 2592x1944
1157853.526: camplus_switch_mode: enter flags=0x0, user_mode: name=(null), width=1296, height=972, xfm=0, user_mode->flags=0x1
1157853.591: camplus_switch_mode [0]: request_mode: name=2592x1944, width=2592, height=1944, xfm=3
1157859.327: camplus_switch_mode: exit, mode used 2592x1944
```
With ```-md 6``` I get
```
1179573.835: camplus_switch_mode: enter flags=0x0, user_mode: name=(null), width=1296, height=972, xfm=0, user_mode->flags=0x1
1179573.901: camplus_switch_mode [0]: request_mode: name=640x480p60, width=640, height=480, xfm=3
1179617.158: camplus_switch_mode: exit, mode used 640x480p60
1179694.553: camplus_switch_mode: enter flags=0x0, user_mode: name=(null), width=1296, height=972, xfm=0, user_mode->flags=0x1
1179694.618: camplus_switch_mode [0]: request_mode: name=640x480p60, width=640, height=480, xfm=3
1179696.785: camplus_switch_mode: exit, mode used 640x480p60
```

```user_mode``` is the output mode requested by the user. ```request_mode``` is the mode requested off the sensor.
The logging is quite verbose and to a circular buffer, so don't expect to leave it running for long periods before looking for the log lines of interest.

Malvineous, 2021-11-15:
Hmm very interesting.

So when I run `raspivid -w 640 -h 480` it tells me:
```
camplus_switch_mode: enter flags=0x0, user_mode: name=(null), width=640, height=480, xfm=0, user_mode->flags=0x1
camplus_switch_mode [0]: request_mode: name=2x2_BINNED, width=1296, height=972, xfm=3
```
However when I use `raspivid -w 1296 -h 972` it chooses the same mode:
```
camplus_switch_mode: enter flags=0x0, user_mode: name=(null), width=1296, height=972, xfm=0, user_mode->flags=0x1
camplus_switch_mode [0]: request_mode: name=2x2_BINNED, width=1296, height=972, xfm=3
```
I suppose that makes sense, the doco lists 640x480 as "2x2 binned plus skip" so I guess it's resized down in software.

I imagine you'll say that the `2x2_BINNED` mode is fixed and produces 1296x972 regardless (it's not an operator that halves the selected resolution).  But I'm still perplexed how this mode produces effectively a 640x480 image with every pixel doubled to 2x2 to make a 1296x972 video.

Am I misunderstanding how the sensor works?  Does it produce one video pixel for each sensor pixel, or does each sensor pixel contribute to multiple video pixels?

Maybe the issue is that in 1296x972 the quality is low because each 2x2 group of pixels shares some sensor elements, but when this image is resized down to 640x480, it ends up with each pixel being self-contained, so no quality is lost.

I imagine then the same issue happens at 2592x1944, however in the process of shrinking the frame down to 1296x972 each pixel again becomes self-contained. 

If this is correct, then it looks like the solution to my problem is to use `-md 2` for a 1296x972 capture, so I get the full sensor resolution with its overlapping pixels, but then have it resized down so that even at 1296x972 each pixel does not share anything with its neighbours.

Since I'm using V4L2 and V4L2M2M for this, any tips on how to set the equivalent to `-md 2` as a V4L2 control?  I can't seem to find one that matches.  I had a look at the `bcm2835_v4l2` module's parameters but nothing I could see there either.  Google tells me you've said in the past this isn't supported and is onerous to implement, although I wonder whether a kernel module option could offer a more convenient workaround?

Assuming none of these are possible, I wonder whether I could persuade you to change the default mode selection so that it picks the 2592x1944 mode as you go past say 800x600, in order to give a much sharper video?  This would match what happens with 640x480, selecting a much higher mode and then resizing down to give a sharper image.

6by9, 2021-11-15:
https://picamera.readthedocs.io/en/release-1.13/fov.html#sensor-modes for the list of sensor modes.
Mode 4 can produce up to 42fps. Quality is better with downscaling, therefore 640x480 @ 30fps will use mode 4 and downscale. 640x480 @ 42fps will use mode 6. (640x480 @ 61fps will go for mode 7).

https://picamera.readthedocs.io/en/release-1.13/fov.html#background-processes
The sensor produces one of the fixed resolutions. 2592x1944 reads out each pixel. 1292x972 bins together 4 pixels of each colour to produce one output pixel. 640x480 bins together 4 of each pixel, and then skips every other pixel horizontally and vertically.
The ISP (part of the SoC) will then take that image and resize it to whatever size output frame you ask for.

V4L2 doesn't support this form of mode selection as it is not aimed at complex pipelines with resizing.
libcamera does handle the complexities of these pipelines, and controls sensor, CSI2 receiver, and ISP from the kernel, and 3A from userland. 

No 2592x1944 can't be the default video mode as it can only read out at 15fps, which is no good for a 30fps video.

Malvineous, 2021-11-19:
Ah I see, thanks so much for the explanation.  I have done a bit more research and I understand now that it is the Bayer pattern on the sensor responsible for this apparent lower resolution, as multiple video pixels share the same sensor elements.

I was confused about 2592x1944 being limited to 15fps as it seems the 1920x1080 capture resolution uses that sensor mode at 30fps, but I gather from looking at the debug messages it can get away with it because it is reading only a subsection of lines off the sensor and not the full image.

I had a look at libcamera but it seems like it is not yet ready for production use (or at least I could not find any documentation about accessing it through the V4L2 interface it claims to have, so I presume this hasn't been implemented yet.)

Notwithstanding your comments about V4L2 and the sensor mode, I have implemented a V4L2 patch (https://github.com/raspberrypi/linux/pull/4715) to add support for setting the sensor mode through a V4L2 control, which I hope will be merged so we can use this functionality at least until libcamera is ready for use.  libcamera certainly seems like the way to go long term so I will be keeping my eye on it.

Thanks again for your help and explanations, I really appreciate it!

closed 2021-11-19 by Malvineous