raspberrypi/firmware #1651: Add ability to enable hardware watchdog before loading the kernel

zyga, 2021-11-15:
**Describe the bug**
The `tryboot.txt` functionality makes Raspberry Pi boot firmware almost perfectly capable of automatically recovering from any sort of bad system update. With appropriate precautions the system can recover, unharmed and operate normally. The firmware already supports loading the exposing the hardware watchdog with `dtparam=watchdog=on` but there's no apparent way to enable it at the same time.

This could only be done with, for example, synchronized change to enable the watchdog in linux userspace, e.g. with systemd, however it would allow automatically recovering from a kernel that doesn't boot correctly.

**To reproduce**
Add to `tryboot.txt`:
```
dtparam=watchdog=on
kernel=potato
```

Make an empty file in `/boot/potato`

Reboot the system with `reboot "0 tryboot"`.

**Expected behaviour**

The system fails to boot  but self-recovers after several seconds.

**Actual behaviour**

The system hangs until power is cycled externally.

**System**

Tested on Raspberry Pi 4 rev 1.4 with up-to-date raspbian but this is not affected by the userspace or the kernel.
**Logs**

```
pi@zpi:~$ sudo reboot "0 tryboot"
[  117.490774] reboot: Restarting system with command '0 tryboot'
Read start4db.elf bytes  3737864 hnd 0x00000000
Read fixup4db.dat bytes     8420 hnd 0x00000000
Firmware: 51215b4f6e3966401ecd99652a35cf1c25069113 Sep 30 2021 19:21:33
0x00d03114 0x00000000 0x00000fff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Starting start4db.elf @ 0xfec00200 partition 0
+

ASRT:00:00:04.488729:0: ../../../../../vcfw/drivers/chip/vciv/2708/clock_2711.c::clock_wait_busy!2125: timeout != 0
MESS:00:00:04.501015:0: arasan: arasan_emmc_open
MESS:00:00:04.937817:0: brfs: File read: /mfs/sd/tryboot.txt
MESS:00:00:04.941264:0: brfs: File read: 1913 bytes
MESS:00:00:04.996738:0: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:05.006256:0: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:05.015767:0: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:05.025282:0: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:05.034790:0: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:05.044305:0: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:05.053816:0: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:05.063331:0: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:05.072839:0: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:05.082354:0: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:05.086857:0: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:05.112632:0: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:05.122148:0: HDMI1:EDID error reading EDID block 0 attempt 1
MESS:00:00:05.131658:0: HDMI1:EDID error reading EDID block 0 attempt 2
MESS:00:00:05.141173:0: HDMI1:EDID error reading EDID block 0 attempt 3
MESS:00:00:05.150684:0: HDMI1:EDID error reading EDID block 0 attempt 4
MESS:00:00:05.160198:0: HDMI1:EDID error reading EDID block 0 attempt 5
MESS:00:00:05.169706:0: HDMI1:EDID error reading EDID block 0 attempt 6
MESS:00:00:05.179221:0: HDMI1:EDID error reading EDID block 0 attempt 7
MESS:00:00:05.188732:0: HDMI1:EDID error reading EDID block 0 attempt 8
MESS:00:00:05.198247:0: HDMI1:EDID error reading EDID block 0 attempt 9
MESS:00:00:05.202746:0: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:05.213829:0: brfs: File read: /mfs/sd/tryboot.txt
MESS:00:00:09.337747:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:09.345172:0: *** Restart logging
MESS:00:00:09.346471:0: brfs: File read: 1913 bytes
MESS:00:00:09.357280:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:09.367320:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:09.377351:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:09.387390:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:09.397424:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:09.407463:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:09.417495:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:09.427534:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:09.437568:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:09.447608:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:09.452629:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:09.463254:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:09.473293:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:09.483324:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:09.493363:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:09.503397:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:09.513435:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:09.523466:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:09.533505:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:09.543539:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:09.553577:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:09.558599:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:09.564198:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:09.577989:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:09.588024:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 1
MESS:00:00:09.598064:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 2
MESS:00:00:09.608096:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 3
MESS:00:00:09.618135:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 4
MESS:00:00:09.628168:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 5
MESS:00:00:09.638207:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 6
MESS:00:00:09.648239:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 7
MESS:00:00:09.658277:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 8
MESS:00:00:09.668311:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 9
MESS:00:00:09.673341:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:09.683964:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:09.693997:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 1
MESS:00:00:09.704036:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 2
MESS:00:00:09.714067:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 3
MESS:00:00:09.724106:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 4
MESS:00:00:09.734141:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 5
MESS:00:00:09.744179:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 6
MESS:00:00:09.754210:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 7
MESS:00:00:09.764249:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 8
MESS:00:00:09.774283:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 9
MESS:00:00:09.779313:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:09.784911:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:09.793675:0: HDMI0: hdmi_pixel_encoding: 300000000
MESS:00:00:09.799147:0: HDMI1: hdmi_pixel_encoding: 300000000
MESS:00:00:09.809372:0: dtb_file 'bcm2711-rpi-4-b.dtb'
MESS:00:00:09.817666:0: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
MESS:00:00:09.820913:0: Loading 'bcm2711-rpi-4-b.dtb' to 0x100 size 0xc2a9
MESS:00:00:09.840166:0: brfs: File read: 49833 bytes
MESS:00:00:09.854146:0: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
MESS:00:00:09.920991:0: brfs: File read: 1667 bytes
MESS:00:00:09.925880:0: brfs: File read: /mfs/sd/tryboot.txt
MESS:00:00:09.928926:0: dtparam: audio=on
MESS:00:00:09.942185:0: brfs: File read: 1913 bytes
MESS:00:00:09.957179:0: brfs: File read: /mfs/sd/overlays/vc4-fkms-v3d-pi4.dtbo
MESS:00:00:09.978580:0: Loaded overlay 'vc4-fkms-v3d'
MESS:00:00:09.980605:0: dtparam: watchdog=on
MESS:00:00:10.038393:0: brfs: File read: 1446 bytes
MESS:00:00:10.041549:0: brfs: File read: /mfs/sd/cmdline.tx2
MESS:00:00:10.045569:0: Read command line from file 'cmdline.tx2':
MESS:00:00:10.051445:0: 'console=serial0,115200 console=tty1 root=PARTUUID=1a9fcfff-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait IT-WORKS'
MESS:00:00:10.167147:0: brfs: File read: 130 bytes
MESS:00:00:10.174033:0: brfs: File read: /mfs/sd/potato
MESS:00:00:10.176146:0: Loading 'potato' to 0x8000 size 0x0
MESS:00:00:10.181449:0: Device tree loaded to 0x2eff3800 (size 0xc7c4)
MESS:00:00:10.189313:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:10.196766:0: uart: Baud rate change done...
MESS:00:00:10.198788:0:
```

The system hangs here.

**Additional context**

This is the last known limitation of using Raspberry Pi hardware without uboot in industrial setup where reliability is paramount.

pelwell, 2021-11-15:
It would not be difficult for the firmware to detect the `dtparam=watchdog` and enable it early, but that still wouldn't protect against a bad firmware or bad config.txt.

zyga, 2021-11-15:
That's true but it's a big upgrade from having to jump through u-boot to protect against bad kernel. The kernel is a much larger error surface.

popcornmix, 2021-11-15:
One other limitation is the hardware watchdog has a limited maximum duration (of around 40 seconds),
so the arm needs to boot quickly enough to kick/disable the watchdog within that limit.

zyga, 2021-11-15:
Right, I read about that. I found only scraps of information (suggesting 16 seconds). I think it is long enough to boot Linux. If you could produce some firmware that exposes this feature for testing, I would love to give it a try. Given that I work with Yocto and we can have tailored kernels quite easily, we could reach user-space for systemd to start managing the watchdog in time, I think.

timg236, 2021-11-15:
I think we should also modify the start.elf fatal error handle (e.g kernel not found) to reset instead of blocking in tryboot mode OR if config.txt specified the dtparam=watchdog.

Since the USB and TFTP boot-modes have built in timeouts they will generally end up in 'kernel-not-found' if it all goes wrong but as @pelwell says still gaps.

zyga, 2021-11-15:
I think this is a good idea. While QA will certainly catch thinks like broken kernel during testing. Developers doing local development and updating with the OTA code for convenience may report that they expected the update stack to handle this better. Having this addressed in the boot firmware would be most desirable.

popcornmix, 2021-11-16:
> Right, I read about that. I found only scraps of information (suggesting 16 seconds). 

That is correct. 0xfffff x 16uS is the limit.

pelwell, 2021-11-16:
https://drive.google.com/file/d/17oyUcUtSHENN7XwinemKb2upbKpnWoIY/view?usp=sharing is a trial firmware for Pi 4 (just start4.elf and fixup4.dat) that implements the ideas discussed above:
1. With "dtparam=watchdog", the firmware enables the watchdog before launching the kernel.
2. When using tryboot, any path leading to an error pattern on the activity LED will reset after the first time round.

zyga, 2021-11-16:
Woot! Thank you for producing this. I'll get to testing!

zyga, 2021-11-16:
This seems to work beautifully:

```
pi@zpi:~$ sudo reboot "0 tryboot"
[  171.682250] reboot: Restarting system with command '0 tryboot'
Read start4.elf bytes  2239360 hnd 0x00000000
Read fixup4.dat bytes     5352 hnd 0x00000000
Firmware: 99c5aad7da0d1e3b2c28669a3941db77734b9f2d Nov 16 2021 15:30:54
0x00d03114 0x00000000 0x00000fff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Starting start4.elf @ 0xfec00200 partition 0
+

MESS:00:00:03.238951:0: arasan: arasan_emmc_open
MESS:00:00:03.393235:0: brfs: File read: /mfs/sd/tryboot.txt
MESS:00:00:03.396665:0: brfs: File read: 1915 bytes
MESS:00:00:03.451777:0: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:03.461293:0: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:03.470803:0: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:03.480317:0: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:03.489825:0: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:03.499340:0: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:03.508850:0: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:03.518364:0: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:03.527871:0: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:03.537385:0: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:03.541886:0: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:03.567460:0: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:03.576975:0: HDMI1:EDID error reading EDID block 0 attempt 1
MESS:00:00:03.586482:0: HDMI1:EDID error reading EDID block 0 attempt 2
MESS:00:00:03.595996:0: HDMI1:EDID error reading EDID block 0 attempt 3
MESS:00:00:03.605505:0: HDMI1:EDID error reading EDID block 0 attempt 4
MESS:00:00:03.615019:0: HDMI1:EDID error reading EDID block 0 attempt 5
MESS:00:00:03.624527:0: HDMI1:EDID error reading EDID block 0 attempt 6
MESS:00:00:03.634041:0: HDMI1:EDID error reading EDID block 0 attempt 7
MESS:00:00:03.643551:0: HDMI1:EDID error reading EDID block 0 attempt 8
MESS:00:00:03.653065:0: HDMI1:EDID error reading EDID block 0 attempt 9
MESS:00:00:03.657564:0: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:03.668761:0: brfs: File read: /mfs/sd/tryboot.txt
MESS:00:00:07.759827:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:07.767131:0: *** Restart logging
MESS:00:00:07.768550:0: brfs: File read: 1915 bytes
MESS:00:00:07.779359:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:07.789399:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:07.799429:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:07.809465:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:07.819498:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:07.829535:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:07.839566:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:07.849603:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:07.859636:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:07.869673:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:07.874695:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:07.885319:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:07.895357:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:07.905387:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:07.915425:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:07.925458:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:07.935496:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:07.945526:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:07.955564:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:07.965597:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:07.975635:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:07.980656:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:07.986254:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:08.000045:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:08.010079:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 1
MESS:00:00:08.020118:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 2
MESS:00:00:08.030148:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 3
MESS:00:00:08.040186:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 4
[... 447 lines cut ...]
[    5.118249] systemd[1]: Finished Load Kernel Module drm.
[    5.126968] systemd[1]: modprobe@fuse.service: Succeeded.
[    5.133991] systemd[1]: Finished Load Kernel Module fuse.
[    5.142633] systemd[1]: Finished Load Kernel Modules.
[    5.155798] systemd[1]: Mounting FUSE Control File System...
[    5.167278] systemd[1]: Mounting Kernel Configuration File System...
[    5.182377] systemd[1]: Started File System Check Daemon to report status.
[    5.210757] systemd[1]: Starting Apply Kernel Variables...
[    5.239285] systemd[1]: Mounted FUSE Control File System.
[    5.245737] systemd[1]: Mounted Kernel Configuration File System.

Raspbian GNU/Linux 11 zpi ttyS0

zpi login: 
```

In my host system I've applied only one-line change:

```
capi@zpi:~$ cat /etc/systemd/system.conf | grep '^[^#]'
[Manager]
RuntimeWatchdogSec=10
```

This is otherwise a vanilla Raspbian.

zyga, 2021-11-16:
I've also tested this for vanilla booting (no tryboot) and I didn't see any adverse side-effects.

pelwell, 2021-11-16:
The patch has been merged to our internal repo and will appear in future firmware releases.

popcornmix, 2021-11-16:
It's in rpi-update firmware now.

zyga, 2021-11-16:
@popcornmix I was thinking if it needs to be reverted or adjusted. This is introducing a somewhat breaking functionality. Unless the host userspace is picking up the watchdog, like my system did because I altered the default, it will stay in a reboot loop. In case people had dtparam=watchdog before, but were not using the watchdog for real, it may cause regressions.

I'll give it a spin with default systemd settings and report back.

pelwell, 2021-11-16:
I didn't alter the kernel or userspace in any way and experienced no issues.

pelwell, 2021-11-17:
Overnight I was worrying that I hadn't tested successful tryboots, but it all seems OK. For the record, the logic is:
1. With "dtparam=watchdog", the firmware enables the watchdog before launching the kernel _regardless of whether or not `tryboot` is used_. It would be easy to make this conditional on `tryboot`, but I currently see no reason to.
2. When using `tryboot`, any path leading to an error pattern on the activity LED will reset after the first sequence of flashes.

nunojpg, 2021-12-09:
I can not test this to work.

I am using the latest firmware: 

```
strings start4.elf| grep VC_BUILD
VC_BUILD_ID_USER: dom
VC_BUILD_ID_TIME: 11:40:59
VC_BUILD_ID_VARIANT: start
VC_BUILD_ID_TIME: Dec  1 2021
VC_BUILD_ID_BRANCH: bcm2711_2
VC_BUILD_ID_HOSTNAME: buildbot
VC_BUILD_ID_PLATFORM: raspberrypi_linux
VC_BUILD_ID_VERSION: be18b6271a0f99f896fc81249da5da6658422558 (clean)
```

And my config.txt:

```
enable_uart=1
dtparam=watchdog
device_tree=dummy.dtb
kernel=u-boot.bin
arm_64bit=1

```

This dummy.dtb is a device tree from upstream called like this to make it clear that it is not actually used since, but just to make the firmware happy.

Then I get u-boot into interactive console and I expect the watchdog to reboot the SOC, which never happens.

I can't find any evidence that u-boot is keeping the watchdog reset to avoid a reboot.

pelwell, 2021-12-09:
The watchdog is enabled if the loaded Device Tree has a node with the label `watchdog` which is enabled. The normal RPi DTBs have a watchdog node but it is disabled unless the dtparam is used. Why don't you just allow the real DTB to be loaded, even if you ignore it in U-boot, then you won't have this problem?

pelwell, 2021-12-09:
I'm going to assume that the original problem has now been dealt with and close the issue.

closed 2021-12-09 by pelwell

colemickens, 2022-01-10:
Is this available in the "stable" tagged release from 4 days ago? If not, would a build straight from master have it?

colemickens, 2022-01-10:
From testing, it appears the build from Jan6, while not available in a tagged release, but available on the **stable** branch, does reboot me when my u-boot gets stuck. This is great to have, thanks!

pelwell, 2022-01-10:
Correct - it's definitely in the build from the 6th (the lack of tag is an oversight).

dcoder42, 2022-11-16:
I can't seem to get this to work properly. I have the following ```tryboot.txt``` with potato kernel being an empty file
```
dtparam=watchdog
hdmi_force_hotplug=1
hdmi_drive=2
dtparam=audio=on
dtoverlay=vc4-kms-v3d
max_framebuffers=2
disable_overscan=1
kernel=potato
```

When rebooting with ```reboot "0 tryboot"``` the boot process hangs and after 15 seconds the device resets as expected. Removing the ```kernel=potato``` line and rebooting again with ```reboot "0 tryboot"``` the device successfully boots and stays on. No service is triggering ```/dev/watchdog``` so i expected the device to reboot again but this doesn't happen until i manually trigger ```/dev/watchdog``` one time. Am i missing something? 

I also don't fully understand the ```normal RPi DTBs have a watchdog node but it is disabled unless the dtparam is used``` statement from pelwell. What's the "normal DTB" here? The one included in the official image or is the meaning here for RPI2 or lower? Do i need other DTB's to get this to work?

I have tested this with the ```2022-09-22-raspios-bullseye-armhf-lite.img``` i have downloaded today

```vcdbg log msg``` shows ```dtparam: watchdog=true``` instead of ```dtparam: watchdog=on``` like in the logs above. Maybe this is wrong?

```
004008.703: arasan: arasan_emmc_open
004203.849: brfs: File read: /mfs/sd/tryboot.txt
004204.180: brfs: File read: 128 bytes
004284.916: HDMI1:EDID error reading EDID block 0 attempt 0
004285.930: HDMI1:EDID giving up on reading EDID block 0
004287.080: brfs: File read: /mfs/sd/tryboot.txt
004734.425: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
004736.945: *** Restart logging
004736.966: brfs: File read: 128 bytes
004800.864: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
004801.879: hdmi: HDMI1:EDID giving up on reading EDID block 0
004806.920: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
004807.937: hdmi: HDMI1:EDID giving up on reading EDID block 0
004807.953: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
004807.967: HDMI0: hdmi_pixel_encoding: 300000000
004807.980: HDMI1: hdmi_pixel_encoding: 300000000
004812.773: dtb_file 'bcm2711-rpi-4-b.dtb'
004818.143: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
004818.161: Loaded 'bcm2711-rpi-4-b.dtb' to 0x100 size 0xcaf3
004831.119: brfs: File read: 51955 bytes
004846.920: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
004919.925: brfs: File read: 2043 bytes
004920.940: brfs: File read: /mfs/sd/tryboot.txt
004920.961: dtparam: watchdog=true
004929.995: dtparam: audio=on
004940.307: brfs: File read: 128 bytes
004962.716: brfs: File read: /mfs/sd/overlays/vc4-kms-v3d-pi4.dtbo
005054.493: Loaded overlay 'vc4-kms-v3d'
005358.605: brfs: File read: 3914 bytes
005361.964: brfs: File read: /mfs/sd/cmdline.txt
005362.016: Read command line from file 'cmdline.txt':
005362.034: 'console=serial0,115200 console=tty1 root=PARTUUID=eca61bdc-02 rootfstype=ext4 fsck.repair=yes rootwait'
005620.971: brfs: File read: 103 bytes
006596.218: brfs: File read: /mfs/sd/kernel7l.img
006596.246: Loaded 'kernel7l.img' to 0x8000 size 0x6b6ca0
006596.294: Device tree loaded to 0x2eff2e00 (size 0xd1c4)
006601.832: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
009514.880: vchiq_core: vchiq_init_state: slot_zero = 0xdad80000, is_master = 1
009523.618: TV service:host side not connected, dropping notification 0x00000002, 0x00000001, 0x0000005f
```

```
strings start4.elf | grep VC_BUILD
VC_BUILD_ID_USER: dom
VC_BUILD_ID_TIME: 14:03:16
VC_BUILD_ID_VARIANT: start
VC_BUILD_ID_TIME: Aug 26 2022
VC_BUILD_ID_BRANCH: bcm2711_2
VC_BUILD_ID_HOSTNAME: buildbot
VC_BUILD_ID_PLATFORM: raspberrypi_linux
VC_BUILD_ID_VERSION: 102f1e848393c2112206fadffaaf86db04e98326 (clean)
```

dcoder42, 2022-11-18:
I think i have figured it out. The raspberry kernel is configured to take over an already running watchdog at boot. There is a thread running in the kernel to trigger the watchdog until the user space takes over.

```
modprobe configs
zcat /proc/config.gz > .config

CONFIG_WATCHDOG_HANDLE_BOOT_ENABLED=y
CONFIG_WATCHDOG_OPEN_TIMEOUT=0
```

An additional timeout for the user space to take over also exists but this is set to infinite as default. That means it doesn't matter if the user space (rootfs) is not quicky attached, the kernel loads the bcm2835_wdt driver in less than 2 seconds and will continue to trigger the watchdog until the user space takes over. IMO it makes sense to add an additional "watchdog.open_timeout=120" to cmdline.txt to make sure to reset the device if for whatever reason the user space was not able to take over the watchdog.

rschell, 2022-11-18:
I would really like to implement something like this.  I've had two headless 4b machines fail to boot recently, due to I believe DRM issues (see Set DDC pointer commit).  The machines don't even establish an IP connection.  Something about recent changes around vc4-kms-v3d it doesn't like.  These remote machines don't have the capability to plug a hdmi cable into them.  One of the machines, I switched to vc4-fkms-v3d for the time being.

I'd like the initial boot try to a headless vnc configuration which is my standard interface to these machines, but if that fails to come up, I'd like to retry with just a console ssh capability.  Is this possible?