raspberrypi/firmware #1653: Can't boot Raspberry Pi Compute Module 4 from SATA

geerlingguy, 2021-11-16:
**Is this the right place for my bug report?**

Yes

**Describe the bug**

When I attempt to boot my Raspberry Pi Compute Module 4 from a SATA hard drive (SSD or HDD), it does not boot.

**To reproduce**

1. Plug a compatible SATA controller card into the Compute Module 4 IO board.
2. Flash Pi OS onto a SATA SSD or HDD.
3. Plug that drive into the SATA controller.
4. Attempt to boot the Pi.

**Expected behaviour**

The Pi's firmware is able to detect a valid image on any of the attached SATA devices and boot from it.

**Actual behaviour**

The Pi loops through looking for emmc/usb/nvme boot devices.

**System**

* Which model of Raspberry Pi?

Compute Module 4

* Which OS and version (`cat /etc/rpi-issue`)?

```
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 071d3956353f2be964d2d337d3b123f1872b7f53, stage2
```

* Which firmware version (`vcgencmd version`)?

```
Oct 29 2021 10:47:33 
Copyright (c) 2012 Broadcom
version b8a114e5a9877e91ca8f26d1a5ce904b2ad3cf13 (clean) (release) (start)
```

* Which kernel version (`uname -a`)?

```
Linux taco 5.10.63-v8+ #1459 SMP PREEMPT Wed Oct 6 16:42:49 BST 2021 aarch64 GNU/Linux
```

**Logs**

N/A

**Additional context**

Some people have been asking my why SATA boot isn't working, and I wanted to be able to point them to a more definitive 'status' of the feature, so I'm posting this issue—if it's not something that will likely be implemented soon, that's fine, I just wanted to make a more official feature request/report than random musings on the forums.

pelwell, 2021-11-16:
From what I remember, SATA's a nightmare compared to NVME. In an EEPROM which is already a bit short on space I really can't see this happening.

6by9, 2021-11-16:
Which SATA card(s) would you support?

Yes the AHCI interface covers lots, but then what about the RAID controllers from various vendors, or the other random vendors that have their own interfaces? Where do you stop?
IIRC There isn't support for NVMe behind a PCIe switch either, as enumeration there becomes more entertaining, and everything balloons in scale and complexity.

So as pelwell says, I can't see it happening.

jlinton, 2022-02-28:
You can turn on the Ahci driver (or the x86 PCIe option ROM emulator) in edk2/pftf, and try that. It should work with DT, and if you apply the linux/ACPI/PCIe patches will work in ACPI mode.

rcfa, 2022-10-02:
> You can turn on the Ahci driver (or the x86 PCIe option ROM emulator) in edk2/pftf, and try that. It should work with DT, and if you apply the linux/ACPI/PCIe patches will work in ACPI mode.

Sorry, that’s a bit too terse for me to grok.
Could you please be a bit more specific as to what steps to follow to try this?
Also, what do you expect to work or not work? Boot off SATA, or have rootfs on SATA, but /boot on eMMC/SD card?

timg236, 2024-10-01:
There are no plans to support SATA boot on CM4 (or CM5) since it would an unnecessary size and complexity. For high capacity / high performance storage we recommend NVMe. For alternate root file-systems we recommend eMMC boot and letting the Linux kernel handle this via an initramfs since this can handle more exotic storage e.g. RAID

RassK, 2024-10-02:
@timg236 Sorry for my lack of knowledge about the issue. Very new world for me. How does my example work then? Is it not recommended way to go forward?

I just see that Turing RK1 is using similar pattern. Bootloader in eMMC and everything else in external drive. Unfortunately TuringPi2 does not support NVME boot for CM4, so I had to use SATA.

```
ubuntu@cm4-ubuntu-server-3:~$ lsblk
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
loop0          7:0    0  33.7M  1 loop /snap/snapd/21761
sda            8:0    0 238.5G  0 disk
├─sda1         8:1    0   512M  0 part /boot/firmware
└─sda2         8:2    0   238G  0 part /
mmcblk0      179:0    0  29.1G  0 disk
├─mmcblk0p1  179:1    0   512M  0 part
└─mmcblk0p2  179:2    0  28.6G  0 part
mmcblk0boot0 179:32   0     4M  1 disk
mmcblk0boot1 179:64   0     4M  1 disk
```

timg236, 2024-10-02:
When you say SATA, how is it physically connected to the CM4? e.g. via  a USB/SATA adapter or PCIe ?

RassK, 2024-10-02:
Directly to the [Turing Pi 2](https://docs.turingpi.com/docs/turing-pi2-specs-and-io-ports) board, which work only with node 3.

timg236, 2024-10-02:
There is no direct support for SATA so you would need to boot from EMMC / NVMe instead

Harvie, 2025-01-15:
Perhaps having intermediate bootloader might work? eg.:
You can prepare very tiny linux kernel with EFI stub, SATA drivers and embed initramfs in it, initramfs will have scripts that look for kernel on SATA and load it using kexec. All that might fit into single file on (e)MMC. I know it's not super elegant, but it might work.

timg236, 2025-01-20:
As already mentioned native SATA support would add a huge amount of complexity and interop work. This feature is less relevant now that there is native NVMe support and I would imagine that effort in this error would be targeted towards NVMe

For complex boot devices the recommended approach is to have the kernel on local flash memory (eMMC / SD). If this is for an embedded product then rpi-image-gen should make customization of the partition layout much easier for an APT based system
https://github.com/raspberrypi/rpi-image-gen

Closing as not planned since this doesn't exist on any internal roadmap.

Perhaps a forum stick thread for hybrid boot devices might be useful here?

closed 2025-01-20 by timg236

geerlingguy, 2025-01-21:
@timg236 - Good idea; even if not stickied, it'd be nice if there were a central resource for this 'split boot partition' scenario, since the few guides I've seen out there are mostly written back in Pi 2/3/4 days prior to USB boot becoming mainstream. So many of the Google-able results will lead users a bit astray.