raspberrypi/firmware #1654: [RFE] missing kernel modules for some USB gadget functions

zhangyoufu, 2021-11-17:
**Is this the right place for my bug report?**
The `/proc/config.gz` on my RPi doesn't match `defconfig`s from https://github.com/raspberrypi/linux. And the Linux kernel is shipped via firmware update, instead of a raspbian package. So I think this is the right place to report this issue.

**Describe the bug**
I want to make use of USB CDC-NCM (for networking) and TCM (for storage) gadget functions on RPi. However the Linux configuration does not enable aforementioned modules.

**To reproduce**
```
root@raspberrypi:~# modprobe configs
root@raspberrypi:~# zgrep USB_F_ /proc/config.gz
CONFIG_USB_F_ACM=m
CONFIG_USB_F_SS_LB=m
CONFIG_USB_F_SERIAL=m
CONFIG_USB_F_OBEX=m
CONFIG_USB_F_ECM=m
CONFIG_USB_F_SUBSET=m
CONFIG_USB_F_RNDIS=m
CONFIG_USB_F_MASS_STORAGE=m
CONFIG_USB_F_UAC2=m
CONFIG_USB_F_UVC=m
CONFIG_USB_F_MIDI=m
CONFIG_USB_F_HID=m
CONFIG_USB_F_PRINTER=m
```

**Expected behaviour**
expected `CONFIG_USB_F_NCM=m` and `CONFIG_USB_F_TCM=m`

**Actual behaviour**
`CONFIG_USB_F_TCM` and `CONFIG_USB_F_TCM` are not configured.

**System**

* Which OS and version (`cat /etc/rpi-issue`)?
  ```
  root@raspberrypi:~# cat /etc/rpi-issue
  Raspberry Pi reference 2021-10-30
  Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 288b21fc27e128ea6b330777aca68e0061ebf4fe, stage2
  ```
* Which firmware version (`vcgencmd version`)?
  ```
  root@raspberrypi:~# vcgencmd version
  Oct 29 2021 10:49:08
  Copyright (c) 2012 Broadcom
  version b8a114e5a9877e91ca8f26d1a5ce904b2ad3cf13 (clean) (release) (start)
  ```
* Which kernel version (`uname -a`)?
  ```
  root@raspberrypi:~# uname -a
  Linux raspberrypi 5.10.63-v7+ #1459 SMP Wed Oct 6 16:41:10 BST 2021 armv7l GNU/Linux
  ```

pelwell, 2021-11-17:
I think you'll find that the configs file does match the appropriate defconfig (bcm2709_defconfg for a Pi 2, 3 or Zero 2 such as you have), but in a non-obvious way.

The Pi 4 defconfig (bcm2711_defconfig) does include `CONFIG_USB_NET_CDC_NCM=m` and `CONFIG_USB_CONFIGFS_NCM=y`, which results in the following in .config:
```
CONFIG_USB_NET_CDC_NCM=m
CONFIG_USB_F_NCM=m
CONFIG_USB_CONFIGFS_NCM=y
```
The same is also try for Pi 1 and Pi Zero (bcmrpi_defconfig), but _not_ for Pi 2 and Pi 3/Zero 2 (bcm2709_defconfig). That will be because until Zero 2 W (which I suspect is what you have) there has been no way of using gadget mode, so that support is disabled.

I will enable the NCM gadget support.

You also ask for `CONFIG_USB_F_TCM`, saying it is "(for storage)" - that's not what the Kconfig file says:
```
config USB_CONFIGFS_F_TCM
	bool "USB Gadget Target Fabric"
	depends on TARGET_CORE
	depends on USB_CONFIGFS
	select USB_LIBCOMPOSITE
	select USB_F_TCM
	help
	  This fabric is a USB gadget component. Two USB protocols are
	  supported that is BBB or BOT (Bulk Only Transport) and UAS
	  (USB Attached SCSI). BOT is advertised on alternative
	  interface 0 (primary) and UAS is on alternative interface 1.
	  Both protocols can work on USB2.0 and USB3.0.
	  UAS utilizes the USB 3.0 feature called streams support.
```
Are you sure this is what you need, and not the more obvious mass storage support?:
```
config USB_CONFIGFS_MASS_STORAGE
	bool "Mass storage"
	depends on USB_CONFIGFS
	depends on BLOCK
	select USB_F_MASS_STORAGE
	help
	  The Mass Storage Gadget acts as a USB Mass Storage disk drive.
	  As its storage repository it can use a regular file or a block
	  device (in much the same way as the "loop" device driver),
	  specified as a module parameter or sysfs option.
```

zhangyoufu, 2021-11-18:
> the configs file does match the appropriate defconfig (bcm2709_defconfg for a Pi 2, 3 or Zero 2 such as you have), but in a non-obvious way

Didn't know that. Thanks for pointing out.

> Zero 2 W (which I suspect is what you have)

Yes, I'm using USB OTG gadget on RPi Zero 2 W.

> Are you sure this is what you need, and not the more obvious mass storage support?

I'm aware that there is separate Mass Storage function and TCM function. The TCM function handles BOT (Bulk-Only Transfer) and UAS (USB-Attached SCSI). I think that this may help reduce some overhead. I'm not sure, just wanna try.

Please also enable `CONFIG_USB_FUNCTIONFS`. It's required for a userspace MTP responder like [this one](https://github.com/viveris/uMTP-Responder).

pelwell, 2021-11-22:
The following features have been enabled in the Pi 3 (and Zero 2 W) kernel, bringing it in line with the Zero W:
```diff
+CONFIG_USB_CONFIGFS=m
+CONFIG_USB_CONFIGFS_SERIAL=y
+CONFIG_USB_CONFIGFS_ACM=y
+CONFIG_USB_CONFIGFS_OBEX=y
+CONFIG_USB_CONFIGFS_NCM=y
+CONFIG_USB_CONFIGFS_ECM=y
+CONFIG_USB_CONFIGFS_ECM_SUBSET=y
+CONFIG_USB_CONFIGFS_RNDIS=y
+CONFIG_USB_CONFIGFS_EEM=y
+CONFIG_USB_CONFIGFS_MASS_STORAGE=y
+CONFIG_USB_CONFIGFS_F_LB_SS=y
+CONFIG_USB_CONFIGFS_F_FS=y
+CONFIG_USB_CONFIGFS_F_UAC1=y
+CONFIG_USB_CONFIGFS_F_UAC2=y
+CONFIG_USB_CONFIGFS_F_MIDI=y
+CONFIG_USB_CONFIGFS_F_HID=y
+CONFIG_USB_CONFIGFS_F_UVC=y
+CONFIG_USB_CONFIGFS_F_PRINTER=y
```

You'll have to make more of a case for `USB_FUNCTIONFS` and `USB_CONFIGS_F_TCM`. Before editing your post said:

> Please also enable CONFIG_USB_FUNCTIONFS. It's required for a userspace MTP responder like [this one](https://github.com/viveris/uMTP-Responder).

Following the link it says:
> The only requirement is to have the USB FunctionFS (CONFIG_USB_FUNCTIONFS) or GadgetFS (CONFIG_USB_GADGETFS) support enabled in your Linux kernel.

Since USB_GADGETFS is enabled now you shouldn't need USB_FUNCTIONFS.

For `USB_CONFIGFS_F_TCM`, if you want to "just wanna try" then you can build your own kernel: https://www.raspberrypi.com/documentation/computers/linux_kernel.html#building

closed 2021-11-22 by zhangyoufu