raspberrypi/firmware #1655: Firmware not found

golosfar, 2021-11-17:
HI everyone. I am new to Raspberry PI and I really like it, but I got some problems I need help with.

I have 3 SD cards to my RPI4-8GB

1 has Windows 10 as OS and is running fairly okay (32GB)
1 has Twisoter OS and are fantastic and wit no problems at all (32GB)
1 have NOOBS that came pre-installed with my RPI. (64GB)

Yesterday I ran Twister on mu RPI and it was fine, but I had some issues on getting Minecraft JAVA installing and running on it, so I shotdown my RPI and went to bed. 
Tonight I took out the SD card with Twister and inserted the card with NOOBS, but my RPI could not start. I got the "Firmware not Found" message on a black screen. Then I took the card out and inserted my Twister card, and it is running.

What did I do wrong? And how dio I make my NOOBS card work again?

Can anyone help me here?

(I am Danish and 58 yo, so if my English isn't that good, please bare with me).

Best regards Frank aka Golosfar

golosfar, 2021-11-17:
RE. FIRMWARE NOT FOUND

Hello again. I just repeated the action with my NOOBS card. I gave around a minute to try and find the card and the firmware, but it still didn't get started. Then I try to take the card out, but I burned my finger on the SDcard. I got a blister on the tip of my finger, so hot was the card! What on earth is going on with the card? do I have to re-install NOOBS on the card again or???
Why did the SDcard get so hot and could there be any damage to it now? As I wrote in my first message, I am new to this, but I have burned my finger on a computer before?

Best regards Frank aka Golosfar

timg236, 2021-11-17:
"Then I try to take the card out, but I burned my finger on the SDcard. I got a blister on the tip of my finger, so hot was the card! What on earth is going on with the card?"

This normally indicates that there is a crack in the card causing a short-circuit which will make the card very hot.

Please examine the card carefully and if you can see any signs of stress (e.g. a crack) or it bends slightly then replace the card.

popcornmix, 2021-11-17:
If the sdcard is that hot it is irrecoverably broken. Usually this occurs after physical stress (e.g. a hairline crack in the card).

golosfar, 2021-11-17:
Hi popcornmix.
Yep, it is broken. Don't know how, but it is broken. Gotta go buy a new one tomorrow and re-install NOOBS one more time.
Thanx for your help. :-)
Frank

popcornmix, 2021-11-17:
I've not heard of a broken sdcard damaging the Pi, but it's conceivable if it shorts something.
However if you booting Twister OS to desktop it seems unlikely the Pi is damaged.
See how you get on installing NOOBS with the new sdcard.

Gerardhub, 2022-12-11:
I too am now facing the same problem, which is apparently common. No idea what caused it, but my RPi 400 suddenly won't boot and reports to me: "firmware not found". I tried several SD cards but apparently it is not because of an SD card.
1. What could have caused this, in other words, how could this have been prevented?
2. How can this problem be fixed?

pelwell, 2022-12-11:
Boot or reboot? There was a problem with the firmware which could affect reboot, and a fix will appear in the next update.

Can you upload photos of the diagnostic messages when booting fails?

Gerardhub, 2022-12-11:
Hi Pelwell,

The only thing I can do is to use Esc to change the page to the one containing the text "Firmware not found".

As for this fix, it seems this problem has existed for quite some time. An update is not possible because the LXTerminal is not accesible.

I cant make screenshots because of the same reasons ("scrot" isn't accesible, because the LXTYerminal is not accessible) but in this same thread there are pictures that show the same as mine.

popcornmix, 2022-12-11:
> I cant make screenshots because of the same reasons ("scrot" isn't accesible, because the LXTerminal isnt accessible) but in this same thread there are pictures that show the same as mine.

A photo from a phone would be fine.

Gerardhub, 2022-12-12:
> A photo from a phone would be fine.

Not everyone has a boring life or considers themselves so important that a pocket computer (also called a smartphone because you can also use it to make calls) should always be wthin reach. ;) I don't have one because a quiet life suits me very well. ;)

dannysilence, 2023-03-18:
Hello. Facing same problem, already tried running PI with EEPROM restore images for boot from both devices - as SD card same as USB (EEPROM restore image was written from Pi Imager application, so not sure I could have a mistake here.) 

[image]

dannysilence, 2023-03-18:
> 

hope it helps :)

rwzdoorn, 2023-03-18:
I'm with team @dannysilence  .. have the same issue since yesterday update. Anyone has a clue what happend?

pelwell, 2023-03-18:
What USB storage do you two have attached?

rwzdoorn, 2023-03-19:
@pelwell SSD (Samsung 870 EVO 1TB)

rwzdoorn, 2023-03-22:
Fixed my issues following this topic: https://forums.raspberrypi.com/viewtopic.php?t=343161 

_You can try recovering by getting a fresh copy of the OS (the Lite version will do) on another card or USB device. Copying all the files in the first partition to your old card - except **config.txt** and **cmdline.txt**._

closed 2024-01-06 by timg236