raspberrypi/firmware #1656: Pi 400 wake only by F10 possible?

fanoush, 2021-11-20:
as per https://forums.raspberrypi.com/viewtopic.php?t=324115 
Would it be possible to allow Pi 400 to boot fully only if F10 was pressed, to prevent unattended boot when power is lost? Also it would prevent SD card corruption if power is lost again.

Idea is to shutdown in bootloader like if F10 was used if device was booted from cold state by attaching power.

pelwell, 2021-11-20:
The boot EEPROM can't distinguish between applying the power and pushing F10 - that's handled by a dedicated microcontroller with preprogrammed firmware. No amount of software changes will change that

fanoush, 2021-11-20:
OK, thanks. The article https://www.raspberrypi.com/news/have-you-tried-turning-it-off-and-on-again/ says "If the power key (F10 or Fn+F10) is pressed, it wakes up the Power Management Unit (PMU) chip via the GLOBAL_EN pin"

So was not sure if it could be later determined from PMU chip that this happened or that power was applied to USB-C connector.

So it is not possible. Never mind. Thanks anyway.

pelwell, 2021-11-22:
I'm not a hardware engineer, but the PMIC becoming powered before `GLOBAL_EN` going high is the normal situation - there's an R/C circuit to delay `GLOBAL_EN` in order to allow power to stabilise, and the microcontroller has the ability to delay that signal indefinitely (and drive it low at shutdown to trigger the power-off sequence).

closed 2021-11-22 by pelwell