raspberrypi/firmware #1657: execution of start.elf is delayed by 0.7 seconds

SyotaTanaka, 2021-11-20:
https://github.com/raspberrypi/firmware/commit/1df55790fb191704c0ce630d4d0713a8beb43a7d
This commit seems to slow down execution by about 0.7 seconds.

Is this delay the intended change?
Also, is there any way to avoid this delay?

Device: Raspberry Pi 1 Model B
config.txt: None

pelwell, 2021-11-23:
The delay is due to an intentional change - unplugging your camera would avoid it.

SyotaTanaka, 2021-11-24:
> The delay is due to an intentional change - unplugging your camera would avoid it.

I don't use a camera.
External connection is only USB serial to GPIO.

pelwell, 2021-11-24:
That's interesting. What does `raspi-gpio get 0-3` report? And `tail -3 /proc/cpuinfo`?

SyotaTanaka, 2021-11-24:
```
pi@raspberrypi:~$ raspi-gpio get 0-3
GPIO 0: level=1 fsel=4 alt=0 func=SDA0
GPIO 1: level=1 fsel=4 alt=0 func=SCL0
GPIO 2: level=1 fsel=0 func=INPUT
GPIO 3: level=1 fsel=0 func=INPUT
```
```
pi@raspberrypi:~$ tail -3 /proc/cpuinfo
Revision        : 000d
Serial          : 0000000024168025
Model           : Raspberry Pi Model B Rev 2
```

The measurement was done this way
https://forums.raspberrypi.com/viewtopic.php?t=84734

closed 2024-01-08 by timg236