raspberrypi/firmware #1658: Raspberry Pi 3B+ & Official 7" Display fades to white

mishu28nmv, 2021-11-21:
Hello to all. I just updated my RPi 3b+ to Bullseye and now the display stoped working.
After reboot I see the rainbow square then the raspberry screen (with the version type) and then it fades to WHITE. Sometimes I can SSH, sometimes not.
The display is the official 7" touch screen. I checked all the cables and they seem ok, even the DSI connector.
* Which model of Raspberry Pi? **Pi 3B+**
* Which OS and version? **Bullseye, 2021.10.30 image I think**
* Which kernel version? **latest I think, 5.10**
* Official Power supply
* The image loaded on SD card is fresh
This is my latest config file where I try to disable all. Please HELP.
[config.txt](https://github.com/raspberrypi/firmware/files/7576127/config.txt)

DonAyers, 2021-11-22:
I have this exact problem as well.

mishu28nmv, 2021-11-22:
It is all good now. With that config.txt and with the 4 cables connected from RPi to display (red, black, green and yellow) it's working. And also uncomment all those driver things from config.txt

closed 2021-11-22 by mishu28nmv

6by9, 2021-11-22:
Unless you have an original Pi1 model A or B, then you should ONLY connect the red and black wires to provide power. The I2C wires are only required on those boards as the I2C bus was not wired onto the FFC connector.

mishu28nmv, 2021-11-22:
Then I think it was because that **V3D driver was uncommented**.
Now that I have # in front of those settings it is all good! I will detach the 2 wires as well!
Thank you for your reply!

6by9, 2021-11-22:
Disabling vc4-kms-v3d means you've just disabled all 3D acceleration. 

Duplicate of https://github.com/raspberrypi/linux/issues/4686. There is a known issue on the Pi3 (and possibly earlier) that is still being investigated.
Adding ```dtparam=i2c_vc_baudrate=50000``` to config.txt is the current workaround without killing 3D acceleration.

mishu28nmv, 2021-11-22:
Ok. Thanks for that. I will try. Sorry I'm not that much into programming. And sorry for the duplicate thread.

systeemgod, 2021-11-22:
Adding "dtparam=i2c_vc_baudrate=50000" to /boot/config.txt did not help for me on a Pi2

systeemgod, 2021-11-22:
> Adding "dtparam=i2c_vc_baudrate=50000" to /boot/config.txt did not help for me on a Pi2

disabling vc4-kms-v3d did work. That's acceptable for me as I don't need any 3D for this project anyway.

mishu28nmv, 2021-11-22:
As I have only MagicMirror installed and youtube music, do I need 3d acceleration?