raspberrypi/firmware #1660: Firmware doesn't pick the right DTB for the CM4 with upstream_kernel=y

mripard, 2021-11-26:
**Describe the bug**

On a CM4, when using upstream_kernel=y, it looks like the firmware will try to load the RPi4 device tree (bcm2711-rpi-4-b.dtb) unless the device_tree option is explicitly set.

**To reproduce**

On a CM4, with a config.txt file containing

```
upstream_kernel=y
```

And a boot partition with both device trees compiled from the upstream source tree (bcm2711-rpi-4-b.dtb and bcm2711-rpi-cm4-io.dtb), the device tree used by the kernel will be bcm2711-rpi-4-b.dtb unless `device_tree=bcm2711-rpi-cm4-io.dtb` is set explicitly.

**Expected behaviour**
The firmware should pick the `bcm2711-rpi-cm4-io.dtb` device tree

**Actual behaviour**
The firmware picks the `bcm2711-rpi-4-b.dtb` device tree.

Upstream has used a slightly different naming convention for the CM4, and its support is fairly recent (5.16), so I assume there's just no code to handle the alternative naming yet.

**System**

* Which model of Raspberry Pi?  Compute Module 4
* Which OS and version (`cat /etc/rpi-issue`)? Upstream Debian
* Which firmware version (`vcgencmd version`)? Firmware hash is `564e5f9b852b23a330b1764bcf0b2d022a20afd0`, variant `start`, compiled on 2021-02-25T12:10:40
* Which kernel version (`uname -a`)? v5.16-rc1 and later.

**Additional context**
Upstream discussion about this:
https://lore.kernel.org/netdev/20211022143337.jorflirdb6ctc5or@gilmour/

pelwell, 2021-11-26:
Indeed - the firmware hasn't been updated yet. I'll leave the fallback as bcm2711-rpi-4-b.dtb, since it exists upstream and doesn't result in a completely dead device.

CyrilBrulebois, 2021-11-26:
Similarly, I have to set `device_tree=bcm2837-rpi-cm3-io3.dtb` on the CM3.

Without it, the failure mode is more spectacular than with the CM4: stuck on the rainbow boot screen.

pelwell, 2021-11-26:
@mripard Trial start4/fixup4 here: https://drive.google.com/file/d/13GdEpuoTQ84mIN1jbuewv3eEKIbtTi3O/view?usp=sharing

@CyrilBrulebois Why did nobody report this before? I'll get this fixed as well.

CyrilBrulebois, 2021-11-26:
@pelwell Back when I started working on CM3, I integrated an older version of the `raspi-firmware` package in Debian that shuffled files according to RPi names, and continued using that particular version in a product, successfully.

Since then, I discovered the `raspi-firmware` package [switched to trusting `upstream_kernel` would do the job, and deploying DTB files under their mainline names](https://salsa.debian.org/debian/raspi-firmware/-/commit/165f43a6ca14d240f199e8ab8924e503ca5f427d). Seeing what happens with the CM4 / Pi 4 B, I suspected CM3 support might have been broken in passing, which I've only confirmed recently.

pelwell, 2021-11-26:
Trial start/fixup here: https://drive.google.com/file/d/1CwCXlsp_RV6R1zd9HeA-r6JQ43nsGwy2/view?usp=sharing
This one's untested, but after the CM4 change it should be a no-brainer.

pelwell, 2021-12-01:
The CM1-4 upstream DTB changes are in the latest rpi-update firmware.

CyrilBrulebois, 2021-12-05:
Thanks, @pelwell!

It took me a little longer than I would have hoped for, but I've checked what happens with the updated bootloader files.

Methodology:

- build a Debian 11 image using the usual Debian tooling (https://salsa.debian.org/raspi-team/image-specs): `make raspi_3_bullseye.img raspi_4_bullseye.img` …
- but adjusted to inject all the files normally packaged in the `raspi-firmware` binary package from the `boot/` directory of the https://github.com/raspberrypi/firmware repository;
- and since Debian 11 comes with a 5.10 kernel (which didn't feature a DTB for the CM4), I'm also copying manually the CM4 DTB built from v5.16-rc1 (I know there's no guarantee that it works with a different kernel, but it does).

I've verified that all of those work fine:

- Pi 3 B + Rev 1.3: no apparent regressions;
- Pi CM 3+ Rev 1.0: it used to be blocked on the rainbow boot screen, it now boots fine;
- Pi 4 B Rev 1.4: no apparent regressions;
- Pi CM 4 Rev 1.0: it used to believe it was a Pi 4 B, now it knows it's a CM 4, and both onboard USB and onboard Ethernet are working fine (which they didn't when it believed it was a Pi 4 B).

(Full disclosure: since I had forgotten about the missing CM4 DTB for my first attempt, I've also verified that the bootloader still defaults to the Pi 4 B DTB on the CM4 if the CM4 DTB is missing.)

In other words, everything looks good to me, thanks again!

closed 2021-12-05 by pelwell