raspberrypi/firmware #1661: Pin limitations in gpio-shutdown dtoverlay?

alexwhittemore, 2021-12-03:
I'm trying to hack some functionality into an already-spun CM4 carrier board. I want to use the boot select button to gracefully shut down. Best I can tell, there's no ill effect of toggling nRPI_BOOT high>low once the CM4 is already booted. Since it's attached to GPIO40, I can watch this pin as a trigger for clean shutdown.

In a perfect world, it seems like the cleanest way to do this would be with 
`dtoverlay=gpio-shutdown,gpio-pin=40,active_low=1`
but that doesn't seem to work at all. As a test,
`dtoverlay=gpio-shutdown,gpio-pin=3,active_low=1`
works EXACTLY as expected, though of course on the wrong pin for my application. In fact, the line with gpio-pin=40 behaves exactly as the line with gpio-pin=3, so it seems like the gpio-pin parameter is getting totally ignored for the default of 3.

On the hunch that there was some kind of blessed list of "approved" pins in the dtoverlay, I decompiled gpio-shutdown.dtbo. I'm guessing it's something to do with this chunk:
```
	__overrides__ {
		gpio_pin = < 0x02 0x6770696f 0x733a3400 0x01 0x6272636d 0x2c70696e 0x733a3000 >;
		gpio_pull = < 0x01 0x6272636d 0x2c70756c 0x6c3a3000 >;
		active_low = < 0x02 0x6770696f 0x733a3800 >;
		debounce = < 0x02 0x6465626f 0x756e6365 0x2d696e74 0x65727661 0x6c3a3000 >;
	};
```
but I have to admit, I have very little idea what I'm doing with device trees, so this doesn't make much sense to me. None of those addresses seem to match up to anything in the main blob.

So I guess my question is: is gpio40 indeed "locked out" in gpio-shutdown.dtbo, and if so, what is the mechanism that does it?

pelwell, 2021-12-03:
`-` != `_`.  Try:
```
dtoverlay=gpio-shutdown,gpio_pin=40,active_low=1
```

pelwell, 2021-12-03:
By the way, did you know that you get help on overlays using the `dtoverlay` command in Raspberry Pi OS?
```
Name:   gpio-shutdown

Info:   Initiates a shutdown when GPIO pin changes. The given GPIO pin
        is configured as an input key that generates KEY_POWER events.

        This event is handled by systemd-logind by initiating a
        shutdown. Systemd versions older than 225 need an udev rule
        enable listening to the input device:

                ACTION!="REMOVE", SUBSYSTEM=="input", KERNEL=="event*", \
                        SUBSYSTEMS=="platform", DRIVERS=="gpio-keys", \
                        ATTRS{keys}=="116", TAG+="power-switch"

        Alternatively this event can be handled also on systems without
        systemd, just by traditional SysV init daemon. KEY_POWER event
        (keycode 116) needs to be mapped to KeyboardSignal on console
        and then kb::kbrequest inittab action which is triggered by
        KeyboardSignal from console can be configured to issue system
        shutdown. Steps for this configuration are:

            Add following lines to the /etc/console-setup/remap.inc file:

                # Key Power as special keypress
                keycode 116 = KeyboardSignal

            Then add following lines to /etc/inittab file:

                # Action on special keypress (Key Power)
                kb::kbrequest:/sbin/shutdown -t1 -a -h -P now

            And finally reload configuration by calling following commands:

                # dpkg-reconfigure console-setup
                # service console-setup reload
                # init q

        This overlay only handles shutdown. After shutdown, the system
        can be powered up again by driving GPIO3 low. The default
        configuration uses GPIO3 with a pullup, so if you connect a
        button between GPIO3 and GND (pin 5 and 6 on the 40-pin header),
        you get a shutdown and power-up button. Please note that
        Raspberry Pi 1 Model B rev 1 uses GPIO1 instead of GPIO3.

Usage:  dtoverlay=gpio-shutdown,<param>=<val>

Params: gpio_pin                GPIO pin to trigger on (default 3)
                                For Raspberry Pi 1 Model B rev 1 set this
                                explicitly to value 1, e.g.:

                                    dtoverlay=gpio-shutdown,gpio_pin=1

        active_low              When this is 1 (active low), a falling
                                edge generates a key down event and a
                                rising edge generates a key up event.
                                When this is 0 (active high), this is
                                reversed. The default is 1 (active low).

        gpio_pull               Desired pull-up/down state (off, down, up)
                                Default is "up".

                                Note that the default pin (GPIO3) has an
                                external pullup. Same applies for GPIO1
                                on Raspberry Pi 1 Model B rev 1.

        debounce                Specify the debounce interval in milliseconds
                                (default 100)

```

alexwhittemore, 2021-12-03:
> `-` != `_`. Try:
> 
> ```
> dtoverlay=gpio-shutdown,gpio_pin=40,active_low=1
> ```

UGH. I WAS referencing /boot/overlays/README - didn't know about the dtoverlay command, but that's very neat!

If only I could properly read either. The base-case working exactly as expected by pure happenstance really threw me off the scent of myself. Of course that works perfectly - thanks for the extremely helpful response!

closed 2021-12-03 by pelwell