raspberrypi/firmware #1663: Headless rpi02w not connecting to ssh using disk image from rpi0w

dakschnitzer, 2021-12-07:
I have a disk image, created using the steps described here (https://www.raspberrypi.com/documentation/computers/getting-started.html), from a rpi0w. I would like to run this disk image on a rpi02w.

The disk image contains the OS and various software I've put onto the rpi0w. I've validated that this disk image works on other rpi0w units, can connect to wifi, and ssh. 

When I try using this disk image in a rpi02w, I see that it is booting (LED lights up and shows activity), but it does not appear on my network.

What I've tried to do:
- run update and upgrade on the software while the SD card is in the rpi0w to make sure it's fully up to date. 
- run full-upgrade;
- manually upgrade from Buster to Bullseye following these steps: https://www.tomshardware.com/how-to/upgrade-raspberry-pi-os-to-bullseye-from-buster
- add a fresh wpa_supplicant.conf file to the SD card
- add a fresh ssh empty file to the SD card

None of these steps have resulted in seeing the rpi02w on the network or being able to SSH. After each of these steps, I validated that the SD card still worked fine in the rpi0w.

cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"

Please help!

pelwell, 2021-12-07:
I'm a bit surprised that what you have done so far has not been sufficient. What does the following command return:
```
$ strings /boot/start.elf | grep VC_BUILD
```

pelwell, 2021-12-07:
And also:
```
$ ls -l /lib/firmware/brcm/brcmfmac43436*
```

dakschnitzer, 2021-12-10:
> I'm a bit surprised that what you have done so far has not been sufficient. What does the following command return:
> 
> ```
> $ strings /boot/start.elf | grep VC_BUILD
> ```

VC_BUILD_ID_USER: dom
VC_BUILD_ID_TIME: 15:07:06
VC_BUILD_ID_VARIANT: start
VC_BUILD_ID_TIME: Dec  1 2021
VC_BUILD_ID_HOSTNAME: buildbot
VC_BUILD_ID_BRANCH: stable_20211201
VC_BUILD_ID_PLATFORM: raspberrypi_linux
VC_BUILD_ID_VERSION: 71bd3109023a0c8575585ba87cbb374d2eeb038f (clean)

dakschnitzer, 2021-12-10:
> 

-rw-r--r-- 1 root root 413934 Nov  1 22:07 /lib/firmware/brcm/brcmfmac43436-sdio.bin
-rw-r--r-- 1 root root  11209 Nov  1 22:07 /lib/firmware/brcm/brcmfmac43436-sdio.clm_blob
-rw-r--r-- 1 root root   1613 Nov  1 22:07 /lib/firmware/brcm/brcmfmac43436-sdio.txt
-rw-r--r-- 1 root root 428103 Nov  1 22:07 /lib/firmware/brcm/brcmfmac43436s-sdio.bin
-rw-r--r-- 1 root root   1092 Nov  1 22:07 /lib/firmware/brcm/brcmfmac43436s-sdio.txt

dakschnitzer, 2021-12-10:
> And also:
> 
> ```
> $ ls -l /lib/firmware/brcm/brcmfmac43436*
> ```

-rw-r--r-- 1 root root 413934 Nov  1 22:07 /lib/firmware/brcm/brcmfmac43436-sdio.bin
-rw-r--r-- 1 root root  11209 Nov  1 22:07 /lib/firmware/brcm/brcmfmac43436-sdio.clm_blob
-rw-r--r-- 1 root root   1613 Nov  1 22:07 /lib/firmware/brcm/brcmfmac43436-sdio.txt
-rw-r--r-- 1 root root 428103 Nov  1 22:07 /lib/firmware/brcm/brcmfmac43436s-sdio.bin
-rw-r--r-- 1 root root   1092 Nov  1 22:07 /lib/firmware/brcm/brcmfmac43436s-sdio.txt

dakschnitzer, 2021-12-10:
I ran these commands on the version of the SD Card where i did not upgrade to bullseye. 

I used the SD Card where i ran the full upgrade to bullseye to re-build my entire stack from scratch. I did a clean install of everything and it now works in both the Pi02w and the pi0w.

I still don't understand why my original SD card that was working in the pi0w is not working in the pi02w, even after running updates and upgrades.