raspberrypi/firmware #1664: doesn't boot form usb mass storage when emmc corrupted in cm4

fengpingsh, 2021-12-08:
A cm4 with emmc but no firmware write to emmc.
plug a mass storage usb disk which has flashed the lastet img.

the formware in the usb stick can not be started, here attach the log. Seems that it can't find the usb disk.
Does there any special action I have to take?

root dir cluster 2 sectors 0 entries 0
FAT32 clusters 516190
Read config.txt bytes     1796 hnd 0x00000000
Firmware not found
ERROR: 4
Boot mode: USB-MSD (04) order f256
PCIe timeout: 0x0000008f
USB xHC init failed
Boot mode: NVME (06) order f25
PCIe timeout: 0x0000008f
Failed to open device: 'nvme'
Boot mode: BCM-USB-MSD (05) order f2
XHCI-STOP
xHC ver: 272 HCS: 01000140 0c0000f1 07ff000a HCC: 0220fe65
xHC ports 1 slots 64 intrs 1
XHCI-STOP
xHC ver: 272 HCS: 01000140 0c0000f1 07ff000a HCC: 0220fe65
USBSTS 0
USB MSD timed out after 25 seconds
Boot mode: NETWORK (02) order f
GENET: RESET_PHY

timg236, 2021-12-08:
It looks as though there is no firmware on the USB memory stick or the USB port is not working. Make sure that you don't have the micro-USB cable connected if you are using a CM4IO board.

fengpingsh, 2021-12-27:
that's right, I unplugged the micro-USB cable and it works.
Thanks.

closed 2021-12-27 by fengpingsh