raspberrypi/firmware #1666: HAT-EEPROM with CM3 not working

iluminat23, 2021-12-09:
We have different boards with CM3 and CM3+. With the current issues on getting chip we need to rework many of them. To tackle these variants we want to use an EEPROM on the I2C0 (HAT-EEPROM). We tested this with one of our boards. But this seems not to work. The same with a CM4 based board (IOBOARD and a custom one) works fine.

It would be great if you could add support for "HAT-EEPROMs" for CM3 as well.

JamesH65, 2021-12-09:
As an aside, if you are using the EEPROM to identify the type of board and making software work as per the device found, you can actually identify the board type using the data from cat /proc/cpuinfo. Check out this page. https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#raspberry-pi-revision-codes

If that is not what you need it for, please ignore!

iluminat23, 2021-12-09:
It's not about the CM3 or BCM2835 revision. It is about our own board. We have issues obtaining some chips. And we need to replace them. In turn we need to change the devicetree overlay. To support all boards with the same image and config.txt we wanted to use an "HAT-EEPROM" to choose the correct overlay and to provide additional information.

JamesH65, 2021-12-09:
I did wonder, but didn't want you to miss a trick, so just making sure! EEPROM is a good approach then.

pelwell, 2021-12-09:
`force_eeprom_read=1`?

iluminat23, 2021-12-09:
Hi Phil,

thank you for your fast reply. This works!

Best Regards,
Philipp

closed 2021-12-09 by iluminat23