raspberrypi/firmware #1667: Default core voltage on Pi zero is 1.80V with recent firmware

IanSB, 2021-12-17:
I'm working on this bare metal project for the Pi zero:
https://github.com/IanSB/RGBtoHDMI/wiki

I recently started using newer firmware (bootcode.bin, fixup_cd.dat, start_cd.elf) but the default core voltage has increased from 1.35V to 1.80V on the Pi zero or zeroW on all firmware from the 14-jul-2021 commit.
The most recent one that works for me is the commit on 21-apr-2021

Other pi models (pi1, 2 ,3 zero2) have sensible voltages between 1.20 and 1.30 volts with recent firmware

My software doesn't attempt to change the voltage either programmatically or using config.txt

21-apr-2021 commit:
[image]

14-jul-2021 commit and all later versions:
[image]

pelwell, 2021-12-17:
A voltage scaling bug that only affected Pi Zeros was introduced in May 2021 and fixed on 14th November. Although it reports a voltage of 1.8V, the hardware actually caps this to just under 1.4V - not ideal, but not catastrophic.

Firmwares built since 14th November, include the current [`stable`](https://github.com/raspberrypi/rpi-firmware/tree/stable)  and [`oldstable`](https://github.com/raspberrypi/rpi-firmware/tree/oldstable) branches includes this patch.

IanSB, 2021-12-17:
>Firmwares built since 14th November, include the current stable and oldstable branches includes this patch.

Thanks, I'd been using the files from the most recent raspberry Pi OS image I had which were dated 29/10/21 and I'd been getting some reports of lockups with my latest beta releases which is why I started looking at it. I thought I'd also tested the most recent firmware build but I guess not.

>the hardware actually caps this to just under 1.4V - not ideal, but not catastrophic.

I noticed that vcgencmd returned 1.4V

hoglet67, 2021-12-17:
@pelwell Is there a Github issue for the voltage scaling bug?

pelwell, 2021-12-17:
I think you are reading it.

IanSB, 2021-12-17:
@pelwell 
>the hardware actually caps this to just under 1.4V - not ideal, but not catastrophic.

Has this set the over voltage warranty bit? (The project uses force_turbo=1 but doesn't use overvoltage settings)

popcornmix, 2021-12-17:
> Has this set the over voltage warranty bit? (The project uses force_turbo=1 but doesn't use overvoltage settings)

No, you need an explicit `over_voltage` settings (plus `force_turbo=1`) to set the warranty bit.