raspberrypi/firmware #1668: boot_ramdisk bug at CM4

Sfinx, 2021-12-17:
**Describe the bug**
Seeing boot_ramdisk bug at CM4 - seems like wrong DTB size calculation is performed.

**To reproduce**
Download sample tar from [https://drive.google.com/file/d/1wEINzrBzzL1_MNT8vFScXBpcibtE_6Mb](https://drive.google.com/file/d/1wEINzrBzzL1_MNT8vFScXBpcibtE_6Mb)

Untar, try to boot with 'rpiboot -d .' - you will see something like

```
MESS:00:00:25.045053:0: Device tree loaded to 0x2e345e00 (size 0xd179)
```

Kernel will not boot, start4.elf just hangs

Next dd the same boot.img from tar to CM4 eMMC, boot and see:

```
MESS:00:00:14.214337:0: Device tree loaded to 0x2e345e00 (size 0xd193)
```

Kernel will boot.

**Expected behaviour**
Kernel must boot when using boot_ramdisk, DTB size must be correct is this case.

**Actual behaviour**
Kernel do not boot with boot_ramdisk, DTB size is different

**System**
No raspinfo. Sample boot.img provided

* CM4, 32Gb eMMC, 8Gb RAM/WIFI/BT
* pieeprom-2021-11-22
* 5.15.8 aarch64

peterharperuk, 2021-12-20:
With dtb issues it might be helpful for us if you capture uart logging and add dtdebug=1 into config.txt

Sfinx, 2021-12-20:
Did you tried to boot my example ? The uart output in both cases is _the_ same except the mentioned dtb size. Ok, will supply it. I've noticed that this happens only with 64 bit kernel. 32bit one boots ok in both modes.

peterharperuk, 2021-12-20:
Yes - I can reproduce what you're seeing with the size difference

pelwell, 2021-12-20:
Given that the firmware modifies the DTB before passing it on, and that one of the modifications is an indication of the boot medium, a size difference is not that surprising.

The output of
```
$ dtc -I fs /proc/ device-tree
```
in both cases should help to clear up source of the change.

Sfinx, 2021-12-20:
: dtc -I fs /proc/device-tree

The kernel hangs at memory init stage - no chance to reach the init and issue this command

Sfinx, 2021-12-20:
Good and bad logs attached.
[badlog.txt](https://github.com/raspberrypi/firmware/files/7745223/badlog.txt)
[goodlog.txt](https://github.com/raspberrypi/firmware/files/7745227/goodlog.txt)

Sfinx, 2021-12-20:
> Yes - I can reproduce what you're seeing with the size difference. But I don't really see any difference in boot behaviour - unless I'm missing something. So having your logs might help.

Are the boot_ramdisk kernel fully booted at your case ? If yes - what the CM4 model do you have ?

pelwell, 2021-12-20:
> no chance to reach the init and issue this command

Understood - that comment was aimed at @peterharperuk.

peterharperuk, 2021-12-20:
Thanks - here's my log for reference [boot_test_rpiboot.txt](https://github.com/raspberrypi/firmware/files/7745453/boot_test_rpiboot.txt). It's a CM4 Lite - I wouldn't expect it to make a difference but I'll try and find a heavy.

peterharperuk, 2021-12-20:
I don't get a command prompt - presumably I'd have to mess with init to run dtc. But it's working for me so I'm not sure what I can do to help further. Do you have a camera attached? Maybe that has some effect...?

Sfinx, 2021-12-20:
No camera, even no any additional peripherial at all - it is pure CM4 inserted in Waveshare i/o board with hdmi && usb touch && wired ethernet  attached only. Even if it would attached - how do you explain that pure boot works but boot_ramdisk do not ?

BTW: the kernel version do not matters - I'm seeing this at stock 5.10.x that is currently mainstream here.

Sfinx, 2021-12-20:
It is pity that you do not have the full CM4 board variants. I thought that firmware team can afford this...

peterharperuk, 2021-12-20:
I've tried a heavy - it still works. But the fact you're using a Waveshare board is new information

Sfinx, 2021-12-20:
Hmm, so you are passing the memory init ? It is Mini Base Board (B) - simplest one. No serious addons there except USB hub & fan controller.

pelwell, 2021-12-20:
@peterharperuk There's a Mini Base Board B on my desk (with mods to make it reboot reliably), in case that helps.

Sfinx, 2021-12-20:
Are these mods h/w ones ? Will be happy if you can share them so I can try. Thanks !

peterharperuk, 2021-12-20:
It's running init - looks the same as your good log.

I extracted the image again and repeated the test and it does fail to boot unless you remove the "typo in earlycon". Does it work for you when the zero is there?

Sfinx, 2021-12-20:
It is still unclear for me why the same board & kernel is booting ok from mmc but do not boot from ramdisk. What can be the possible reason ? My logic here is simple - we do not change the kernel & board but change the boot way. So the bug is in boot way.

Sfinx, 2021-12-20:
> It's running init - looks the same as your good log.
> 
> I extracted the image again and repeated the test and it does fail to boot unless you remove the "typo in earlycon". Does it work for you when the zero is there?

Adding forgotten zero just allows you to see the early kernel log. It doesn't influence anything as uart console is "write and forget" thing"

pelwell, 2021-12-20:
> uart console is "write and forget" thing

UARTs don't have infinite FIFOs, so it is necessary to poll for free space before adding new characters for output.

Sfinx, 2021-12-20:
> > It doesn't influence anything
> 
> Well it seems to stop boot via RPIBOOT working for me. Does it work for you if you put the zero back? I'm just interested in getting confirmation of what I'm seeing.

Hmm, do not catch this "stop boot via RPIBOOT working for me". Is it stopped or working ?

The logs that I've posted here contain correct pl011 address  (0xfe201000) :+1: 
```
[    0.000000] earlycon: pl11 at MMIO32 0x00000000fe201000 (options '')
```

Sfinx, 2021-12-20:
> > uart console is "write and forget" thing
> 
> UARTs don't have infinite FIFOs, so it is necessary to poll for free space before adding new characters for output.

I think FIFO is used with RTS/CTS. For dumb write the UART_TX_READY flag is checked - it is just time needed to send one char:

```
static void pl010_console_putchar(struct uart_port *port, int ch)
{
        struct uart_amba_port *uap =
                container_of(port, struct uart_amba_port, port);
        unsigned int status;

        do {
                status = readb(uap->port.membase + UART01x_FR);
                barrier();
        } while (!UART_TX_READY(status));
        writel(ch, uap->port.membase + UART01x_DR);
}
```

peterharperuk, 2021-12-20:
Then I'm out of ideas - it works for me and gets as far as running init.

BTW: The hw mod Phil is referring to is discussed in here https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/188 The comment on 9-Nov seems useful.

pelwell, 2021-12-20:
FIFO use makes no difference - the TX READY flag is what matters. An I/O address or memory location with a zero in that location means your boot stalls.

Sfinx, 2021-12-21:
Debugging further: weird, kernel never returns from arm64_memblock_init()

Sfinx, 2021-12-21:
> I don't get a command prompt - presumably I'd have to mess with init to run dtc. But it's working for me so I'm not sure what I can do to help further. Do you have a camera attached? Maybe that has some effect...?

When you are saying about camera - does it means that CM4 bootloader reads something from I2C0 ? This board has RTC & Fan controller at I2C0. I guess it could read something and misinterpret this data in 64 bit mode

Sfinx, 2021-12-21:
Got it working when passing mem=3G to kernel ! Any thoughts ?

Sfinx, 2021-12-21:
> Then I'm out of ideas - it works for me and gets as far as running init.

Can you please tell me - how much memory does your CM4 has ? Mine is 8Gb

Sfinx, 2021-12-21:
from RPi4_UEFI_Firmware project:

```
* A 3 GB RAM limit is enforced __by default__, even if you are using a Raspberry Pi 4
  model that has 4 GB or 8 GB of RAM, on account that the OS **must** patch DMA access,
  to work around a hardware bug that is present in the Broadcom SoC.  
```

pelwell, 2021-12-21:
And with `total_mem=3072`?

I find it strange that your boot.txz includes a fixup4.dat but not a start4.elf. If you are relying on an existing start4.elf then supplying a fixup4.dat is unlikely to be helpful, and possibly harmful.

Sfinx, 2021-12-21:
total_mem=3072 freezes in the same way - no return from arm64_memblock_init(). 

You can safely delete fixup4.dat. Nor it nor start4.elf is used when booting ramdisk using rpiboot -d. The start4.elf/fixup4.dat must be present _inside_ boot.img

Sfinx, 2021-12-21:
Additional info - mem=3956M works but mem=3957M doesn't work anymore. This is 0xf7400000

timg236, 2021-12-21:
If start4.elf is missing it will failover to start.elf but that’s probably not what you want. Fix the image to have a start4.elf and fixup4.dat otherwise the GPU mem config will be different to what you are expecting.

Sfinx, 2021-12-21:
Seems like you've missed the point again. start4.elf is not used for loading boot.img:

```
# strace ../usbboot/rpiboot -d . > out 2>&1
# grep start out 
<nothing>
```
start4.elf & fixup4.dat is definitely present inside boot.img

timg236, 2021-12-21:
> Seems like you've missed the point again. start4.elf is not used for loading boot.img:
> 
> ```
> # strace ../usbboot/rpiboot -d . > out 2>&1
> # grep start out 
> <nothing>
> ```
> 
> start4.elf & fixup4.dat is definitely present inside boot.img

See previous comment from pelwell who said the file seemed to be missing. Anyway, this doesn’t seem to be related to RAM disks / rather a UEFI config issue

Sfinx, 2021-12-21:
UEFI is not used here. So the image boots ok from SD card. Still waiting the reply from peterharperuk about his CM4 memory size. I'm now pretty sure that nobody here checked the 64bit ramdisk option with CM4 modules > 4GB in size.

timg236, 2021-12-21:
I’m pretty sure we did and since the VPU can only address the first gigabyte and the RAM disk is at 512MB that’s not relevant

Sfinx, 2021-12-21:
> I’m pretty sure we did and since the VPU can only address the first gigabyte and the RAM disk is at 512MB that’s not relevant

Where I can see the results of this particular boot_ramdisk test at 8GB CM4 module ? Please point me to it

pelwell, 2021-12-21:
> Seems like you've missed the point again

I think you need to watch your tone unless you want to be deprioritised.

peterharperuk, 2021-12-21:
I've tested 1GB and 4GB CM4 devices. I don't have access to 8GB atm as I'm not in the office. I assume you've fixed your cmdline.txt and it's still failing to boot? In which case I'll retest asap.

Sfinx, 2021-12-21:
> > Seems like you've missed the point again
> 
> I think you need to watch your tone unless you want to be deprioritised.

do you think you can scare me with de-prioritization? All you have achieved is that I no longer seeing the rpi as a trustworthy and adequately supported platform - just a closed thing within itself with people do not bothering to read code but stating something unexisting. Farewell

Sfinx, 2021-12-21:
> I've tested 1GB and 4GB CM4 devices. I don't have access to 8GB atm as I'm not in the office. I assume you've fixed your cmdline.txt and it's still failing to boot? In which case I'll retest asap.

I've changed the subject - I was really hoped you used the exact device that was mentioned in bug report but it is appeared that was used lite first, next non-8GB version. Anyway I do not care anymore about this bug as switched from rpi so you can close this

closed 2021-12-21 by JamesH65