raspberrypi/firmware #1669: CPU frequency is frozen to max since latest stable firmware

MichaIng, 2021-12-17:
**Describe the bug**
Since latest stable firmware/kernel `5.10.63 #1488`, the CPU frequency is frozen to max on some RPi modules, RPi 4 and RPi Zero (1) W at least, but not on RPi 2. On RPi Zero W the same is true with `5.10.85+ #1501`. It is not a CPU governor issue, but only one available frequency is reported:
```
# cat /sys/devices/system/cpu/cpufreq/policy0/scaling_available_frequencies
1100000
## without overclocking
# cat /sys/devices/system/cpu/cpufreq/policy0/scaling_available_frequencies
1000000
```
This is true with or without overclocking settings.

**To reproduce**
1. Update an RPi 4 or Zero (1) W to lastest stable kernel/firmware packages.
2. `reboot`
3. `cat /sys/devices/system/cpu/cpufreq/policy0/scaling_available_frequencies`

**Expected behaviour**
Available frequencies should range from 600/700 MHz to the maximum, based on overclocking, e.g. 1000 MHz on RPi Zero (1) W and 1500 MHz on RPi 4

**Actual behaviour**
Only the highest CPU frequency is available, depending on overclocking.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? RPi 4 and RPi Zero (1) W
* Which OS and version (`cat /etc/rpi-issue`)? Debian Bullseye
* Which firmware version (`vcgencmd version`)?
```
Dec  1 2021 11:48:03
Copyright (c) 2012 Broadcom
version be18b6271a0f99f896fc81249da5da6658422558 (clean) (release) (start)
# and others
```
* Which kernel version (`uname -a`)?
```
Linux Amizero 5.10.85+ #1501 Thu Dec 16 15:58:52 GMT 2021 armv6l GNU/Linux
# or
Linux DietPi 5.10.63-v7l+ #1488 SMP Thu Nov 18 16:15:28 GMT 2021 armv7l GNU/Linux
# and others
```

**Logs**
<details>
<summary>dmesg (nothing below warn severity)</summary>

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.85+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1501 Thu Dec 16 15:58:52 GMT 2021
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] random: fast init done
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Reserved memory: created CMA memory pool at 0x0bc00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000] On node 0 totalpages: 114688
[    0.000000]   Normal zone: 1008 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 114688 pages, LIFO batch:31
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113680
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=Composite-1:720x480@60i smsc95xx.macaddr=B8:27:EB:51:02:04 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  root=PARTUUID=0e2887fb-02 rootfstype=ext4 rootwait fsck.repair=yes net.ifnames=0 console=tty1
[... 278 lines cut ...]
[   16.337366] random: crng init done
[   16.337396] random: 7 urandom warning(s) missed due to ratelimiting
[   18.581628] vc4-drm soc:gpu: bound 20400000.hvs (ops vc4_hvs_ops [vc4])
[   18.597021] Registered IR keymap rc-cec
[   18.597316] rc rc0: vc4 as /devices/platform/soc/20902000.hdmi/rc/rc0
[   18.597629] input: vc4 as /devices/platform/soc/20902000.hdmi/rc/rc0/input1
[   18.616465] vc4-drm soc:gpu: bound 20902000.hdmi (ops vc4_hdmi_ops [vc4])
[   18.616955] vc4-drm soc:gpu: bound 20806000.vec (ops vc4_vec_ops [vc4])
[   18.617569] vc4-drm soc:gpu: bound 20004000.txp (ops vc4_txp_ops [vc4])
[   18.618100] vc4-drm soc:gpu: bound 20206000.pixelvalve (ops vc4_crtc_ops [vc4])
[   18.618616] vc4-drm soc:gpu: bound 20207000.pixelvalve (ops vc4_crtc_ops [vc4])
[   18.626002] vc4-drm soc:gpu: bound 20807000.pixelvalve (ops vc4_crtc_ops [vc4])
[   18.626414] vc4-drm soc:gpu: bound 20c00000.v3d (ops vc4_v3d_ops [vc4])
[   18.690969] [drm] Initialized vc4 0.0.0 20140616 for soc:gpu on minor 0
[   18.725482] Console: switching to colour frame buffer device 90x30
[   18.745099] vc4-drm soc:gpu: [drm] fb0: vc4drmfb frame buffer device
[   21.727694] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[   23.362300] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   33.123511] cam1-reg: disabling
[   33.123541] cam-dummy-reg: disabling
```
</details>

**Additional context**
Originally reported here: https://github.com/MichaIng/DietPi/issues/5088

popcornmix, 2021-12-17:
Not on a default install of RPiOS on Pi4. e.g.
```
pi@pi4:~ $ vcgencmd version ; uname -a ; cat /sys/devices/system/cpu/cpufreq/policy0/scaling_available_frequencies
Nov 18 2021 16:16:49 
Copyright (c) 2012 Broadcom
version d9b293558b4cef6aabedcc53c178e7604de90788 (clean) (release) (start)
Linux domnfs 5.10.63-v7l+ #1488 SMP Thu Nov 18 16:15:28 GMT 2021 armv7l GNU/Linux
600000 700000 800000 900000 1000000 1100000 1200000 1300000 1400000 1500000 1600000 1700000 1800000 
```
What do you have in config.txt? I suspect it's a setting like, force_turbo, initial_turbo, arm_freq, arm_freq_min etc, so some change to the default cpufreq governor.

MichaIng, 2021-12-17:
It is the `rpi-firmware`/master version applied when testing the `vc4-kms-v3d,noaudio` issue on the other thread. The link I added above contains cases with stable firmware package.

I tested with and without any overclocking settings (including `initial_turbo`, and I never use `force_turbo` on my boards as a principle 😉). And again same was done with RPi 4 on the linked issue, here the `config.txt`: https://github.com/MichaIng/DietPi/issues/5088#issuecomment-996167320

What is true is that at least in case of the RPi 4 it is not using Raspbian but Debian. Though the same `archive.raspberrypi.org`, so that shouldn't make a difference.

@ayo-x
If you find time, could you try to comment every setting in `config.txt`, reboot and check again, just to rule that out:
```sh
cp -a /boot/config.txt{,_bak}
sed -Ei 's/^([^#])/#\1/' /boot/config.txt
reboot
# after reboot
vcgencmd version ; uname -a ; cat /sys/devices/system/cpu/cpufreq/policy0/scaling_available_frequencies
```
To revert:
```sh
cp -a /boot/config.txt{_bak,}
```

popcornmix, 2021-12-17:
Adding `noaudio` had no effect. Nor did `initial_turbo=1`
`force_turbo=1` does have the expected effect:
```
pi@pi4:~ $ cat /sys/devices/system/cpu/cpufreq/policy0/scaling_available_frequencies
1800000 
```
as does `arm_freq_min=1800`.

MichaIng, 2021-12-17:
Just downgraded kernel/firmware on RPi Zero W again and commented all config.txt settings:
```
2021-12-17 19:21:12 root@Amizero:~# vcgencmd version; uname -a ;cat /sys/devices/system/cpu/cpufreq/policy0/scaling_available_frequencies; vcgencmd get_config int
Nov 18 2021 16:18:09
Copyright (c) 2012 Broadcom
version d9b293558b4cef6aabedcc53c178e7604de90788 (clean) (release) (start)
Linux Amizero 5.10.63+ #1488 Thu Nov 18 16:14:04 GMT 2021 armv6l GNU/Linux
1000000
aphy_params_current=547
arm_freq=1000
arm_freq_min=700
audio_pwm_mode=514
config_hdmi_boost=5
core_freq=400
disable_auto_turbo=1
disable_commandline_tags=2
display_hdmi_rotate=-1
display_lcd_rotate=-1
dphy_params_current=547
dvfs=3
enable_tvout=1
force_eeprom_read=1
force_pwm_open=1
force_turbo=1
framebuffer_ignore_alpha=1
framebuffer_swap=1
gpu_freq=300
ignore_lcd=1
init_uart_clock=0x2dc6c00
max_framebuffers=-1
over_voltage_avs=0x249f0
pause_burst_frames=1
program_serial_random=1
sdram_freq=450
total_mem=512
hdmi_force_cec_address:0=65535
hdmi_force_cec_address:1=65535
hdmi_pixel_freq_limit:0=0x9a7ec80
```
Ehm, did you just successfully replicate this issue as well?
> pi@pi4:~ $ cat /sys/devices/system/cpu/cpufreq/policy0/scaling_available_frequencies
1800000

Only one (the highest) frequency!

Yes `noaudio` is completely unrelated to this issue, obviously every other `config.txt` setting is as well, though I can test the default one shipped with RPi OS again.

popcornmix, 2021-12-17:
I can only replicate your issue if I add `force_turbo=1` or `arm_freq_min=1800` (which is the expected behaviour as only 1800 is possible).

What does:
```
vcgencmd get_config int
```
report?

MichaIng, 2021-12-17:
> if I add `force_turbo=1` or `arm_freq_min=1800`

Ah yes that makes sense of course. However none of us who is facing it sets any of those two.

See `vcgencmd get_config int` above on RPi Zero W with all settings in `config.txt` commented. `arm_freq_min=700` is reported correctly but not effective.

MichaIng, 2021-12-17:
Oh what? `force_turbo=1` is shown in my `vcgencmd get_config int`, but it is definitely not touched!

MichaIng, 2021-12-17:
Found it: We have the following comment (!) in our default `config.txt`: https://github.com/MichaIng/DietPi/blob/8c7254e/config.txt#L80
```
# NB: "enable_uart=1" will forcefully set "core_freq=250" on RPi models with onboard WiFi, unless "force_turbo=1" is set.
```
Removing it solves the issue, re-adding it creates it again. Interestingly:
```
# force_turbo=1
# "force_turbo=1"
```
do not cause the issue. I also removed and re-added the newline, suspecting Windows CR LF line endings being the culprit, but it isn't. Last change we did on this comment was: https://github.com/MichaIng/DietPi/commit/a1d1cee
Copy&pasting the old text solves the issue, copy&pasting the new text reintroduces it 😄. Now that question which magic pattern turns a comment into an effective setting? Highly problematic of course since this may have lead to void warranties already. Though to void the warranty `over_voltage=7` or higher is required in combination with `force_turbo=1`?

MichaIng, 2021-12-17:
Fixed for new images with: https://github.com/MichaIng/DietPi/commit/934459f

MichaIng, 2021-12-17:
So to sum it up:
- The `config.txt` parser internally somehow applies a line break at column/character 100.
- Hence if by change any valid setting is starting in a comment exactly at line 100, it becomes an effective setting.
- Interesting in case of `force_turbo=1`, this means that `force_turbo=1" other text` is still interpreted like `force_turbo=1`, instead of being invalid, which is an issue in itself as well?
- So to avoid any havoc, any (commnt) lines in `config.txt` must not be longer than 99 characters.

While the `config.txt` shipped by RPi OS has a lower line length cap already, and for readability it is generally good practice, this is not obvious and I'd call it as very serious bug, though one that has any effect only in very rare unlucky cases.

6by9, 2021-12-20:
https://www.raspberrypi.com/documentation/computers/config_txt.html#file-format

> There is a 98-character line length limit (previously 78) for entries - any characters past this limit will be ignored.

If it isn't ignored then that is a bug, but not supporting lines over 98 chars is not.

MichaIng, 2021-12-20:
Thanks for the link. Makes sense, also since I'm not aware of any setting with could reach that length, probably an overlay when defining every possible parameter with a long value.

6by9, 2021-12-20:
If you need to specify a load of parameters to a single overlay, then you can use ```dtparam=``` on a second line to act as a continuation, eg
```
dtoverlay=foo,param1=1234,param2=2345,param3=3456
dtparam=param4=4567,param5=5678
```
Documented at https://www.raspberrypi.com/documentation/computers/configuration.html#part3.2

MichaIng, 2021-12-20:
Good to know, not that I ever faced a situation where it would have been required, but also I do only use and work with a small subset of all overlays and especially their parameters.

popcornmix, 2021-12-21:
If you want to test [here](https://drive.google.com/uc?id=1CCxkdyMsCS_zczzPgmQQLJCijib_JKVP&export=download) is a firmware that should handle overly long config.txt settings correctly.

MichaIng, 2021-12-21:
Many thanks for the quick update, indeed it solves the issue 👍.

popcornmix, 2021-12-21:
Thanks for testing. Will be in next rpi-update (and apt in future).

popcornmix, 2021-12-22:
Fix now in rpi-update firmware.

MichaIng, 2022-01-01:
@popcornmix 
As recognised in this context, at least for binary settings trailing characters are still ignored, hence it is e.g. possible to enable the primary UART (on onboard WiFi models) via:
```
enable_uart=1abc
```
Not critical, but I guess this is actually not intended, is it? I'd vote for being strict in such cases and skip such as invalid setting, at least when there is no space and/or hash as inline comment (which are at least not explicitly mentioned in the documentation as supported).

MichaIng, 2022-01-04:
Okay, agreed. Difficult to remove something that worked in the past, even when not explicitly intended/supported. The UART case is actually a major argument as there may be systems which are accessible via UART only, and having it disabled after a firmware upgrade + reboot, because such an inline comment suddenly renders the setting invalid, would be pretty bad.

closed 2022-02-12 by MichaIng