raspberrypi/firmware #1671: No boot, 7 blinks after updating firmware in the last commit

gentoo-root, 2021-12-24:
**Describe the bug**

Commit 536baf33ae9007d027bffc8f42cb603664ecb5c4 brought the new firmware files (fixup_x.dat, start_x.elf). Updating these files prevents my Raspberry Pi 3B v1.2 from booting, the green LED blinks 7 times in a loop.

Commit 1a0297bfbf7c970f657c11edc42b22a671170d18, which has the previous version of firmware, works fine (tested by replacing the two files mentioned above).

**To reproduce**

There are the following files on by first partition on the microSD card:

- bootcode.bin, fixup_x.dat, start_x.elf taken from this repository.
- cmdline.txt, config.txt.
- dtbs - a directory with device tree files and overlays.
- kernel8.img - a custom build of 64-bit linux 5.7.
- rootfs.img - a custom initramfs.
- A bunch of text files, unrelated to the boot process.

The contents of config.txt:

```
device_tree=dtbs/broadcom/bcm2710-rpi-3-b.dtb
overlay_prefix=dtbs/overlays/
initramfs rootfs.img followkernel
start_x=1
enable_uart=1
gpu_mem=512
```

**Expected behaviour**
Linux should boot with the firmware taken from commit 536baf33ae9007d027bffc8f42cb603664ecb5c4.

**Actual behaviour**
Linux doesn't boot with the firmware taken from commit 536baf33ae9007d027bffc8f42cb603664ecb5c4, the green LED blinks 7 times in a loop. However, linux does boot with the previous version of the firmware taken from commit 1a0297bfbf7c970f657c11edc42b22a671170d18.

**System**

Raspberry Pi 3B v1.2.
Custom OS.
`Linux version 5.7.19-v8+ (gcc version 8.3.0 (GCC), GNU ld (GNU Binutils) 2.34) #1 SMP PREEMPT`

paralin, 2021-12-24:
Raspberry pi 4 does not boot with "firmware not found" error.

paralin, 2021-12-25:
Note: my issue might be related to buildroot only.

popcornmix, 2021-12-26:
Hi - no immediate idea what has broken this for you (and apparently not for others). I'm away so only have limited ability to help.
I have a [test firmware](https://drive.google.com/uc?id=17p7mQrVnc-Yxrv31y7y6Zvb1mcQwVoic&export=download) that has one commit reverted ("ldconfig: Discard subsequent chunks from a truncated line") which may narrow down the problem if you can confirm if this build is good/bad.

gentoo-root, 2021-12-29:
@popcornmix, I tested your test firmware, and it doesn't work either: the same 7 blinks of the green LED.

popcornmix, 2021-12-30:
Thanks, that means it's not ("ldconfig: Discard subsequent chunks from a truncated line").
It's not obvious which of the other commits would cause this regression.
Can you provide minimal files that show the 7 blinks issue?
i.e. config.txt, rootfs.img, bcm2710-rpi-3-b.dtb (and whatever else is needed).

graysky2, 2022-01-01:
Not sure if anyone called this out, but one Arch ARM user is [reporting](https://archlinuxarm.org/forum/viewtopic.php?f=15&t=15759)  this bug affecting specific models:
> RPi4 and Zero W are fine but 3B+, 3B and Zero 2 W stuck at boot with 7 flashes.

bravebug, 2022-01-01:
The same problem on rpi3b with Archlinuxarm:
```
uname -a
Linux charm 5.15.5-1-ARCH #1 SMP Wed Dec 1 01:26:36 UTC 2021 aarch64 GNU/Linux
```
Boot was broken after update raspberrypi-bootloader --> 20211222

andyten, 2022-01-02:
The same problem on rpi3b with Archlinuxarm:

Boot was broken after update raspberrypi-bootloader --> 20220101

pelwell, 2022-01-04:
Current top-of-tree rpi-update kernel + firmware (5.10.89) and the above-mentioned 5.10.88 release both work on a 3B+ with RPiOS Bullseye.

Can someone with a serial cable and putty/minicom etc. try with `enable_uart=1` and `uart_2ndstage=1` in config.txt and post the output?

pelwell, 2022-01-04:
The 64-bit kernel location is broken for BCM2837 - an explicit `kernel=kernel8.img` fixes it for me.

popcornmix, 2022-01-04:
rpi-update firmware includes the fix for kernel8.img.

pelwell, 2022-01-05:
Closing in anticipation of a successful outcome.

closed 2022-01-05 by pelwell

graysky2, 2022-01-05:
> Closing in anticipation of a successful outcome.
@pelwell - At least [one user](https://archlinuxarm.org/forum/viewtopic.php?f=15&t=15759#p68508) confirmed it :+1:

bravebug, 2022-01-05:
```
$ uname -a
Linux charm 5.15.5-1-ARCH #1 SMP Wed Dec 1 01:26:36 UTC 2021 aarch64 GNU/Linux
$ pacman -Qi raspberrypi-bootloader | grep Version
Version         : 20220104-1
```
boot process is fine

andyten, 2022-01-06:
Raspberry Pi 3 Model B Plus Rev 1.3

`uname -a
Linux alarm 5.10.90-1-rpi-ARCH #1 SMP PREEMPT Wed Jan 5 14:52:08 UTC 2022 aarch64 GNU/Linux
`

boot process is fine