raspberrypi/firmware #1672: RPi 3 Mod. B+ vc4-kms-v3d: vc4-drm soc:gpu: [drm] *ERROR* Timed out waiting for commit

raoulbhatia, 2022-01-02:
Similar to #1648 and similar reports, I have display issues with `dtoverlay=vc4-kms-v3d` on my `Raspberry Pi 3 Model B Plus Rev 1.3`.

Switching back to `dtoverlay=vc4-fkms-v3d` (**f**kms!) makes the RPi boot successfully.

My issue is with recent LibreELEC builds.  But given the other bug reports, I figured it is OK to file the issue here.

**To reproduce**
Install `LibreELEC-RPi2.arm-11.0-nightly-20220101-0e04fbf.tar` from https://test.libreelec.tv/

**Expected behaviour**
1. System boots
2. Kodi starts
3. Kodi home screen is displayed

**Actual behaviour**
1. System is stuck at "LibreELEC boot screen" (i.e. showing `waiting for network`).
2. System is reachable over the network.

`dmesg` excerpt
```
[ 1302.693532] [drm:drm_atomic_helper_wait_for_flip_done] *ERROR* [CRTC:72:crtc-3] flip_done timed out
[ 1312.934063] [drm:drm_crtc_commit_wait] *ERROR* flip_done timed out
[ 1312.934121] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CRTC:72:crtc-3] commit wait timed out
[ 1322.961226] [drm:drm_crtc_commit_wait] *ERROR* flip_done timed out
[ 1322.961284] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:73:plane-4] commit wait timed out
[ 1332.988372] [drm:drm_crtc_commit_wait] *ERROR* flip_done timed out
[ 1332.988434] vc4-drm soc:gpu: [drm] *ERROR* Timed out waiting for commit
```

The mentioned workarounds i.e. adding `video=HDMI-A-1:1920x1080M@60` to cmdline.txt do **not** work.

The only solution is to switch back to `dtoverlay=vc4-kms-v3d`.

**System**
FYI `raspinfo` not available on LibreELEC.

`Raspberry Pi 3 Model B Plus Rev 1.3` directly connected to a `Samsung 43" QLED 4K Q60T (2020)` (identified as "PC").
```
# vcgencmd version
Dec  1 2021 11:48:58
Copyright (c) 2012 Broadcom
version be18b6271a0f99f896fc81249da5da6658422558 (clean) (release) (start_x)

# uname -a
Linux LibreELEC 5.15.10 #1 SMP Sat Jan 1 17:25:10 CET 2022 armv7l GNU/Linux

# find /sys/devices/platform/ -name edid
/sys/devices/platform/soc/soc:gpu/drm/card0/card0-Writeback-1/edid
/sys/devices/platform/soc/soc:gpu/drm/card0/card0-HDMI-A-1/edid

# cat '/sys/devices/platform/soc/soc:gpu/drm/card0/card0-HDMI-A-1/edid' | openssl base64
AP///////wBMLRZwAA4AAQEeAQOAXzZ4Cqgzq1BFpScNSEi974BxT4HAgQCBgJUA
qcCzANHABHQAMPJwWoCwWIoAUB10AAAeVl4AoKCgKVAwIDUAUB10AAAaAAAA/QAY
Sw9RHgAKICAgICAgAAAA/ABTQU1TVU5HCiAgICAgAVoCA07wUl8QHwQTBRQgISJd
XmJkBxYDEiwJBwcVB1BXBwBnVACDAQAA4gBP4wUDAW4DDAAgALg8IACAAQIDBOMG
DQHlDmBhZWblAYuEkAECOoAYcTgtQFgsRQBQHXQAAB4AAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAARQ==
```

**Logs**
See above.

pelwell, 2022-01-04:
Possibly related to https://github.com/raspberrypi/linux/pull/4808#issuecomment-1005125491

raoulbhatia, 2022-01-07:
I can confirm this issue is resolved on the latest LibreELEC 11 build that incorporates the change from raspberrypi/linux#4808 (Kernel 5.15.13 for RPi)

```
LibreELEC (community): nightly-20220107-00c51c7 (RPi2.arm)

LibreELEC:~ # uname -a
Linux LibreELEC 5.15.13 #1 SMP Fri Jan 7 15:26:42 CET 2022 armv7l GNU/Linux

LibreELEC:~ # cat /flash/cmdline.txt
boot=/dev/mmcblk0p1 disk=/dev/mmcblk0p2 quiet video=HDMI-A-1:1920x1080M@60

LibreELEC:~ # grep vc4-kms-v3d /flash/distroconfig.txt
dtoverlay=vc4-kms-v3d
```

Feel free to close this issue.

closed 2022-01-07 by popcornmix

raoulbhatia, 2022-01-07:
Thanks @popcornmix 🇬🇧