raspberrypi/firmware #1674: Raspberry Pi 4 Model B won't boot

canaryGrapher, 2022-01-08:
**The bug**
No matter what Distro I use (including Raspberry Pi OS - latest), my Raspberry Pi 4 Model B (4GB RAM) won't boot. I keep getting this error screen for a split second, and then it disappears, and the device shuts down, repeating the step.
[image]

**To reproduce**
I have no idea how it got to this. I kept aside my device for a few months and then it won't boot. I tried reflashing the bootloader and I got the green screen, but even then, the issue persists.

**Expected behaviour**
After booting my device, it should load the Operating system, but it doesn't. I have tried with Pop OS, Raspberry Pi OS, and even Ubuntu Server

**Actual behaviour**
[image]

blitztide, 2022-01-17:
What's the contents of your config.txt file?

WombleWoo7547, 2022-04-22:
looks like your bootloader is corrupted. in the raspberry pi imager go "Misc Utility Images">"Bootloader">"SD Card Boot". Flash that to a card and wait for a green screen. Try booting again and it will hopefully work.

closed 2024-01-08 by timg236