raspberrypi/firmware #1675: rpi-firmware, arm-trusted-firmware, and u-boot : expectations

colemickens, 2022-01-10:
1. I want to kexec, I think this require PSCI and I think that requires ARM-TF.
2. There seems to be an ARM-TF for RPI4 in the reference implementation.

I have successfully built the ARM-TF bl31.bin for the rpi4 and booted into u-boot with it.

However, any u-boot boot method fails with "reserving fdt region failed".

I also think I understand the ARM-TF will modify the FDT further. Is it possible that u-boot doesn't know what to do with this? (The kernel+dtb that *it* is loading is the 5.15 upstream kernel+rpi4 dtb)

What should my expectations around this be? Thanks!

colemickens, 2022-01-10:
After disabling the new early watchdog, it actually seems like it's booting despite this (soft?) error in u-boot,... but only with a single core. I get messages after the kernel loads about not being able to bring up the other CPUs.

This is with the 2022-01-06 firmware and the 2021-12-02 eeprom.

pelwell, 2022-01-10:
I'm afraid that's a question for either the TF or U-boot maintainers - we don't support either - but from a brief search it seems the TF (bl31.bin) modifies the FDT from the firmware before passing it on. How have you configured U-boot's FDT handling? I would recommend setting it up to use the firmware's (and in your case TF's) FDT, unless you have a good reason not to.

colemickens, 2022-01-10:
If I edit my extlinux.conf and prevent it from loading its own tree from fdtdir then it boots with/on all cores. I have not had success with kexec, however, when using this (which would only work on nixos systems):

```
b=/run/booted-system
sudo kexec -l \
  "${b}/kernel" \
  --dtb "/sys/firmware/fdt" \
  --initrd "${b}/initrd" \
  --append "$(cat ${b}/kernel-params)"

sudo systemctl kexec
```

feel free to close this but for now I'll leave it open to see if you/others have more advice.

closed 2024-01-08 by timg236