raspberrypi/firmware #1676: [Question] Firmware maintainer of BCM43430/1 in Raspberry Pi 3?

Matheus-Garbelini, 2022-01-11:
Hi, I'm trying to understand to what company to report a firmware vulnerability (DoS) for Raspberry Pi 3.
I understand that at some point the Cypress/Infineon acquired a part of broadcom may be responsible for further updates to BCM43430 chipset used in Raspberry Pi 3B. However, such chipset is quite old. Should I contact Infineon or Broadcom regarding firmware issues?
Specifically to brcm/brcmfmac43430-sdio firmware.

Regards.

Matheus-Garbelini, 2022-01-11:
@pelwell thanks a lot for your fast reply. Will contact Infineon asap.

Matheus-Garbelini, 2022-01-17:
Hi @pelwell, It seems Infineon is not replying for about a week.
Since you are an official raspberrypi maintainer, is it possible to sent the report directly to you via encrypted email?

Regards.

pelwell, 2022-01-17:
I've asked our main Infineon contact how they want to handle this, and will get back to you with their answer.

grsm17, 2022-01-17:
Hi Matheus - Could you confirm that you followed the process listed here for highlighting issues?
https://www.infineon.com/dgdl/Vulnerability_Notification_Process.pdf?fileId=8ac78c8b7ca35f7b017ca6d98b1c0000

If yes then i will chase our internal teams, if not then please do so and i will follow up once received by the team.

Thanks
Graham

pelwell, 2022-01-17:
Thank you, Graham.

Matheus-Garbelini, 2022-01-17:
Hi, @grsm17 and @pelwell , thanks a lot for the reply.

@grsm17 Yes, the report was sent to Infineon PSIRT (psirt@infineon.com) at January 12th, 10:15AM GMT, from my university email (matheus_garbelini@mymail.sutd.edu.sg). I've just sent a reply to the same email to help tracking it down by your team.

Regards and thanks for the attention.

pelwell, 2022-01-17:
@Matheus-Garbelini I've heard that Infineon support are now looking into the report and will contact you directly as and when necessary 

Thanks for bringing this to our attention.

closed 2022-01-17 by pelwell

Matheus-Garbelini, 2022-01-17:
@pelwell Indeed, everything is being handled now.
Thanks for your support and the work on Raspberry Pi platform.