raspberrypi/firmware #1677: please specify raspberrypi/linux tag for corresponding firmware stable updates

colemickens, 2022-01-12:
**Is this the right place for my bug report?**

This corresponds to: 

When I look at: https://github.com/raspberrypi/firmware/tree/stable/boot the most recent commit history entry is "Update firmware and kernel to latest master".

I'm not sure how to track that to a commit in raspberrypi/linux. I even pulled the built kernel but couldn't find a git rev in the kernel version string.

Would it be possible to specify the git rev used from the linux repo, whenever they're dropped in here? It seems like git tags aren't always pushed and particularly I don't see a new stable tag for the most recent new stable kernel dropped.

This would greatly help me in keeping rpi4 stuff up to date for NixOS.

colemickens, 2022-01-12:
Alternatively, if you built the git rev into the kernel version string, I could extract this myself without any other modifications to release/tag workflow.

pelwell, 2022-01-12:
I think https://github.com/raspberrypi/firmware/blob/stable/extra/git_hash is what you are looking for.

colemickens, 2022-01-12:
Thank you @pelwell !!

closed 2022-01-12 by colemickens

pelwell, 2022-01-23:
Things don't get pushed to those repos unless they are to be released. If you take the latest commits the worst that should happen is that you get ahead of RPiOS.

colemickens, 2022-01-23:
Thank you very much! I really appreciate the help!