raspberrypi/firmware #1678: Raspberry Pi 1B+ freezes at boot

Iam-lcl, 2022-01-12:
**Describe the bug**
Flashing a clean Raspberry Pi OS Lite for use on the RPi 1 B+ v1.2 will not boot up and freezes on bootloading or kernel panic.
I can see it start to load but stalls.
It will boot only if I change in the config file with
`start_file=start_x.elf`
But then it will not load drivers?
USB is not detected, neither does ethernet, maybe just USB?

I tested to download the master from here and just copy the content from boot folder into the SD cards boot partition.
This for some reason made it boot with `start_file=start_x.elf` in config file and drivers loaded.

I managed to login, run apt update|upgrade and everything looked fine.

Rebooting stops it from loading the drivers again.

**To reproduce**
Run Raspberry Pi Imager and flash Raspberry Pi OS lite (32bit) 2021-10-30 onto SD card.
run it as it is will freeze boot.

adding only `start_file=start_x.elf` to config file and now it boots all the way to login but without USB.

Running this same SD card and power supply will boot on Rpi 0 (non W) 
I will also boot with `start_file=start_x.elf`
Both with usb available (don't have usb ethernet available, but keyboard works)

**Expected behaviour**
Boot normaly after flash with Raspberry Pi Imager.

**Actual behaviour**
Freezes on boot or gets kernel panic

**System**
RPi 1B+
Raspberry Pi OS lite (32bit) 2021-10-30
5 different SD cards from Samsung, Sandisk and Kingston, 16-32GB, Kingston is brand new
3 different Power supply, last one official Raspberry pi power supply

**Logs**
unable to get logs, if there is no way to get the logs written into SD?

**Additional context**
Tested different versions from Raspberry Pi Imager, similar issues, some get stuck at rainbow splash some black screens.
Risc OS Pi works without problems

pelwell, 2022-01-21:
Sorry I took a while to get round to looking at this, but a completely clean install of Raspberry Pi OS Lite (32bit) 2021-10-30 boots just fine on my favourite rev 1.2 B+ (if very slowly compared to a 4B), with working USB and Ethernet. The non-Lite version also eventually boots to a desktop.

Since you've already tried different power supplies and SD cards it sound like your B+ could be showing its age. You could try with `over_voltage=2` in `config.txt` to see if that helps, but it may be time for an upgrade.

Iam-lcl, 2022-01-21:
Hmm..

Well I upgraded to Rpi2 when it got released then upgraded again to Rpi3 (non +) when that got released.
It's just that the Rpi1B+ has not been in use since I replaced it.
Just wanted to try and use it as a ip<sat server.

The `over_voltage=2` made no difference.

But this is probably some hardware issues or something..
The SD-card I use at the moment is the Kingston Canvas select plus A1 V10 U1 32GB microSDHC.
all of the SD cards is of microSDHC.

Can it have something to do with this?
May be the Pi has problem reading from these SD-cards?

Also I managed to get som logs now via serial terminal.
This is fresh flashed SD card, and with `over_voltage=2`

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.63+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1459 Wed Oct 6 16:40:27 BST 2021
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Model B Plus Rev 1.2
[    0.000000] random: fast init done
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Reserved memory: created CMA memory pool at 0x0bc00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113680
[    0.000000] Kernel command line: coherent_pool=1M snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  console=ttyAMA0,115200 console=tty1 root=PARTUUID=0a164481-02 rootfstype=ext4 fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 177408K/458752K available (8683K kernel code, 1324K rwdata, 2824K rodata, 420K init, 837K bss, 19200K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 30717 entries in 60 pages
[    0.000000] ftrace: allocated 60 pages with 4 groups
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x324/0x560 with crng_init=1
[    0.000027] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000114] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000229] bcm2835: system timer (irq = 27)
[    0.000999] Console: colour dummy device 80x30
[    0.001667] printk: console [tty1] enabled
[    0.001777] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060434] pid_max: default: 32768 minimum: 301
[    0.060791] LSM: Security Framework initializing
[    0.061181] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.061262] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.063755] cgroup: Disabling memory control group subsystem
[    0.064317] CPU: Testing write buffer coherency: ok
[    0.066692] Setting up static identity map for 0x8200 - 0x8238
[    0.068335] devtmpfs: initialized
[    0.082315] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.082840] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.082948] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.120712] pinctrl core: initialized pinctrl subsystem
[    0.123145] NET: Registered protocol family 16
[    0.128211] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.135240] audit: initializing netlink subsys (disabled)
[    0.137075] thermal_sys: Registered thermal governor 'step_wise'
[    0.138090] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.138200] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.138557] Serial: AMBA PL011 UART driver
[... 128 lines cut ...]
[    5.654600] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
[    5.673562] VFS: Mounted root (ext4 filesystem) readonly on device 179:2.
[    5.691588] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00, bcdDevice= 2.00
[    5.710741] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    5.734856] smsc95xx v2.0.0
[    5.749446] devtmpfs: mounted
[    5.771294] Freeing unused kernel memory: 420K
[    5.786520] Kernel memory protection not selected by kernel config.
[    5.803857] Run /sbin/init as init process
[    6.556007] libphy: smsc95xx-mdiobus: probed
[    6.575933] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-20980000.usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:2d:85:64
[    6.964641] systemd[1]: System time before build time, advancing clock.
[    7.144332] NET: Registered protocol family 10
[... cut ...]
[    7.409404] systemd[1]: Set hostname to <raspberrypi>.
[    7.471741] random: systemd: uninitialized urandom read (16 bytes read)
[    7.490991] systemd[1]: Initializing machine ID from random generator.
[    7.510367] systemd[1]: Installed transient /etc/machine-id file.
[    8.994342] uart-pl011 20201000.serial: no DMA platform data
[   10.589365] systemd[1]: Caught <SEGV>, core dump failed (child 74, code=killed, status=11/SEGV).
[   10.610857] systemd[1]: Freezing execution.
[  181.843401] random: crng init done
```

pelwell, 2022-01-21:
You can also try reducing arm_freq and sdram_freq - start with `arm_freq=600` and `sdram_freq=250`, increasing on success.

Iam-lcl, 2022-01-21:
no, that did not work. I lowered it down to `arm_freq=500` and `sdram_freq=200` still freezing execution.

But setting `start_file=start_x.elf`
make it boot into login.
but with no usb, ethernet and even no serial terminal this time.
`start_x=1` and it behaves as without any start_x thingys.

Also `Risc OS Pi` is the only OS that works.

pelwell, 2022-01-21:
I suspect there is a memory fault, and using the `x` version of the firmware changes the memory layout enough to make the difference. Adding `gpu_mem=64` may have a similar effect.

Iam-lcl, 2022-01-21:
Interesting.
I removed all othere parameters and only use `gpu_mem`.
`gpu_mem=64` same experience as before
`gpu_mem=32` kernel paniks
`gpu_mem=128` black screen only no other signs of life
`gpu_mem=256` Boots, usb work, ethernet works. running `sudo apt update` fails after a few procents. `shutdown now` fails....
`gpu_mem=384` Boots, usb work, ethernet works, running `sudo apt update` works, `sudo apt upgrade` works, `shutdown now` works.
Not much memory left for OS, but now it's showing some signs of life.

So I guess you are right, there is some RAM faults here, reserving it to GPU makes OS get a part which still is good or something.
That's a bummer..

Iam-lcl, 2022-01-30:
I'm closing this as this is a issue with my hardware.
I was hoping to repurpose my old Rpi 1B+ from dust collecting into something more useful, but looks like it's going back to dust collecting. If I don't try to transplant the ram chip or something..

Thanks for the help

closed 2022-01-30 by Iam-lcl