raspberrypi/firmware #1679: Raspberry Pi Camera module not detected!

Mr-mongoose-com, 2022-01-14:
When Raspberry Pi camera module is inserted into the slot beside the headphone jack, with the blue sticker facing the Ethernet port, when running `vcgencmd get_camera` returns `supported=1 detected=0`. run `sudo raspi-config` and the connection to the peripheral is enabled.

popcornmix, 2022-01-17:
Unless you are saying this problem only occurs with a an updated version of firmware, it doesn't sound like a firmware issue.

If the camera is not detected, it is likely the camera is faulty, the cable is faulty or it is not connected correctly.

6by9, 2022-01-17:
```vcgencmd get_camera``` only applies if the legacy camera stack is enabled.
If using Bullseye with libcamera configured then it will report no cameras detected.

popcornmix, 2022-01-17:
Perhaps we should change the output of `get_camera` with libcamera (e.g. `command not supported with libcamera`).
What's the simplest check that camera is connected in libcamera world?

6by9, 2022-01-17:
Add an "ignored" flag to ```camera_subsystem_get_attached_devices``` (exported function, so there will be a few references), and increment that flag for any cameras that are configured for libcamera.
Leave it with me, but higher priorities at present.

6by9, 2022-02-07:
Firmware patch created that reports the number of camera interfaces configured for libcamera.

popcornmix, 2022-07-13:
What does `vcgencmd version` report on system that doesn't work?

Torxed, 2023-08-13:
For those ending up here with newer hardware, `vcgencmd` equivalent would be:
```
$ cam --list
$ cam --camera 1 --list-properties
$ cam --camera 1 --list-controls
$ cam --camera 1 --capture=10 --file=./
```
It's not a 1:1 copy, but it will test functionality and capabilities.