raspberrypi/firmware #1681: Intermittent "bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)" on boot (5.10).

bhjelstrom, 2022-01-15:
**Describe the bug**

Using 5.10 and associated firmware, the following error occurs during some boots. When the error occurs, the 5647 camera module is unusable. 

The frequency of the error appeared to increase as I optimized boot time by disabling EDID reads, disabling kernel tracing, etc.

[    5.276137] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
[    5.276346] dwc2 3f980000.usb: supply vusb_d not found, using dummy regulator
[    5.294908] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
[    5.298558] dwc2 3f980000.usb: supply vusb_a not found, using dummy regulator
[    5.303372] : Unregister from media controller
[    5.303379] (efault): Unregistering node (null)[0] device node /dev/video0
[    5.303383] (efault): Unregistering node (null)[0] device node /dev/video0
[    5.303386] (efault): Unregistering node (null)[0] device node /dev/video0
[    5.303390] (efault): Unregistering node (null)[0] device node /dev/video0

Below are logs from two different 5.10 kernel versions (each with correct firmware versions). It looked like the camera subsystem was getting a lot of attention lately, so I checked 5.10.90 and the 318e3373496 firmware (Jan 13), but the issue persisted.

**To reproduce**
Boot a CM3 with an OV5647 attached. Use 5.10.36 or 5.10.90 kernel with associated rpi extended firmware. The failure does not always occur - on 5.10.90 I had 20 successful boots followed by a boot with the above error.

**Expected behaviour**
The error never occurs, and the camera is always available for streaming. This error does not occur at all (with hundreds of tested boots) on 4.19.120

**Actual behaviour**
The error shown in the kernel output above occurs, and the camera is unavailable. From my testing, it appears as though a different critical issue I'm facing is fixed in 5.10 (https://github.com/raspberrypi/linux/issues/3702), but I cannot upgrade until the camera subsystem is reliable.

**System**
CM3+

**Logs**

Linux version 5.10.90-v7
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    3.592229] systemd[1]: Condition check resulted in Load Kernel Modules being skipped.
[    3.605081] systemd[1]: Starting Apply Kernel Variables...
[    3.615609] systemd[1]: Starting Coldplug All udev Devices...
[    3.691376] systemd[1]: Started Journal Service.
[    4.114109] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    4.267793] systemd-journald[108]: Received client request to flush runtime journal.
[    4.497849] audit: type=1334 audit(1634052246.719:2): prog-id=5 op=LOAD
[    4.506279] audit: type=1334 audit(1634052246.729:3): prog-id=6 op=LOAD
[    4.711343] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
[    4.720637] mc: Linux media interface: v0.10
[    4.751709] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
[    4.765329] [vc_sm_connected_init]: start
[    4.768127] videodev: Linux video capture interface: v2.00
[    4.782731] [vc_sm_connected_init]: installed successfully
[    4.821042] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[    4.821046] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[    4.826360] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
[    4.842026] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[    4.864495] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
[    4.881168] bcm2835_isp: module is from the staging directory, the quality is unknown, you have been warned.
[    4.963695] enc28j60 spi0.0: Ethernet driver 1.02 loaded
[    5.190934] audit: type=1334 audit(1634052247.409:4): prog-id=7 op=LOAD
[    5.199650] audit: type=1334 audit(1634052247.409:5): prog-id=8 op=LOAD
[    5.276137] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
[    5.276346] dwc2 3f980000.usb: supply vusb_d not found, using dummy regulator
[    5.294908] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
[    5.298558] dwc2 3f980000.usb: supply vusb_a not found, using dummy regulator
[    5.303372] : Unregister from media controller
[    5.303379] (efault): Unregistering node (null)[0] device node /dev/video0
[    5.303383] (efault): Unregistering node (null)[0] device node /dev/video0
[    5.303386] (efault): Unregistering node (null)[0] device node /dev/video0
[    5.303390] (efault): Unregistering node (null)[0] device node /dev/video0
[    5.413208] dwc2 3f980000.usb: DWC OTG Controller
[    5.420123] dwc2 3f980000.usb: new USB bus registered, assigned bus number 1
[    5.431017] dwc2 3f980000.usb: irq 66, io mem 0x3f980000
[    5.439760] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.10
[... 280 lines cut ...]
[    5.680705] audit: type=1334 audit(1612279790.899:4): prog-id=7 op=LOAD
[    5.689194] audit: type=1334 audit(1612279790.899:5): prog-id=8 op=LOAD
[    6.210526] 8021q: 802.1Q VLAN Support v1.8
[    6.314177] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    6.353158] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    6.446838] enc28j60 spi0.0 eth0: link down
[    6.452626] enc28j60 spi0.0 eth0: multicast mode
[    6.459126] enc28j60 spi0.0 eth0: multicast mode
[    6.486657] enc28j60 spi0.0 eth0: multicast mode
[    6.492965] enc28j60 spi0.0 eth0: multicast mode
[    6.523865] enc28j60 spi0.0 eth0: multicast mode
[    6.530029] enc28j60 spi0.0 eth0: multicast mode
[    7.352373] enc28j60 spi0.0 eth0: link up - Half duplex
[    7.358930] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[    7.386465] enc28j60 spi0.0 eth0: multicast mode
[    7.392357] enc28j60 spi0.0 eth0: multicast mode
[    9.155402] random: crng init done
[    9.159946] random: 7 urandom warning(s) missed due to ratelimiting
[    9.440419] enc28j60 spi0.0 eth0: multicast mode
[    9.446283] enc28j60 spi0.0 eth0: multicast mode
[    9.451956] enc28j60 spi0.0 eth0: multicast mode
[   15.961274] enc28j60 spi0.0 eth0: multicast mode

**Additional context**
Add any other relevant context for the problem.

bhjelstrom, 2022-04-26:
Still occurs on 5.10.110, 32 bit, legacy camera stack

[7.449622] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
[    7.474795] mc: Linux media interface: v0.10
[    7.476420] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
[    7.502259] [vc_sm_connected_init]: start
[    7.515101] [vc_sm_connected_init]: installed successfully
[    7.607023] videodev: Linux video capture interface: v2.00
[    7.729473] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[    7.739879] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[    7.766837] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[    7.848193] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
[    7.853054] bcm2835_isp: module is from the staging directory, the quality is unknown, you have been warned.
[    7.937173] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
[    7.985297] audit: type=1334 audit(1634052248.319:4): prog-id=7 op=LOAD
[    7.997520] audit: type=1334 audit(1634052248.319:5): prog-id=8 op=LOAD
[    8.043512] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
[    8.046714] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
[    8.067867] dwc2 3f980000.usb: supply vusb_d not found, using dummy regulator
[    8.080910] : Unregister from media controller
[    8.080921] (efault): Unregistering node (null)[0] device node /dev/video0
[    8.080927] (efault): Unregistering node (null)[0] device node /dev/video0
[    8.080933] (efault): Unregistering node (null)[0] device node /dev/video0
[    8.080939] (efault): Unregistering node (null)[0] device node /dev/video0

The output of vcgencmd get_camera is:
`supported=2 detected=1`

`sudo v4l2-ctl --list-devices
Cannot open device /dev/video0, exiting.`

bhjelstrom, 2022-04-27:
This occurs on 5.15.32 on both CM3 and CM3+ modules. I cannot reproduce it reliably, but it seems to occur once every 15-20 boot cycles.

bhjelstrom, 2022-04-28:
vcdbg output when the failure occurs:

002200.907: brfs: File read: /mfs/sd/config.txt
002201.575: brfs: File read: 398 bytes
002203.338: brfs: File read: /mfs/sd/dt-blob.bin
002209.934: brfs: File read: 16134 bytes
002210.483: brfs: File read: /mfs/sd/config.txt
003211.103: gpioman: gpioman_get_pin_num: pin DISPLAY_SDA not defined
003211.133: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
003830.822: gpioman: gpioman_get_pin_num: pin FLASH_0_ENABLE not defined
003830.850: gpioman: gpioman_get_pin_num: pin FLASH_0_INDICATOR not defined
003830.898: gpioman: gpioman_get_pin_num: pin FLASH_0_ENABLE not defined
003830.926: gpioman: gpioman_get_pin_num: pin FLASH_0_INDICATOR not defined
003830.997: gpioman: gpioman_get_pin_num: pin BT_ON not defined
003831.016: gpioman: gpioman_get_pin_num: pin WL_ON not defined
003861.119: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
003861.179: *** Restart logging
003861.204: brfs: File read: 398 bytes
003861.381: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
003861.402: HDMI0: hdmi_pixel_encoding: 162000000
003861.657: vec: vec_middleware_power_on: vec_base: 0x7e806000 rev-id 0x00002708 @ vec: 0x7e806100 @ 0x00000420 enc: 0x7e806060 @ 0x00000220 cgmsae: 0x7e80605c @ 0x00000000
003862.733: dtb_file 'bcm2710-rpi-cm3.dtb'
003862.750: Trying Device Tree file 'bcm2710-rpi-cm3.dtb'
003868.631: brfs: File read: /mfs/sd/bcm2710-rpi-cm3.dtb
003868.655: Loading 'bcm2710-rpi-cm3.dtb' to 0x100 size 0x72e6
003878.911: dtdebug: delete_node(/**local_fixups**)
003882.098: brfs: File read: 29414 bytes
003890.459: dtdebug: using platform 'bcm2835'
003890.476: dtdebug: overlay map not loaded
003892.180: dtdebug: /aliases:i2c_vc=i2c0
003897.744: dtdebug: /**symbols**:i2c_vc=i2c0
003901.658: dtdebug: /**overrides**:i2c_vc=i2c0
003909.471: dtdebug: /**overrides**:i2c_vc_baudrate=i2c0_baudrate
003911.209: dtdebug: /aliases:i2c=i2c1
003916.910: dtdebug: /**symbols**:i2c=i2c1
003920.815: dtdebug: /**overrides**:i2c=i2c1
003922.544: dtdebug: /aliases:i2c_arm=i2c1
003928.193: dtdebug: /**symbols**:i2c_arm=i2c1
003932.162: dtdebug: /**overrides**:i2c_arm=i2c1
003940.132: dtdebug: /**overrides**:i2c_baudrate=i2c1_baudrate
003948.178: dtdebug: /**overrides**:i2c_arm_baudrate=i2c1_baudrate
003960.279: dtparam: uart0_clkrate=48000000
003963.328: dtdebug: /**overrides** has no uart0_clkrate property
003963.344: Unknown dtparam 'uart0_clkrate' - ignored
003963.913: brfs: File read: /mfs/sd/config.txt
003964.103: brfs: File read: 398 bytes
003972.083: dtdebug: Opened overlay file 'overlays/miniuart-bt.dtbo'
003973.707: brfs: File read: /mfs/sd/overlays/miniuart-bt.dtbo
003981.071: dterror: can't find symbol 'uart0_pins'
003981.085: Failed to resolve overlay 'miniuart-bt'
003981.142: dtparam: spi=on
003984.042: dtdebug: found override spi
003984.081: dtdebug:   override spi: string target 'status'
003988.633: brfs: File read: 1819 bytes
003993.706: dtdebug: Opened overlay file 'overlays/enc28j60.dtbo'
003995.402: brfs: File read: /mfs/sd/overlays/enc28j60.dtbo
004005.880: Loaded overlay 'enc28j60'
004014.910: dtdebug: merge_fragment(/soc/spi@7e204000,/fragment@0/**overlay**)
004014.938: dtdebug:   +prop(#address-cells)
004015.593: dtdebug:   +prop(#size-cells)
004016.243: dtdebug:   +prop(status)
004019.704: dtdebug: merge_fragment(/soc/spi@7e204000/enc28j60@0,/fragment@0/**overlay**/enc28j60@0)
004019.729: dtdebug:   +prop(compatible)
004020.354: dtdebug:   +prop(reg)
004021.019: dtdebug:   +prop(pinctrl-names)
004021.710: dtdebug:   +prop(pinctrl-0)
004022.426: dtdebug:   +prop(interrupt-parent)
004023.076: dtdebug:   +prop(interrupts)
004023.772: dtdebug:   +prop(spi-max-frequency)
004024.513: dtdebug:   +prop(status)
[... 169 lines cut ...]
004774.870: dtparam: cam0-led=2
004777.827: dtdebug: found override cam0-led
004777.859: dtparam: cam1-pwdn=3
004780.985: dtdebug: /**overrides** has no cam1-pwdn property
004780.998: Unknown dtparam 'cam1-pwdn' - ignored
004781.031: dtparam: cam1-led=30
004784.149: dtdebug: /**overrides** has no cam1-led property
004784.162: Unknown dtparam 'cam1-led' - ignored
004784.217: dtparam: arm_freq=1200000000
004787.175: dtdebug: found override arm_freq
004787.231: dtdebug:   override arm_freq: cell target clock-frequency @ offset 0 (size 4)
004795.457: dtdebug:   override arm_freq: cell target clock-frequency @ offset 0 (size 4)
004803.721: dtdebug:   override arm_freq: cell target clock-frequency @ offset 0 (size 4)
004812.049: dtdebug:   override arm_freq: cell target clock-frequency @ offset 0 (size 4)
004820.449: dtparam: core_freq=400000000
004823.581: dtdebug: /**overrides** has no core_freq property
004823.594: Unknown dtparam 'core_freq' - ignored
004832.446: gpioman: gpioman_get_pin_num: pin WL_ON not defined
004832.463: dtdebug: delete_node(/hat)
004840.456: brfs: File read: 65 bytes
005430.235: brfs: File read: /mfs/sd/zImage
005430.259: Loading 'zImage' to 0x8000 size 0x64dd30
005430.290: Device tree loaded to 0x2eff8500 (size 0x7a35)
005432.607: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
008077.778: vchiq_core: vchiq_init_state: slot_zero = 0xf4180000, is_master = 1

pelwell, 2022-04-28:
Those two logs are identical, which is almost impossible - did you paste the same one twice?

bhjelstrom, 2022-04-28:
I fixed the previous post. Here is more output from a recent reproduction of the issue (vcdbg, dmesg, journalctl). Interestingly, the vcdbg output shows that the camera was found, but we still see "bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)"

config:
```
start_file=start_x.elf
fixup_file=fixup_x.dat
boot_delay=1
kernel=zImage
gpu_mem=128
hdmi_ignore_hotplug=1
hdmi_ignore_edid=0xa5000080
dtoverlay=miniuart-bt
disable_splash=1
dtparam=spi=on
dtoverlay=enc28j60
dtoverlay=pi3-disable-bt
enable_uart=1
dtoverlay=dwc2,dr_mode=host
```
The issue continues to occur (intermittently, roughly once per 20-30 boots) after increasing CMA size to 256MB from 64MB

**VCDBG**
```
$ sudo vcdbg log msg >> log.txt
002251.735: brfs: File read: /mfs/sd/config.txt
002252.405: brfs: File read: 398 bytes
002254.167: brfs: File read: /mfs/sd/dt-blob.bin
002260.765: brfs: File read: 16134 bytes
002261.326: brfs: File read: /mfs/sd/config.txt
003261.944: gpioman: gpioman_get_pin_num: pin DISPLAY_SDA not defined
003261.973: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
003881.660: gpioman: gpioman_get_pin_num: pin FLASH_0_ENABLE not defined
003881.687: gpioman: gpioman_get_pin_num: pin FLASH_0_INDICATOR not defined
003881.735: gpioman: gpioman_get_pin_num: pin FLASH_0_ENABLE not defined
003881.763: gpioman: gpioman_get_pin_num: pin FLASH_0_INDICATOR not defined
003881.834: gpioman: gpioman_get_pin_num: pin BT_ON not defined
003881.853: gpioman: gpioman_get_pin_num: pin WL_ON not defined
003911.939: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
003911.998: *** Restart logging
003912.024: brfs: File read: 398 bytes
003912.212: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
003912.233: HDMI0: hdmi_pixel_encoding: 162000000
003912.488: vec: vec_middleware_power_on: vec_base: 0x7e806000 rev-id 0x00002708 @ vec: 0x7e806100 @ 0x00000420 enc: 0x7e806060 @ 0x00000220 cgmsae: 0x7e80605c @ 0x00000000
003913.566: dtb_file 'bcm2710-rpi-cm3.dtb'
003913.583: Trying Device Tree file 'bcm2710-rpi-cm3.dtb'
003919.465: brfs: File read: /mfs/sd/bcm2710-rpi-cm3.dtb
003919.490: Loading 'bcm2710-rpi-cm3.dtb' to 0x100 size 0x72e6
003929.754: dtdebug: delete_node(/__local_fixups__)
003932.943: brfs: File read: 29414 bytes
003941.299: dtdebug: using platform 'bcm2835'
003941.316: dtdebug: overlay map not loaded
003943.021: dtdebug: /aliases:i2c_vc=i2c0
003948.584: dtdebug: /__symbols__:i2c_vc=i2c0
003952.498: dtdebug: /__overrides__:i2c_vc=i2c0
003960.311: dtdebug: /__overrides__:i2c_vc_baudrate=i2c0_baudrate
003962.051: dtdebug: /aliases:i2c=i2c1
003967.752: dtdebug: /__symbols__:i2c=i2c1
003971.658: dtdebug: /__overrides__:i2c=i2c1
003973.387: dtdebug: /aliases:i2c_arm=i2c1
003979.037: dtdebug: /__symbols__:i2c_arm=i2c1
003983.006: dtdebug: /__overrides__:i2c_arm=i2c1
003990.976: dtdebug: /__overrides__:i2c_baudrate=i2c1_baudrate
003999.022: dtdebug: /__overrides__:i2c_arm_baudrate=i2c1_baudrate
004011.123: dtparam: uart0_clkrate=48000000
004014.173: dtdebug: /__overrides__ has no uart0_clkrate property
004014.188: Unknown dtparam 'uart0_clkrate' - ignored
004014.758: brfs: File read: /mfs/sd/config.txt
004014.948: brfs: File read: 398 bytes
004022.930: dtdebug: Opened overlay file 'overlays/miniuart-bt.dtbo'
004024.555: brfs: File read: /mfs/sd/overlays/miniuart-bt.dtbo
[... 775 lines cut ...]
Mar 11 07:33:37 scll kernel: enc28j60 spi0.0 eth0: multicast mode
Mar 11 07:33:37 scll kernel: enc28j60 spi0.0 eth0: multicast mode
Mar 11 07:33:37 scll dhcpcd[153]: eth0: waiting for carrier
Mar 11 07:33:37 scll dhcpcd[153]: eth0: carrier acquired
Mar 11 07:33:37 scll systemd[1]: Started Serial Getty on ttyAMA0.
Mar 11 07:33:37 scll systemd[1]: Reached target Login Prompts.
Mar 11 07:33:37 scll dhcpcd[153]: eth0: carrier lost
Mar 11 07:33:38 scll avahi-daemon[140]: Server startup complete. Host name is scll.local. Local service cookie is 4135720468.
Mar 11 07:33:38 scll kernel: bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
Mar 11 07:33:38 scll kernel: bcm2835-v4l2-0: mmal_init: vchiq mmal init failed -19
Mar 11 07:33:38 scll kernel: bcm2835-v4l2-0: bcm2835_mmal_probe: mmal init failed: -19
Mar 11 07:33:39 scll kernel: enc28j60 spi0.0 eth0: link up - Half duplex
Mar 11 07:33:39 scll kernel: IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Mar 11 07:33:39 scll dhcpcd[153]: eth0: carrier acquired
Mar 11 07:33:39 scll kernel: enc28j60 spi0.0 eth0: multicast mode
Mar 11 07:33:40 scll kernel: random: crng init done
Mar 11 07:33:40 scll kernel: random: 7 urandom warning(s) missed due to ratelimiting
Mar 11 07:33:40 scll systemd[1]: Started OpenSSH server daemon.
Mar 11 07:33:40 scll sshd[207]: Server listening on 0.0.0.0 port 22.
Mar 11 07:33:40 scll sshd[207]: Server listening on :: port 22.
Mar 11 07:33:40 scll systemd[1]: Started Apache HTTP Server.
Mar 11 07:33:40 scll systemd[1]: Reached target Multi-User System.
Mar 11 07:33:40 scll systemd[1]: Startup finished in 3.502s (kernel) + 7.368s (userspace) = 10.870s.
Mar 11 07:33:40 scll dhcpcd[153]: eth0: rebinding lease of 192.168.1.45
Mar 11 07:33:40 scll dhcpcd[153]: eth0: probing address 192.168.1.45/24```

bhjelstrom, 2022-04-29:
I appreciate this @pelwell. I patched my 5.15.32 with your change since 5.15.36 breaks my cm3+ ethernet.

**This is what I see when it works:**
``````bash
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.15.32-v7 
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Compute Module 3 Plus Rev 1.0
[    0.000000] random: fast init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x34000000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x0000000037ffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000037ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000037ffffff]
[    0.000000] percpu: Embedded 16 pages/cpu s35660 r8192 d21684 u65536
[    0.000000] pcpu-alloc: s35660 r8192 d21684 u65536 alloc=16*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 227360
[    0.000000] Kernel command line: coherent_pool=1M snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:CB:3F:95 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  root=/dev/mmcblk0p2 rootwait console=tty1 console=ttyAMA0,115200 vchiq.core_msg_log_level=6
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 825324K/917504K available (10240K kernel code, 1341K rwdata, 3096K rodata, 1024K init, 568K bss, 26644K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 33175 entries in 98 pages
[    0.000000] ftrace: allocated 98 pages with 3 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x520/0x710 with crng_init=1
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000001] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000025] Switching to timer-based delay loop, resolution 52ns
[    0.000342] Console: colour dummy device 80x30
[    0.001175] printk: console [tty1] enabled
[    0.001254] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001315] pid_max: default: 32768 minimum: 301
[    0.001502] LSM: Security Framework initializing
[    0.001711] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001764] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.003288] cgroup: Disabling memory control group subsystem
[    0.003582] CPU: Testing write buffer coherency: ok
[    0.004157] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.005398] Setting up static identity map for 0x100000 - 0x10003c
[    0.005609] rcu: Hierarchical SRCU implementation.
[    0.006554] smp: Bringing up secondary CPUs ...
[    0.007722] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.008991] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.010285] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.010434] smp: Brought up 1 node, 4 CPUs
[    0.010492] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.010528] CPU: All CPU(s) started in HYP mode.
[    0.010555] CPU: Virtualization extensions available.
[    0.011479] devtmpfs: initialized
[    0.029222] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.029542] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.029609] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.032827] pinctrl core: initialized pinctrl subsystem
[    0.034247] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[... 1814 lines cut ...]
[   14.932299] vchiq: Rcvd: 00000000: 6d 6d 61 6c 09 00 00 00 b4 21 35 3f 0e 00 00 00 mmal.....!5?....
[   14.944289] vchiq: Sent: 00000000: 6d 6d 61 6c 08 00 00 00 00 00 00 00 0e 00 00 00 mmal............
[   14.956323] vchiq: Sent Msg DATA(5) to mmal s:5 d:89 len:36
[   14.963427] vchiq: Rcvd Msg DATA(5) from mmal s:89 d:5 len:316
[   14.970718] vchiq: Rcvd: 00000000: 6d 6d 61 6c 08 00 00 00 b4 21 35 3f 0e 00 00 00 mmal.....!5?....
[   14.982744] vchiq: Sent: 00000000: 6d 6d 61 6c 0a 00 00 00 00 00 00 00 0e 00 00 00 mmal............
[... cut ...]
[   15.098070] vchiq: Sent: 00000000: 6d 6d 61 6c 06 00 00 00 30 00 00 00 0e 00 00 00 mmal....0.......
[   15.110096] vchiq: Sent Msg DATA(5) to mmal s:5 d:89 len:28
[   15.121233] vchiq: Rcvd Msg DATA(5) from mmal s:89 d:5 len:28
[   15.128439] vchiq: Rcvd: 00000000: 6d 6d 61 6c 06 00 00 00 b4 21 35 3f 0e 00 00 00 mmal.....!5?....
[   15.140400] bcm2835-v4l2-0: Broadcom 2835 MMAL video capture ver 0.0.2 loaded.
[   15.312242] enc28j60 spi0.0 eth0: multicast mode
[   33.112022] cam-dummy-reg: disabling
```

bhjelstrom, 2022-04-29:
Failure case filtered (| grep vchiq):
```
[    1.920532] vchiq: Rcvd Msg CONNECT(1) from ???? s:0 d:0 len:0
[    7.368011] vchiq: Sent Msg CONNECT(1) to ???? s:0 d:0 len:0
[    7.379497] vchiq: Open Service SMEM SrcPort:0
[    7.392083] vchiq: Add Service KEEP SrcPort:1
[    7.454004] vchiq: Sent Msg OPEN(2) to ???? s:0 d:0 len:12
[    7.463206] vchiq: Rcvd Msg OPENACK(3) from SMEM s:31 d:0 len:2
[    7.471787] vchiq: Rcvd: 00000000: 01 00                                           ..
[    7.493980] vchiq: Sent: 00000000: 09 00 00 00 01 00 00 00 02 00 00 00             ............
[    7.494004] vchiq: Sent Msg DATA(5) to SMEM s:0 d:31 len:12
[    7.495486] vchiq: Rcvd Msg DATA(5) from SMEM s:31 d:0 len:12
[    7.552682] vchiq: Rcvd: 00000000: 09 00 00 80 01 00 00 00 02 00 00 00             ............
[    7.575426] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[    7.576596] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[    7.576601] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[    7.691407] vchiq: Open Service mmal SrcPort:2
[    7.713824] vchiq: Sent Msg OPEN(2) to ???? s:2 d:0 len:12
[    7.722514] vchiq: Open Service mmal SrcPort:3
[    7.723613] vchiq: Rcvd Msg OPENACK(3) from mmal s:85 d:2 len:2
[    7.746887] vchiq: Rcvd: 00000000: 10 00                                           ..
[    7.749029] vchiq: Sent: 00000000: 6d 6d 61 6c 04 00 00 00 cc b9 09 83 01 00 00 00 mmal............
[    7.775934] vchiq: Sent Msg DATA(5) to mmal s:2 d:85 len:160
[    7.786175] vchiq: Sent Msg OPEN(2) to ???? s:3 d:0 len:12
[    7.855107] vchiq: Open Service mmal SrcPort:4
[    7.864406] vchiq: Sent Msg OPEN(2) to ???? s:4 d:0 len:12
[    7.903488] vchiq: Rcvd Msg CLOSE(4) from mmal s:0 d:3 len:0
[    7.912746] vchiq: Rcvd Msg CLOSE(4) from mmal s:0 d:4 len:0
[    7.912787] vchiq: Rcvd Msg DATA(5) from mmal s:85 d:2 len:44
[    7.912827] vchiq: Rcvd: 00000000: 6d 6d 61 6c 04 00 00 00 a4 03 36 3f 01 00 00 00 mmal......6?....
[    7.912909] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
[    7.913231] vchiq: Sent: 00000000: 6d 6d 61 6c 08 00 00 00 6d 6d 61 6c 01 00 00 00 mmal....mmal....
[    7.913248] vchiq: Sent Msg DATA(5) to mmal s:2 d:85 len:36
[    7.913408] vchiq: Rcvd Msg DATA(5) from mmal s:85 d:2 len:316
[    7.913446] vchiq: Rcvd: 00000000: 6d 6d 61 6c 08 00 00 00 a4 03 36 3f 01 00 00 00 mmal......6?....
[    7.913679] vchiq: Sent: 00000000: 6d 6d 61 6c 08 00 00 00 6d 6d 61 6c 02 00 00 00 mmal....mmal....
[    7.913694] vchiq: Sent Msg DATA(5) to mmal s:2 d:85 len:36
[    7.915526] vchiq: Rcvd Msg DATA(5) from mmal s:85 d:2 len:316
[    7.915559] vchiq: Rcvd: 00000000: 6d 6d 61 6c 08 00 00 00 a4 03 36 3f 02 00 00 00 mmal......6?....
[    7.915822] vchiq: Sent: 00000000: 6d 6d 61 6c 08 00 00 00 6d 6d 61 6c 03 00 00 00 mmal....mmal....
[    7.915831] vchiq: Sent Msg DATA(5) to mmal s:2 d:85 len:36
[    7.915885] vchiq: Rcvd Msg DATA(5) from mmal s:85 d:2 len:316
[    7.915905] vchiq: Rcvd: 00000000: 6d 6d 61 6c 08 00 00 00 a4 03 36 3f 03 00 00 00 mmal......6?....
[    7.915950] vchiq: Sent: 00000000: 6d 6d 61 6c 08 00 00 00 6d 6d 61 6c 04 00 00 00 mmal....mmal....
[    7.915957] vchiq: Sent Msg DATA(5) to mmal s:2 d:85 len:36
[    7.916002] vchiq: Rcvd Msg DATA(5) from mmal s:85 d:2 len:316
[    7.916026] vchiq: Rcvd: 00000000: 6d 6d 61 6c 08 00 00 00 a4 03 36 3f 04 00 00 00 mmal......6?....
[    7.916090] vchiq: Sent: 00000000: 6d 6d 61 6c 08 00 00 00 6d 6d 61 6c 05 00 00 00 mmal....mmal....
[    7.916099] vchiq: Sent Msg DATA(5) to mmal s:2 d:85 len:36
[    7.916153] vchiq: Rcvd Msg DATA(5) from mmal s:85 d:2 len:316
[    7.916174] vchiq: Rcvd: 00000000: 6d 6d 61 6c 08 00 00 00 a4 03 36 3f 05 00 00 00 mmal......6?....
[    7.917845] vchiq: Sent: 00000000: 6d 6d 61 6c 0f 00 00 00 50 be 10 80 06 00 00 00 mmal....P.......
[    7.917865] vchiq: Sent Msg DATA(5) to mmal s:2 d:85 len:40
[    7.918127] vchiq: Rcvd Msg DATA(5) from mmal s:85 d:2 len:284
[    7.918155] vchiq: Rcvd: 00000000: 6d 6d 61 6c 0f 00 00 00 a4 03 36 3f 06 00 00 00 mmal......6?....
[    7.919966] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
[    7.960786] vchiq: Sent: 00000000: 6d 6d 61 6c 0f 00 00 00 50 be 10 80 06 00 00 00 mmal....P.......
[    7.974782] vchiq: Sent Msg DATA(5) to mmal s:2 d:85 len:40
[    7.998383] vchiq: Rcvd Msg DATA(5) from mmal s:85 d:2 len:108
[    8.187243] vchiq: Rcvd: 00000000: 6d 6d 61 6c 0f 00 00 00 a4 03 36 3f 06 00 00 00 mmal......6?....
[    8.313523] vchiq: Sent: 00000000: 6d 6d 61 6c 0f 00 00 00 50 be 10 80 06 00 00 00 mmal....P.......
[    8.334755] vchiq: Sent Msg DATA(5) to mmal s:2 d:85 len:40
[    8.349028] vchiq: Rcvd Msg DATA(5) from mmal s:85 d:2 len:80
[    8.349063] vchiq: Rcvd: 00000000: 6d 6d 61 6c 0f 00 00 00 a4 03 36 3f 06 00 00 00 mmal......6?....
[    8.393954] vchiq: Sent: 00000000: 6d 6d 61 6c 0f 00 00 00 00 ef 7d b3 06 00 00 00 mmal......}.....
[    8.412131] vchiq: Sent Msg DATA(5) to mmal s:2 d:85 len:40
[    8.425567] vchiq: Rcvd Msg DATA(5) from mmal s:85 d:2 len:40
[    8.440383] vchiq: Rcvd: 00000000: 6d 6d 61 6c 0f 00 00 00 a4 03 36 3f 06 00 00 00 mmal......6?....
[    8.532072] vchiq: Open Service mmal SrcPort:3
[    8.538077] vchiq: Sent Msg OPEN(2) to ???? s:3 d:0 len:12
[    8.545131] vchiq: Rcvd Msg CLOSE(4) from mmal s:0 d:3 len:0
[... cut ...] CHI service connection (status=-1)
[    8.614128] vchiq: Sent: 00000000: 6d 6d 61 6c 05 00 00 00 2d 00 00 00 06 00 00 00 mmal....-.......
[    8.614139] vchiq: Sent Msg DATA(5) to mmal s:2 d:85 len:28
[    8.614369] vchiq: Rcvd Msg DATA(5) from mmal s:85 d:2 len:28
[    8.652413] vchiq: Rcvd: 00000000: 6d 6d 61 6c 05 00 00 00 a4 03 36 3f 06 00 00 00 mmal......6?....
[    8.664595] vchiq: Sent Msg CLOSE(4) to mmal s:2 d:85 len:0
[    8.665229] vchiq: Rcvd Msg CLOSE(4) from mmal s:85 d:2 len:0
```

bhjelstrom, 2022-04-29:
Your experience made me think that this had something to do with our camera application. I delayed our camera application for one minute and confirmed that it is not causing the issue. I would love to transition to libcamera, but I'm experiencing some difficulties: https://forums.raspberrypi.com/viewtopic.php?t=333753#p1998390

bhjelstrom, 2022-05-03:
Another data point - this occurs on 5.10.20 as well.

pelwell, 2022-05-03:
The "reboot if no error" loop has run for almost 6000 iterations without a MMAL/VCHIQ failure.. The whole thing feels like a race condition, but without a means to reproduce it making progress is impossible.

Is there a shorter start-up delay that still avoids the problem and is acceptable as a workaround?

closed 2022-11-12 by bhjelstrom