raspberrypi/firmware #1682: ttyAMA0/serial0 not working when using firmware from `next`

frezbo, 2022-01-16:
**Describe the bug**
Kernel console logs are missing when using firmware of `next` branch. Kernel doesn't seem to detect `ttyAMA0`, works fine with firmware from `master` branch

**To reproduce**
List the steps required to reproduce the issue.

**Expected behaviour**
Add a clear and concise description of what you expected to happen.

**Actual behaviour**
Add a clear and concise description of what actually happened.

**System**
* Raspbeery Pi 4 loading u-boot loading Grub loading kernel

*config.txt*

```bash
```

**Logs**
With Firmware

**Additional context**
Add any other relevant context for the problem.

pelwell, 2022-01-16:
Does it work if the firmware loads the kernel directly?

pelwell, 2022-01-17:
Starting with a copy of your config.txt and commenting out `armstub=bl31.bin` and `kernel=u-boot.bin` leads to a working ttyAMA0 for me:
```
...
[    5.129602] systemd[1]: Mounted Kernel Configuration File System.

Raspbian GNU/Linux 11 raspberrypi ttyAMA0

raspberrypi login: pi
Password:
Linux raspberrypi 5.15.13-v8+ #1510 SMP PREEMPT Wed Jan 5 12:28:30 GMT 2022 aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Jan 17 10:14:15 GMT 2022 on tty1
pi@raspberrypi:~$
```
```
EFI stub: Booting Linux Kernel...
EFI stub: ERROR: FIRMWARE BUG: kernel image not aligned on 64k boundary
```
Does the EFI stub understand that kernel8.img is gzip-compressed, reducing it from >20MB to ~8MB? What is the size of the 5.10 kernel8.img that is working for you?

frezbo, 2022-01-17:
I tried `gzipping` the kernel if I use it, it keeps on rebooting. The un-compressed kernel size is 54M and compressed is 54M

```bash
❯ du -sh vmlinuz 
54M	vmlinuz
❯ du -sh Image.gz 
20M	Image.gz
```

In my case grub boots and un-compressed kernel image

frezbo, 2022-01-17:
logs when `config.txt` is updated with `kernel=kernel8.img` and `cmdline.txt` added

```bash
Read start4.elf bytes  2240224 hnd 0x00027dfd 
Read fixup4.dat bytes     5354 hnd 0x00018bce 
Firmware: 58e03c94953762222f2b838390dde54d46c38381 Jan  5 2022 12:29:20
0x00d03114 0x00000000 0x00000fff
MEM GPU: 32 ARM: 991 TOTAL: 1023
Starting start4.elf @ 0xfeb00200 partition 0
+

MESS:00:00:06.150819:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:06.153727:0: brfs: File read: 596 bytes
MESS:00:00:06.208491:0: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:06.218008:0: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:06.227516:0: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:06.237030:0: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:06.246540:0: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:06.256054:0: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:06.265561:0: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:06.275076:0: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:06.284586:0: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:06.294100:0: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:06.298598:0: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:06.324082:0: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:06.333597:0: HDMI1:EDID error reading EDID block 0 attempt 1
MESS:00:00:06.343107:0: HDMI1:EDID error reading EDID block 0 attempt 2
MESS:00:00:06.352621:0: HDMI1:EDID error reading EDID block 0 attempt 3
MESS:00:00:06.362129:0: HDMI1:EDID error reading EDID block 0 attempt 4
MESS:00:00:06.371643:0: HDMI1:EDID error reading EDID block 0 attempt 5
MESS:00:00:06.381153:0: HDMI1:EDID error reading EDID block 0 attempt 6
MESS:00:00:06.390667:0: HDMI1:EDID error reading EDID block 0 attempt 7
MESS:00:00:06.400175:0: HDMI1:EDID error reading EDID block 0 attempt 8
MESS:00:00:06.409689:0: HDMI1:EDID error reading EDID block 0 attempt 9
MESS:00:00:06.414190:0: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:06.420490:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:07.612920:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:07.620216:0: *** Restart logging
MESS:00:00:07.621636:0: brfs: File read: 596 bytes
MESS:00:00:07.631629:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:07.641668:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:07.651698:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:07.661736:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:07.671768:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:07.681806:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:07.691836:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:07.701874:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:07.711907:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:07.721945:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:07.726966:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:07.737591:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:07.747629:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:07.757659:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:07.767697:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:07.777730:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:07.787768:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:07.797797:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:07.807836:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:07.817868:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:07.827907:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:07.832928:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:07.850247:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:07.860287:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 1
MESS:00:00:07.874122:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 2
MESS:00:00:07.884219:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 3
MESS:00:00:07.894324:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 4
MESS:00:00:07.904420:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 5
MESS:00:00:07.914506:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 6
MESS:00:00:07.924601:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 7
MESS:00:00:07.934690:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 8
[... 25 lines cut ...]
MESS:00:00:08.272239:0: brfs: File read: 1073 bytes
MESS:00:00:08.275242:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:08.279440:0: Read command line from file 'cmdline.txt':
MESS:00:00:08.285291:0: 'talos.platform=metal console=tty0 console=ttyAMA0,115200 init_on_alloc=1 slab_nomerge pti=on consoleblank=0 nvme_core.io_timeout=4294967295 random.trust_cpu=on printk.devkmsg=on ima_template=ima-ng ima_appraise=fix ima_hash=sha
MESS:00:00:08.417865:0: brfs: File read: 249 bytes
MESS:00:00:08.424420:0: brfs: File read: /mfs/sd/bl31.bin
MESS:00:00:08.426753:0: Loading 'bl31.bin' to 0x0 size 0xb06b
MESS:00:00:08.432255:0: brfs: File read: 45163 bytes
MESS:00:00:09.647195:0: brfs: File read: /mfs/sd/kernel8.img
[... cut ...]
MESS:00:00:16.524539:0: uart: Baud rate change done...
MESS:00:00:16.526592:0: uart: Baud rate change done...
MESS:00:00:16.532309:0: bfs_xhci_stop
MESS:00:00:16.534834:0: XHCI-STOP
MESS:00:00:16.538028:0: xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
MESS:00:00:16.545186:0: USBSTS 18
NOTICE:  BL31: v2.6(release):
NOTICE:  BL31: Built : 16:06:15, Dec 27 2021
Read start4.elf bytes  2240224 hnd 0x00027dfd 
Read fixup4.dat bytes     5354 hnd 0x00018bce 
Firmware: 58e03c94953762222f2b838390dde54d46c38381 Jan  5 2022 12:29:20
```

frezbo, 2022-01-17:
I'm not sure if this is a bug on u-boot/grub/dtbs, but in this case the console is broken only when the overlays are changed, the u-boot.grub and upstream kernel version remains the same

pelwell, 2022-01-17:
Can you explain what you mean by these two statements?:
    "the console is broken only when the overlays are changed"
and:
    "this has something to do with overlays, when using 1.20211029 firmware dtb's, directly loading the kernel from config.txt works, switching back to the next bracnh console is broken again"

How are you changing just the overlays?

I don't think there is anything fundamentally wrong with the latest firmware and/or the 5.15 kernels used in the `next` branch, but I acknowledge that you have a problem with your configuration.

frezbo, 2022-01-17:
Scenario one:

* the first partition contains files from `1.20211029` tag of firmware repository

Serial console over ttyAMA0 works

Second scenario:

* the first partition files are overwritten from files from `boot` folder from `next` branch of firmware

serial console over ttyAMA0 doesn't work.

In both scenarios the upstream kernel version is unchanged

I hope this helps, let me know if you need anything extta

pelwell, 2022-01-17:
I understand - "overlays" didn't just refer to the `overlays` directory, but rather the whole of the firmware.

> I hope this helps, let me know if you need anything extra

This isn't a problem I'll be investigating further. I've established that the kernel and firmware in the `next` branch have a working ttyAMA0 serial port when used with the majority of your config.txt. If somebody more familiar with U-boot and Grub - software we don't support - can help to pinpoint the underlying cause (starting with reproducing the issue) and demonstrate that we've broken something then we'll obviously try to fix it, but until then this won't get any attention.

closed 2022-01-17 by pelwell

frezbo, 2022-01-17:
Thanks, but removing u-boot+grub combo and directly booting the kernel also has the same effect.

pelwell, 2022-01-17:
> The kernel used is from `torvalds/linux` not using the `raspberrypi/linux`

Editing a comment to add significant information like this is bad practise - there is no notification of the change, so the person who has read the original comment in the email will probably miss it.

ssh-ing in to look at the kernel logs would remove the current doubt over whether this is non-working serial port or a non-booting kernel. Adding `earlycon=pl011,0xfe201000` might also get you some output if it's crashing early on.

frezbo, 2022-01-17:
i apologize for the edits and confusion. the kernel indeed boots up and I can access it via the network, it's just that the console is not working, I'll will try adding the `earlycon` and report back

pelwell, 2022-01-17:
earlycon is unlikely to help then, but it's still worth a try.

Have you thought to post the kernel log retrieved over ssh?

frezbo, 2022-01-17:
After adding the `earlycon` I can see the console logs. Here's console logs with `earlycon` enabled copied from serial console:

```bash
Read start4.elf bytes  2240224 hnd 0x00027dfd 
Read fixup4.dat bytes     5354 hnd 0x00018bce 
Firmware: 58e03c94953762222f2b838390dde54d46c38381 Jan  5 2022 12:29:20
0x00d03114 0x00000000 0x00000fff
MEM GPU: 32 ARM: 991 TOTAL: 1023
Starting start4.elf @ 0xfeb00200 partition 0
+

MESS:00:00:06.145336:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:06.148212:0: brfs: File read: 597 bytes
MESS:00:00:06.203014:0: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:06.212530:0: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:06.222037:0: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:06.231552:0: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:06.241062:0: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:06.250577:0: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:06.260084:0: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:06.269599:0: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:06.279109:0: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:06.288623:0: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:06.293121:0: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:06.318615:0: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:06.328130:0: HDMI1:EDID error reading EDID block 0 attempt 1
MESS:00:00:06.337640:0: HDMI1:EDID error reading EDID block 0 attempt 2
MESS:00:00:06.347154:0: HDMI1:EDID error reading EDID block 0 attempt 3
MESS:00:00:06.356662:0: HDMI1:EDID error reading EDID block 0 attempt 4
MESS:00:00:06.366176:0: HDMI1:EDID error reading EDID block 0 attempt 5
MESS:00:00:06.375686:0: HDMI1:EDID error reading EDID block 0 attempt 6
MESS:00:00:06.385200:0: HDMI1:EDID error reading EDID block 0 attempt 7
MESS:00:00:06.394708:0: HDMI1:EDID error reading EDID block 0 attempt 8
MESS:00:00:06.404222:0: HDMI1:EDID error reading EDID block 0 attempt 9
MESS:00:00:06.408723:0: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:06.415044:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:07.507843:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:07.515139:0: *** Restart logging
MESS:00:00:07.516559:0: brfs: File read: 597 bytes
MESS:00:00:07.526551:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:07.536591:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:07.546621:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:07.556659:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:07.566692:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:07.576730:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:07.586760:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:07.596798:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:07.606830:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:07.616868:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:07.621889:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:07.632514:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:07.642552:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:07.652582:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:07.662619:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:07.672651:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:07.682689:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:07.692718:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:07.702757:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:07.712789:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:07.722828:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:07.727848:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:07.745175:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:07.755215:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 1
MESS:00:00:07.769041:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 2
MESS:00:00:07.779139:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 3
MESS:00:00:07.789228:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 4
MESS:00:00:07.799340:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 5
MESS:00:00:07.809427:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 6
MESS:00:00:07.819521:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 7
MESS:00:00:07.829609:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 8
[... 1034 lines cut ...]
 SUBSYSTEM=pci
 DEVICE=+pci:0000:01:00.0
192.168.15.202: kern:    info: [2022-01-17T14:07:57.052016495Z]: hub 4-0:1.0: USB hub found
 SUBSYSTEM=usb
 DEVICE=+usb:4-0:1.0
192.168.15.202: kern:    info: [2022-01-17T14:07:57.052924495Z]: hub 4-0:1.0: 4 ports detected
 SUBSYSTEM=usb
 DEVICE=+usb:4-0:1.0
192.168.15.202: kern:     err: [2022-01-17T14:07:57.155444495Z]: pinctrl-bcm2835 fe200000.gpio: could not add GPIO chip
 SUBSYSTEM=platform
 DEVICE=+platform:fe200000.gpio
192.168.15.202: kern:     err: [2022-01-17T14:07:57.161028495Z]: pinctrl-bcm2835 fe200000.gpio: could not add GPIO chip
 SUBSYSTEM=platform
 DEVICE=+platform:fe200000.gpio
[... cut ...]
192.168.15.202: kern:  notice: [2022-01-17T14:07:57.174214495Z]: cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
192.168.15.202: kern: warning: [2022-01-17T14:07:57.175170495Z]: platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
 SUBSYSTEM=platform
 DEVICE=+platform:regulatory.0
192.168.15.202: kern:    info: [2022-01-17T14:07:57.176863495Z]: cfg80211: failed to load regulatory.db
192.168.15.202: kern:    info: [2022-01-17T14:07:57.204933495Z]: ALSA device list:
```

frezbo, 2022-01-17:
Update: after adding `earlycon=pl011,0xfe201000`, with `next` firmware, serial console with u-boot+grub combo also works, but `ttyAMA0`  is not present under `/dev`. Falling back to `1.20211029` firmware, `ttyAMA0` is present under `/dev`

pelwell, 2022-01-17:
It might save time if you can run `dtc -I fs /proc/device-tree > tx.txt` and then upload the output somewhere (don't be alarmed by the warnings).

frezbo, 2022-01-17:
Uploaded the output of `dtc -I fs /proc/device-tree` at https://gist.github.com/frezbo/a158688ad5aca8a7cfba81cb9fa1a1ff

frezbo, 2022-01-17:
I tried the `bcm2711-rpi-4-b.dtb` from the upstream kernel, but then it fails to load the `disable-bt` overlay

```bash

MESS:00:00:07.964630:0: dtb_file 'bcm2711-rpi-4-b.dtb'
MESS:00:00:07.968162:0: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
MESS:00:00:07.972737:0: Loading 'bcm2711-rpi-4-b.dtb' to 0x100 size 0x67a9
MESS:00:00:07.989268:0: brfs: File read: 26537 bytes
MESS:00:00:07.999023:0: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
MESS:00:00:08.037595:0: brfs: File read: 1703 bytes
MESS:00:00:08.041239:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:08.045230:0: brfs: File read: 598 bytes
MESS:00:00:08.051440:0: brfs: File read: /mfs/sd/overlays/disable-bt.dtbo
MESS:00:00:08.058311:0: dterror: no symbols found
MESS:00:00:08.060102:0: Failed to resolve overlay 'disable-bt'
MESS:00:00:08.073133:0: dterror: no symbols found
MESS:00:00:08.076980:0: dterror: no symbols found
MESS:00:00:08.083235:0: brfs: File read: 1073 bytes
MESS:00:00:08.086542:0: Failed to open command line file 'cmdline.txt'
MESS:00:00:08.093551:0: dterror: no symbols found
MESS:00:00:08.097857:0: dterror: no symbols found
MESS:00:00:08.110037:0: dterror: no symbols found
MESS:00:00:08.115964:0: dterror: no symbols found
MESS:00:00:08.121673:0: dterror: no symbols found
MESS:00:00:08.156264:0: dterror: no symbols found
MESS:00:00:08.170604:0: dterror: no symbols found
MESS:00:00:08.199227:0: brfs: File read: /mfs/sd/u-boot.bin
MESS:00:00:08.201693:0: Loading 'u-boot.bin' to 0x80000 size 0x8f848
MESS:00:00:08.207778:0: Device tree loaded to 0x2eff9000 (size 0x6f9a)
MESS:00:00:08.215757:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:08.223088:0: uart: Baud rate change done...
MESS:00:00:08.225110:0:

```

pelwell, 2022-01-17:
The DT dump shows that the DT node for UART0 is enabled, and yet the driver isn't starting (other than the `AMBA PL011 UART driver` message from the init method). It's likely that there is a driver required by the downstream dtb that isn't present in the upstream kernel, and that is preventing the UART0 device instance from being created.

The upstream kernel doesn't acknowledge DT overlays - if you want to change the configuration you have to write a new dts/dtb file for it. The standard upstream configuration is to have UART0 for Bluetooth and UART1 for the console, with no way to switch. Is there some reason you have to use the upstream kernel rather than one of ours?

frezbo, 2022-01-17:
Thanks for taking time to look through this. We can't use downstream rpi kernel since out product is based on upstream kernel and generic (talos.dev). Initially I got the PoE fan working by using upstream kernel and combining `bcm2711-rpi-4-b.dts` with the poe overlay. Since rpi-5.15 has the upstream driver and the right overlay it just worked. (Sorry for the tangent on the PoE hat, it was what initially made me use the `next` branch of firmware). But it's strange that the upstream kernel works fine with the `1.20211029` firmware and serial console is only broken when the `next` branch is used. As a workaround I can add `earlycon=pl011,0xfe201000` to the cmdline

pelwell, 2022-01-17:
The earlycon workaround is only useful for console output - /dev/ttyAMA0 won't be available for anything else. If you don't care about /dev/ttyAMA0 then use the upstream dtb as is without the earlycon patch and you should get serial output from /dev/ttyS0 instead - be sure to change your cmdline.txt to say "console=ttyS0,115200".

frezbo, 2022-01-17:
yeh, just the kernel logs