raspberrypi/firmware #1683: No boot, eight short flashes in a row, no video output - Raspberry PI 4

Fledermausi, 2022-01-16:
For several days now, I've been trying to make a supposedly brand new Raspberry pi 4 boot. I am a complete newbie to Raspberry PI, and this is my first Raspberry Pi board. So far this has been exceptionally frustrating. The green LED light produces eight quick flashes in a sequence (not the four  long ones followed by four short ones, but just a burst of eight rapid flashes followed by nothing. After that, only the red light stays on). I've tried three different SD cards (and will probably try a third one), I tried two different HDMI cables as well as a "normal" full size HDMI cable connected with an adapter. I've also tried formatting and installing OS on the micro SD cards in a Linux computer, and I've tried two different memory card readers.  I cannot find any references to eight rapid greed LED flashes in a sequence. Any advice or help would be appreciated.

doe300, 2022-01-16:
This looks very similar to #1671 , have you had a loot at that issue?

JamesH65, 2022-01-16:
8 short flashes is an SDRAM failure, the device will need to be returned to your supplier and replaced under warranty. Sorry its been a pain, its unusual for this to happen (they are all tested working at the factory), so must have failed since manufacture, perhaps during transport.

closed 2024-01-08 by timg236

KibGim, 2025-04-04:
Hi,
I'm also experiencing random system freezes and sometimes the 8 green flashes on boot (as described here https://github.com/raspberrypi/rpi-eeprom/issues/675 by @timg236  ).  
Occasionally, my Raspberry Pi 4 (2GB model) starts up, shows the boot screen and desktop, but then suddenly displays green vertical lines and becomes unresponsive. Many other times just black screen. Tested with official power suppy only. Tested with a second hdmi cable but that's not the problem (also all official equipment is basically brand new).

My Pi came pre-installed with an SD card and Pi-hole, which worked fine for a while until it stopped booting after a long period of not being used. I'm trying to get it running again, but no success so far.

Do you think this patch or issue might also apply to the Pi 4? 
I'm still within the warranty period ** only for +/- a week ** and I'm considering sending it back—but I want to make sure it's a real hardware issue and not something they could blame on me. I just want to confirm that it's truly not functioning properly.

Thanks for any insight!

popcornmix, 2025-04-05:
The [forum](https://forums.raspberrypi.com/index.php) would be a better place to ask than a old, closed, issue.