raspberrypi/firmware #1684: Can't boot CM4 via NVMe behind PCIe switch / bridge

geerlingguy, 2022-01-18:
**Is this the right place for my bug report?**

Yes

**Describe the bug**

While I can boot a CM4 using NVMe storage if it is plugged directly into the CM4's PCI Express bus, I _cannot_ boot a CM4 from NVMe if I have the NVMe SSD plugged in through a PCIe switch or bridge (e.g. via a chip like the ASM1184e).

Many use cases for the CM4 include a PCI Express bridge so multiple PCI Express devices can share the x1 Gen 2 lane on the CM4.

**To reproduce**

Assuming you have an NVMe SSD that is already configured to boot when connected directly to the Raspberry Pi Compute Module 4 IO Board's PCIe slot:

  1. Use either an external PCIe expansion board like [this x1 to 4 board](https://pipci.jeffgeerling.com/cards_adapter/zs-zhishang-pcie-x1-to-4-pcie-x16-expansion-riser.html)
  2. Plug an NVMe SSD into one of the four slots
  3. Try booting the Raspberry Pi

The Raspberry Pi will not recognize the NVMe drive behind the bridge as a bootable device.

**Expected behaviour**

The bootloader should recognize NVMe drives behind PCI Express bridges.

**Actual behaviour**

The bootloader does not recognize NVMe drives behind PCI Express bridges.

**System**

* Which model of Raspberry Pi?

Compute Module 4

* Which OS and version (`cat /etc/rpi-issue`)?

```
Raspberry Pi reference 2021-10-30
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, c12b1df4ed6416fb0df33ba1731c5b13c1bdbdf8, stage2
```

* Which firmware version (`vcgencmd version`)?

```
Oct 29 2021 10:47:33 
Copyright (c) 2012 Broadcom
version b8a114e5a9877e91ca8f26d1a5ce904b2ad3cf13 (clean) (release) (start)
```

* Which kernel version (`uname -a`)?

```
Linux turing-node-1 5.10.63-v8+ #1459 SMP PREEMPT Wed Oct 6 16:42:49 BST 2021 aarch64 GNU/Linux
```

**Logs**

N/A

**Additional context**

There has been discussion on the forums, e.g. [this post by peterharperuk on the forums](https://forums.raspberrypi.com/viewtopic.php?p=1848378#p1848378):

> Lack of support for booting from an NVMe bridge/switch is an issue with the way the bootloader works. So until this is fixed you'll have to boot with a sd card I'm afraid. I'll shout on here when we have something to try.

However I couldn't find any issue in the bootloader repository (nor anything tracking it in any of the other repositories), so I thought I'd file an issue here.

fanoush, 2022-01-19:
maybe this belongs to https://github.com/raspberrypi/rpi-eeprom/issues as "firmware" is the stuff that already sits on nvme drive in this case

timg236, 2022-01-19:
It’s not supported

closed 2022-01-19 by pelwell

fanoush, 2022-01-19:
We know it is currently not supported. Note the quote by  @peterharperuk "booting from an NVMe bridge/switch is an issue with the way the bootloader works. So until this is fixed you'll have to boot with a sd card I'm afraid. I'll shout on here when we have something to try." This looks like it may be fixed in future so it may make sense to have issue open for that.

pelwell, 2022-01-19:
If it can be reopened now, it can be reopened later.

timg236, 2022-01-19:
It's a known feature request but there's no plan to do it in the near future. 
At the very least the PCIe (NVMe) reset investigation that @peterharperuk has been looking at is a prerequisite but there's also a fairly large list of higher priority items ahead of this.

geerlingguy, 2022-01-19:
I'm okay with open or closed, my main purpose was to have a place on GitHub to track the actual issue, since it's a lot harder on the Pi Forums ;)

arqtv, 2022-01-31:
Closing an Issue with "Fixed" or "Will not fix" is fine but what is the reason for closing this issue? I do very much appreciate the heavy workload and that there are other priorities but my concern with closing this issue because of this is that it will not get any attention any time soon?  And like Jeff, I can't find this issue or the "known feature request" that Tim refers to above anywhere else? If it is "live" as a feature request or an issue somewhere else, Tim could you say exactly where please? And we can track it there. If not perhaps it should be re-opened as a bootloader issue?

timg236, 2022-01-31:
It’s a feature request not a bug and would part of rpi-eeprom (although you’d need new firmware as well). This isn’t being worked on and I don’t believe it’s a planned feature. New features are announced first via beta testing on the forums or blogs.

arqtv, 2022-02-01:
Yes having to raise a feature request as an "issue" is an anomaly with Github not having a separate thread for "feature requests".  I am aware there is limited space left in the eeprom and perhaps this is the root reason why a feature like this will not or perhaps cannot be implemented. No fault in saying cannot if true. As much if not all of what is needed for this feature is wrapped up in closed source it is hard for non RPi people to see the trees. If this is a "can't be done" (for lack of eeprom space or other technical reasons) rather than a "won't be done" then I for one will move on. If there is no technical reason and it "can be done" but there is no available technical resource within RPi at this time to do it then is this something one of RPi's Approved Design Partners could tackle? Guidance would be much appreciated.

JamesH65, 2022-02-01:
This is not something that could be done by design partners as they do not have access to the required information. I suspect it can be done, but is so low on the list of priorities (Very low use case vs considerable amounts of other work that needs doing) that it will never be done.

hpmerkel, 2022-02-01:
In my opinion booting from NVME is a major topic especially for CM4 boards. More an more new devboards are using bridged bus to support multive devices. Of course i understand that there are plenty of feature request. But using Raspberry Pi with AI components opens a totally new customer area by using Dual Edge TPUs. This would bring lots of performance on AI however slow SD card currently.

timg236, 2022-02-01:
The recommended method is to use a CM4 with EMMC, or a Lite +SD or even a USB2 MSD for the boot-partition.

JamesH65, 2022-02-01:
I believe it's B. It requires access to closed firmware source, VC compilers and knowledge that is only available to Pi engineers.

fanoush, 2024-06-05:
> If it can be reopened now, it can be reopened later.

As per https://github.com/raspberrypi/firmware/issues/1833 it looks like this is working for Pi5.
Maybe this issue could be reopened for CM4 too?

JamesH65, 2024-06-05:
@peterharperuk FYI

peterharperuk, 2024-06-05:
It's more complicated for cm4 (because of start.elf) and the code we added for Pi5 didn't work on CM4, so more investigation needed. But yes, we've talked about adding support for CM4 when we have time.

arqtv, 2024-06-05:
That sounds like a CM5 might be able to boot directly from an SSD behind a PCIe switch without much if any additional effort, when the CM5’s become available? That will be a bonus.