raspberrypi/firmware #1685: Raspberry Pi 4B not booting kernel image named unconventionally

iostapyshyn, 2022-01-18:
I am having a boot folder with two identical 5.15.15-v8+ kernel images, one named `Image` and the second named `kernel8.img`.

With `kernel=kernel8.img` line in `config.txt`, the system boots normally without issues.

However, if I try to use the differently named kernel image (for example `Image` and obviously `kernel=Image` is in the config), the boot process fails very early (black screen and nothing on serial port).

I recall using arbitrary names for my kernel images just recently on Raspberry Pi 2B V1.1, by putting the filename in the config.txt. Is this a bug, or intended behavior? Am I missing something?

Firmware is up to date.

pelwell, 2022-01-18:
I suspect you are missing `arm_64bit=1`. With non-standard kernel name, the firmware doesn't know that the ARM must be started in 64-bit mode unless you tell it so.

iostapyshyn, 2022-01-19:
> I suspect you are missing `arm_64bit=1`. With non-standard kernel name, the firmware doesn't know that the ARM must be started in 64-bit mode unless you tell it so.

Thanks, that did indeed solve my problem. Does `kernel=kernel8.img` without `arm_64bit=1` force the processor into 64bit mode implicitly?

I believe mentioning this quirk in https://www.raspberrypi.com/documentation/computers/linux_kernel.html would be helpful, as the article mentions ability to use arbitrary filename using `kernel=`. I will prepare a pull request for this.

closed 2022-01-19 by iostapyshyn

pelwell, 2022-01-19:
Yes, because the image name is recognised as being for a 64-bit kernel.