raspberrypi/firmware #1686: Latest kernel/bootloader v1:1.20220118-1 breaks boot

MichaIng, 2022-01-19:
**Describe the bug**
After the latest `raspberrypi-kernel`/`raspberrypi-bootloader` package upgrade to v1:1.20220118-1, all RPi models do not boot anymore.

**To reproduce**
```
apt update
apt upgrade
reboot
```

**Expected behaviour**
Boot

**Actual behaviour**
No boot, more precisely, early boot stage is done but then stalls at rainbow flash, respectively with CPU load LED (when configured) staying lid.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? All models: https://github.com/MichaIng/DietPi/issues/5187
* Which OS and version (`cat /etc/rpi-issue`)? Bullseye
* Which firmware version (`vcgencmd version`)? `version c8fa168f5965406a472b058131772316bddef735 (tainted) (release) (start)`
* Which kernel version (`uname -a`)? `5.10.92`

**Logs**
I'll add them quickly but wanted to do the report earlier, hoping that the update can be set on hold repository wise?

**Additional context**

XECDesign, 2022-01-19:
Strange that nothing like this has come up from rpi-update users or any of the internal testing. Since it's 11PM, I'm partially reverting the repo and we can take a closer look tomorrow.

MichaIng, 2022-01-20:
> Strange that nothing like this has come up from rpi-update users or any of the internal testing.

Indeed. I tested v5.10.8x via `rpi-update` which worked well, but not the newest one. Probably also something with the packaging went wrong.

https://forums.raspberrypi.com/viewtopic.php?p=1963599#p1963599 seems to boot, at least services are running, just the screen black. I'll verify with a fresh Raspberry Pi OS image tomorrow.

popcornmix, 2022-01-20:
> Indeed. I tested v5.10.8x via rpi-update which worked well, but not the newest one

Can you confirm exactly what the last working rpi-update was?

MichaIng, 2022-01-20:
I'm not 100% sure sadly, I only see that `/lib/modules/5.10.85+` is/was present on my RPi Zero W, which was obviously working fine.

MichaIng, 2022-01-20:
Okay, I wasn't able to replicate on a fresh Raspberry Pi OS, so went through all differences and finally identified a device tree overlay which has been invalidated with latest kernel.

In case of less than 16 MiB GPU memory, the VCSM driver fails to load and throws kernel errors at boot, hence we deploy a device tree overlay to prevent it from loading in the first place, when less than 32 MiB GPU memory is applied. The path `/proc/device-tree/soc/vcsm` however does not exist anymore on newest kernel, and it seems that the bootloader denies to boot when a device tree overlay tries to alter an invalid path. I explicitly asked about this and got an answer by RPi engineers that it shouldn't be able to break boot that time 🤔: https://forums.raspberrypi.com/viewtopic.php?t=290607

Only commenting the related `dtoverlay=` entry fixes the failing boot.

Not sure whether it is intended and whether it would be possible to have the bootloader ignoring attempts to alter an invalid path/target?

pelwell, 2022-01-20:
The problem was limited to one function. Although it first appeared that it could have affected `target = <&...>` as well as `target-path = "..."`, in practice the missing symbol would have been caught already and the overlay skipped.

There's a trial firmware available in case you want to test it before the next release: https://drive.google.com/file/d/1xw68Xc_kdvI31q4-LQx840zufSQxoEgF/view?usp=sharing

pelwell, 2022-01-20:
For future reference, the order of fragment merging is:
1. Fragments that can be resolved against other fragments in the overlay, in the order in which they appear in the overlay.
2. All other fragments, in the order in which they appear in the overlay.

Fragment merging stops at the first error, even if it is considered non-fatal. Any fragments already merged will remain. As said previously, overlays that refer to symbols that don't appear in the overlay or the base dtb will be skipped before the fragment merging process. Had this overlay used `target = <&vcsm>;` then it would not have triggered this bug, but there is nothing wrong with the way the overlay was written.

Joulinar, 2022-01-20:
@pelwell 
I did quick testing on my RPI4B arm64. 

- Simply used `rpi-update` to update from `5.10.63-v8+` to `5.10.92-v8+`. 
- As expected system was not booting. 
- Copied your `trial firmware` into `/boot`
- Success, system was booting

```
root@DietPi4:~# uname -a
Linux DietPi4 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64 GNU/Linux
root@DietPi4:~#
```

pelwell, 2022-01-20:
Thanks - we'll get the patch merged internally, then turn the handle on an update.

pelwell, 2022-01-20:
> Does this mean that using symbols is generally and still (after the change) preferred,

I think it depends on the effect you are after. If the aim is to tie the overlay to a specific base DT then the path approach might be better, but for maximum compatibility and to prevent partial merging targeting a label is the best approach.

pelwell, 2022-01-20:
> Just to verify, /proc/device-tree/soc/vcsm has been removed, or just moved elsewhere?

The driver included by the vcsm node was removed in rpi-5.9.y. In rpi-5.10.y I noticed the absence of the driver and removed the DT node, since it was now vestigial.

There is some discussion of vcsm and the switch to vcsm-cma in https://github.com/raspberrypi/linux/issues/4112 and  https://github.com/raspberrypi/linux/issues/4167.

MichaIng, 2022-01-20:
Sounds like a great move so that, if I understand right, also with legacy graphics CMA is used and hence less explicit GPU memory required. This probably means that dedicated GPU memory doesn't need to be increased in any common case, desktops, video playback/games? `raspi-config` still raises it when selecting legacy RPi camera module/`start_x=1`, but I guess this is a specific case.

XECDesign, 2022-01-20:
Updated packages should be up now.

popcornmix, 2022-01-20:
vcsm-cma should also be fine with legacy (no kms or fkms). It is not dependent on display stack.

In general with 5.10 kernel (which doesn't support vcsm) the default gpu_mem should be fine for almost all use cases, and it's almost always a mistake to significantly increase it. You are just wasting memory that could be assigned to cma (where unused space is available to linux).

MichaIng, 2022-01-20:
@XECDesign 
Many thanks, I can verify that it now gracefully handles the invalid device tree overlay 👍.

@popcornmix 
Many thanks for clarifying. The legacy RPi camera module, respectively the additional feature provided by the extended firmware `start_x.elf` still makes use of it? Since `raspi-config` still assigns 128 MiB when legacy camera support is enabled. But we will adjust our config tool to now offer anything higher and that no software install will raise if above default.

Marking this issue hence as closed.

closed 2022-01-20 by MichaIng

popcornmix, 2022-01-21:
I think basic legacy camera usage (raspi-vid/raspi-still) is happy with default gpu_mem setting.
But you can cause extra buffers to be allocated (e.g. with pycamera), so it's hard to say it will always be enough.

With the newer arm side camera driver, cma is generally used for buffers, so gpu_mem is again unwanted.